motioneye-project/motioneyeos #2534: gdrive: command not found

AntouanK, 2020-09-28:
Hi there.

I was trying to follow the [FAQ on syncing with google drive](https://github.com/ccrisan/motioneyeos/wiki/Synchronize-with-Google-Drive), and I realised some binaries are missing.
`wget` was missing, but i found `curl`, so no problem.
I see `gdrive` is missing as well though
```
# gdrive
-sh: gdrive: command not found
```
how can I add it?
From what I understand the image is based on buildroot so there's no package manager.
Was `gdrive` supposed to be installed?

Thank you.

starbasessd, 2020-09-28:
Not necessary.
In the WebGUI
Open the settings
Goto File Storage
Turn on Enable File Uploads
Turn on or off Upload Pictures and Upload Movies as wanted
Change Upload Service to "Google Drive"
Change Location to folder/path you want to use (must already exist in
GDrive) (use a folder called motion in root of GDrive, Location is motion
I turn off Include Subfolders but you may want to leave it on
I turn off Clean Cloud, and manually delete them as I want
Click "Obtain Key" and follow steps to have it generate a key for you on
Google's website
Copy / Paste key into Authorization Key field.
Click on "APPLY" at the top << VERY IMPORTANT!!!
Click Test Service, and wait for 'Test Successful" banner at the top.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-09-28:
@jasaw & @ccrisan Here's one of the Wiki pages that needs serious
updating...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

AntouanK, 2020-09-28:
Great! Thank you @starbasessd 

That all worked great.

One more question.
I wanted to run nginx on top, to enable https and control it a bit better. 
Is that included in the motioneyeos somehow? nginx or something else?
( I know it's off topic, just wondering if I missed it in the wiki )

closed 2020-09-28 by AntouanK

starbasessd, 2020-09-28:
Nope, not currently present. Source code is available to build your own, if
you need it.
Look under 'Assets' for the versions listed here:
https://github.com/ccrisan/motioneyeos/releases

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

AntouanK, 2020-09-28:
hmm, build my own image you mean?
I think that will take me lots of time and I'll get stuck in some steps.

I was thinking of running a hypriot image instead, that has docker pre-installed and run motioneye in docker.
What's the trade-offs of running motioneye in a container?

starbasessd, 2020-09-28:
Yes, BYOI (Build Your Own Image), and add/subtract anything you want.
Trade-offs: Whatever CPU usage for stuff running in the background (Like in
RaspberryPi Lite, or Ubuntu Server, and such)
I run a Lenovo i5 with Ubuntu Server running motion/motionEye, I run a VBox
Ubuntu Server guest with motion/motionEye
and have tried a docker instance, but it had too many issues, and docker
supervisor taking a percentage of CPU with no benefit (for me).
My VBox host has an 8 core/16 thread AMD processor, 32GB RAM, 10TB storage,
3 1GBit ethernet & 1 11ac Wifi, running Ubuntu 20.04 server & VirtualBox.
I run a bunch of cameras permanently and a bunch of others for testing.
I have a Pi1, a couple of Pi2s, a BUNCH of Pi3Bs and 3B+s, and a couple of
Pi4B-4GB and a Pi4B-8GB,
One of my networks is set up for Pi Net Boot.
I can usually test alost any user's environment...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

AntouanK, 2020-09-29:
Ok, I see.
I am planning to use a Pi2 B and I can see that the cpu usage is low with one camera.
I'll try with docker and see what's the difference.
I was asking more about accessing the device and filesystem. Will there be issues with that?
I can see online people post guides about it, so it should be relatively simple once you pass access to the container.

Thank you.