motioneye-project/motioneyeos #2537: Boot panic - caused by network.

Tesla-Nikola-2012, 2020-10-05:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 20200606).

### Board Model

I am using the following board/model: Raspberry Pi 3B+.

### Camera

I am using the following type of camera: Unknown specific type.  Generic raspberry pi cam.

My camera model is: Unknown specific type.  Generic raspberry pi cam.

### Network Connection

My motionEyeOS unit is connected to the network via: This error happens before network configuration is set up immediately after fresh img install to SD card.

### Peripherals

I am using the following peripherals that I consider relevant to this issue:
NONE

### Log Files

I consider the following log files relevant to this issue:
No log files exist.  Pi remains in a constant boot loop.

### Main Issue
Upon power on from a fresh install of the img to an SD card, the Pi goes through a constant boot loop because when it gets to the "Configuring wired network" step it panics if there is no wired connection currently hooked up. 

The panic error says "panic: rebooting in xx seconds (caused by network)".

I know that if you simply hook up an ethernet cable to the Pi, then the boot will continue normally.

However, the Pi should continue to boot normally even if there is no ethernet cable plugged in during boot and let the user configure the network from the command line if they desire instead of forcing the user to physically plug an ethernet cable in during first boot.

Not all, but many users will want to connect to a wireless connection immediately and not have to run an ethernet cable to the Pi upon first boot.

starbasessd, 2020-10-05:
There are several ways to deal with this, all involve creating or copying file(s) to the /boot partition after clean image and before installing to the Pis.
1) copy a wpa_supplicant.conf to the /boot partition:
     https://github.com/ccrisan/thingos/wiki/wpa_supplicant.conf
2) Disable networking discovery:
     create a file called os.conf
     insert a single line:
          OS_NETWORKLESS="true"
3) Create an Access Point setup from the files and instructions in the attached file.
[ConfigFiles.zip](https://github.com/ccrisan/motioneyeos/files/5330589/ConfigFiles.zip)

Let me know if you need further assistance.

Dewey3, 2020-10-07:
@Tesla-Nikola-2012 can you confirm if one of @starbasessd recommended fixes worked for you?  I reported the exact same issue a couple weeks ago, with video of the panic rebooting: [https://github.com/ccrisan/motioneyeos/issues/2526](url).

starbasessd, 2020-10-07:
Did you test?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Tesla-Nikola-2012, 2020-10-07:
@starbasessd @Dewey3 I will try to test this afternoon when I get off work.  It seems like the recommended workaround should work though, but I'll confirm this afternoon.

This might be nitpicking, but would it be better/easier to add in some type of prompt if there is no networking discovered that asks the user to input wireless details during boot?  For people who have experience with Linux and the command line it's a simple process to edit the config files before booting, but for others this may not be the easiest or most straight forward approach.

Tesla-Nikola-2012, 2020-10-07:
@Dewey3 Also, if you have the ability during boot to temporarily connect the Pi via ethernet then you can just use the eth0 IP address to configure the WPA details in the web GUI and not have to roll back to a previous version.

It's a little bit of an inconvenience to do so, but it'll keep you on the most recent release.

starbasessd, 2020-10-07:
I just built a dev20200907 using file:
https://github.com/ccrisan/motioneyeos/releases/download/nightly-dev/motioneyeos-raspberrypi3-dev20200907.img.xz
Installed my wpa_supplicant.conf, and ssh.txt files.
Inserted into Pi3B+
Powered on.
Went through normal boot process, connected to wifi as expected, accessible
through wifi IP address.

I just built a 20200606 using file
https://github.com/ccrisan/motioneyeos/releases/download/20200606/motioneyeos-raspberrypi3-20200606.img.xz
Installed my wpa_supplicant.conf, and ssh.txt files.
Inserted into Pi3B+
Powered on.
Went through normal boot process, connected to wifi as expected, accessible
through wifi IP address.
Both attempts on a 4GB SDCard. Also known good with latest
RaspberryPiOS-Lite 20200820, same 2 files, same results on same card.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Tesla-Nikola-2012, 2020-10-07:
Thank you @starbasessd .  These instructions really should be added to the install instructions.  The install instructions currently state:

"Of course your motionEyeOS needs an IP address before you can communicate with it so you'll have to use the ethernet connection with DHCP enabled, unless you have preconfigured a static IP address."

However for those who may not have the ability to use ethernet then these WIFI instructions should be added as well.

starbasessd, 2020-10-07:
Which link/page do you see that? I am currently organizing a bunch of
updates (Links to latest versions already completed).

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Tesla-Nikola-2012, 2020-10-07:
https://github.com/ccrisan/motioneyeos/wiki/Installation

starbasessd, 2020-10-07:
Thank you. I will look as to how to adjust the instructions.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Tesla-Nikola-2012, 2020-10-07:
No problem, thanks for looking into this.

alsamentertainment, 2020-10-13:
I had this issue and found that, for some weird reason, Windows knocked off the .conf extension at some point along the way.  Re-adding the .conf fixed it.

mikeharris2017, 2020-10-16:
I have the same issue with motioneyeos-raspberrypi-20200606.img on a Raspberry Zero W. Using a Windows PC. 
I only added my SSID and PW in https://github.com/ccrisan/thingos/wiki/wpa_supplicant.conf before copying it to the SD.
I started my Pi and everything went well. I set the camera solution to 1280x720 an hit "apply" and did the reboot. While browsing through the further Camera settings the Zero rebooted without hitting the apply-Button again. 
From then on it started showing "starting wpa_supplicant: failed (brcmfmac)".

Tried to start again from scratch (write image, copy my wpa_supplicant.conf without success. 

I will try the recommendations over the weekend.

starbasessd, 2020-10-16:
Make sure you are using the proper format of wpa_supplicant.comf contents.
Do NOT use Notepad, it doesn't create the file correctly. Use a
unix-friendly editor like Notepad++
-----
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
country=US

network={
    ssid="ssid"
    psk="password"
key_mgmt=WPA-PSK
}
-----

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

mikeharris2017, 2020-10-17:
Hi Starbasessd, thanks.
I use PSPad or TextPad. My first attempt was with Textpad. I realized this morning that my wifi repeaters got an update yesterday an switched wifi off, so the connection to the router was instable as too far. Must be Murphy. I am really sorry for this. 

Setting up the Pi this morning with PSPad to Unix and only the setting:
```
update_config=1
ctrl_interface=/var/run/wpa_supplicant

network={
    ssid="mynet"
    psk="mynet1234"
}
```
The Pi and camera are running fine now. Thanks for the quick support. 

I had a look into the config files today. This is also an interesting option. 
Is there a way to combine both configs? I would like to install the pi in a nesting box to be able to watch the birds breeding. Therefore I can't enter the pi after installation till the tennants move out at the end of September. 
My ideal setting would be to connect with my home wifi at startup. If no wifi, the pi should set up it's own wifi with the config files instead of going into panic mode. "Calm down and set up a wifi" :-)
This would allow me some troubeshooting and modification to repair the connection without disturbing the birds. 

Is there a way to archieve this failover?

Thank you very much!

starbasessd, 2020-10-17:
Unfortunately the AP vs connecting to a home wifi are mutually exclusive,
at least with the single on-board wifi. I doubt it would work even if you
added a 2nd wifi dongle.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

mikeharris2017, 2020-10-18:
What a pity :-( I will try the stability of the setup for a few days and see which way I go. A second wifi dongle is not the best option when running on a powerbar. 
My issue is solved for now. Thanks for the quick and helpful support.

mikeharris2017, 2020-10-18:
Sorry I for asking again: it is not possible to config the setup to 
Try to connect to wifi
If success goto end
If failed 
Then setup as AP
?

Currently the system would do
If failed 
Then reboot after 60s

This way it is not both at the same time. 

Thanks for clarification

starbasessd, 2020-10-18:
That would take a re-write, I believe, for an esoteric situation. I don't
know if @ccrisan would want to consider it. He is always open to others
submitting changes, though, if you would want to do it.
Source code is available to build your own to test...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

mikeharris2017, 2020-10-18:
I would be happy to add new changes. My problem is that my coding skills are just slightly above zero and not very helpful.

ccrisan, 2020-10-18:
> Sorry I for asking again: it is not possible to config the setup to
> Try to connect to wifi
> If success goto end
> If failed
> Then setup as AP
> ?
> 
> Currently the system would do
> If failed
> Then reboot after 60s
> 
> This way it is not both at the same time.
> 
> Thanks for clarification

As a matter of fact, switching to AP mode upon network failure is something that I've been meaning to implement. I won't promise anything, but it's probably one of the first tasks I'm willing to consider, when it comes to the OS.

mikeharris2017, 2020-10-18:
Hi Calin, sounds great. I keep my fingers crossed.

towfeeqf, 2021-02-01:
Device: Raspberry pi Zero
MotionEyeOS version: 20200606 

I struggled for 2 days with the issue and so writing in details for MANY out there who gets stuck due to this SMALL but important error (bear with me for my detailed small solution):

Error:
**“Configuring wired network: no device
Panic: rebooting in 100 seconds (caused by network) …….”**  

This was rebooting my RPI zero in a loop.

I was able to resolve the problem which was causing me trouble due to the file: “wpa_supplicant.conf”

I created this new file and save it in the SD card folder and entered the Wifi configuration settings correctly as shown below (There’s no password and so NONE is shown below):
…….
country=UK
update_config=1
ctrl_interface=/var/run/wpa_supplicant

network={
	scan_ssid=1
	ssid="My network ssid"
	key_mgmt=NONE
}

The problem was **due to the File extension**…………….this was the most important part…..I was missing for 2 days.

There were clear Instructions to remove the .txt extension file but I thought that, adding .conf would remove the txt extension by itself. It was not the case, .txt was still there and therefore during the boot, it wasn't finding the "wpa_supplicant.conf" file which was needed, instead the file available was "wpa_supplicant.conf.txt".
 
**Solution to my problem** : point 7 in the link https://github.com/ccrisan/motioneyeos/wiki/Wifi-Preconfiguration 
..........
   
7. Now, right click on this file and rename it wpa_supplicant.conf. Make sure to remove the .txt file extension.
........

To view the file extensions in a folder I visited this page: 
https://www.quora.com/How-do-I-get-rid-of-the-txt-extension-at-the-end-of-Text-Files-in-Windows 

(sometimes we need to get back to basics to make something work) 
I hope this helps someone!

Osmii, 2021-03-07:
Hello!
I also had problems with wifi droping(totaly first and new setup,Rpi2 with Tp-link dongle) with latest and other releases. Then I connected to a different AP  on other location and now is working ok.

Sven-Massard, 2022-09-26:
I am trying to install on a Raspberry Pi Zero. I have been trying for more than two hours now. Same problem as above.

Tried (nearly unedited on purpose):
`(base) sven@sven-Lenovo-V320-17IKB:/media/sven/254f5940-2de7-477e-9f4e-b858e96dacf3/boot$ sudo touch wpa_supplicant.conf
[sudo] password for sven:
(base) sven@sven-Lenovo-V320-17IKB:/media/sven/254f5940-2de7-477e-9f4e-b858e96dacf3/boot$ nano wpa_supplicant.conf
(base) sven@sven-Lenovo-V320-17IKB:/media/sven/254f5940-2de7-477e-9f4e-b858e96dacf3/boot$ sudo !!
sudo nano wpa_supplicant.conf
(base) sven@sven-Lenovo-V320-17IKB:/media/sven/254f5940-2de7-477e-9f4e-b858e96dacf3/boot$ cat wpa_supplicant.conf
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
country=US

network={
    ssid="Sven und Melissa"
    psk="removed after pasting"
key_mgmt=WPA-PSK
`
I tried instructions

> There are several ways to deal with this, all involve creating or copying file(s) to the /boot partition after clean image and before installing to the Pis.
>
>     1. copy a wpa_supplicant.conf to the [...]

Doesn't work. Used "Raspberry Pi Imager" on Windows, but edited all files on Ubuntu afterwards.

starbasessd, 2022-09-27:
Don't use Raspberry Pi Imager. Use something like Win32DiskImager or even Etcher or dd.
If you use the presets in RPiImager, it will mess up / corrupt the SDCard.
Which version are you using of motionEyeOS? 20200606 or dev20201026?
Use Notepad++ to create / edit the /boot/wpa_supplicant.conf file in Windows rather than using Ubuntu,
There may be an issue with spaces in the SSID name. One way to check is to install  RaspberryPiOS Buster using the same wpa_supplicant.conf file and see if it works there.

Sven-Massard, 2022-09-27:
Omg...
On Ubuntu: `sudo ./writeimage.sh -d /dev/sdd -i "./motioneyeos-raspberrypi-dev20201026.img" -n 'Sven und Melissa:removed'`
It worked right out of the box :)
Thank you very much!