motioneye-project/motioneyeos #2538: Recording every 4 minutes

Mike1236, 2020-10-06:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version 

I am running motionEyeOS version: motionEyeOS 20200606

### Board Model

I am using the following board/model: 

Raspberry Pi 4 2GB as hub
3x Raspberry Pi Zero W

### Camera

I am using the following type of camera: Fast Network Camera

My camera model is:

2x Raspberry Pi Camera v2
1x Raspberry Pi Camera v2 NoIR

### Network Connection

My motionEyeOS unit is connected to the network via: 

Hub is connected by Ethernet
Pi Zeros are wireless

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

None

So my issue is the video is being recorded every 4 minutes and saves 60 seconds. I've set it for continuous recording with maximum movie length of 60 seconds, preserve movies for 1 week, disabled motion detection.
Not sure why it's only recording 60 seconds every 4 minutes instead of every minute?

[image]

starbasessd, 2020-10-06:
May I suggest you install dev20200907 from
https://github.com/ccrisan/motioneyeos/releases
There are many issues in the 20200606 release. Let me know, please.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Mike1236, 2020-10-06:
Hi again,

Should I install dev20200907 on all the devices?

starbasessd, 2020-10-06:
No, just the hub (RPi4-2GB)

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-10-23:
Status?