motioneye-project/motioneyeos #2539: No media being recorded to Network Share or media browser

zdoucette97, 2020-10-07:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 20200606 (0.42.1)

### Board Model

I am using the following board/model: Raspberry PI Zero W

### Camera

I am using the following type of camera: MMAL Camera

My camera model is: Raspberry Pi Camera Module V2

### Network Connection

My motionEyeOS unit is connected to the network via: WiFi

### Peripherals

I am using the following peripherals that I consider relevant to this issue: None

### Details

Hello! This is my first time using motioneyeOS or any Raspberry Pi project for that matter, so please bear with me.

First question; if you have a Network Share enabled, will the Picture and Movie Browser still have media available to watch if the media is not being locally stored on the microsd?

Secondly, the issue I am having is that my Network share is up and running, and Test Share is successful every time, but when I walk back and forth at my East facing front door to test, no media has been recorded so far. 
There is neither media in the two media browsers, OR the Network Share. The Network Share created an empty KEEP file when I tried testing but that's it. I waited 30 minutes in case it just takes a while and nothing is there.
I should also note that I created a local user to use their credentials for the share. On the share, Everyone has Read, MEUser has Change, and Administrators have Full Control.
[image]

I am also currently using a working schedule, which is right now. So hopefully that isn't part of the problem.
[image]

I also used a small mask to hide the road so that passing cars and people won't be recorded. Only those approaching my front door will. 
[image]

Any help would be greatly appreciated!

starbasessd, 2020-10-07:
If you are using an SMB (Windows) share for storage, and you pick it under
File Storage, Storage Device, then no, that is where it will push your
pictures and movies to, without saving a local copy.
If you turn on Upload Media Files, then set it to FTP, or SFTP for local
network, or GDrive or DropBox for cloud storage, it will keep a local file
(If you set your custom path correctly) and upload a copy to the other
drive.
You may need to lower your Motion Detection, Frame Change Threshold to a
VERY low number (like 0.1%) to get it to register correctly.
I also set Motion Gap to 2, Captured Before and Captured after to 6-10, and
Minimum Motion Frames to 2 for most of my cameras. (800x600 or higher
resolution)
Working Schedule looks fine if you want it to only work M-F 7:30A - 5:30P.
I assume you changed the Time Zone in General Settings.
Under Motion Detection, until you have it set the way you want, I would
turn on Show Frame Changes, it'll show you where it thinks there's motion.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

zdoucette97, 2020-10-07:
> 
> If you are using an SMB (Windows) share for storage, and you pick it under File Storage, Storage Device, then no, that is where it will push your pictures and movies to, without saving a local copy. If you turn on Upload Media Files, then set it to FTP, or SFTP for local network, or GDrive or DropBox for cloud storage, it will keep a local file (If you set your custom path correctly) and upload a copy to the other drive. You may need to lower your Motion Detection, Frame Change Threshold to a VERY low number (like 0.1%) to get it to register correctly. I also set Motion Gap to 2, Captured Before and Captured after to 6-10, and Minimum Motion Frames to 2 for most of my cameras. (800x600 or higher resolution) Working Schedule looks fine if you want it to only work M-F 7:30A - 5:30P. I assume you changed the Time Zone in General Settings. Under Motion Detection, until you have it set the way you want, I would turn on Show Frame Changes, it'll show you where it thinks there's motion.
> […](#)
> -- Thanks Kevin Shumaker Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel> N38° 19' 56.52" W85° 45' 8.56" Semper Gumby “Don't tell people how to do things. Tell them what to do and let them surprise you with their results.” - G.S. Patton, Gen. USA Ethics are what we do when no one else is looking. Quis custodiet ipsos custodes? “There is no end to the good you can do if you don’t care who gets the credit.” - C Powell You know we're sitting on four million pounds of fuel, one nuclear weapon and a thing that has 270,000 moving parts built by the lowest bidder. Makes you feel good, doesn't it?

i see. now I just feel silly. I set frame change to 5% so it would stop recording the shadows of my mum's lilacs. thanks a bunch!

closed 2020-10-07 by zdoucette97