motioneye-project/motioneyeos #2541: Using both Network Share and Google Drive for File Storage yields no recordings when computer is off

zdoucette97, 2020-10-08:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 20200606 (0.42.1)

### Board Model

I am using the following board/model: Raspberry PI Zero W

### Camera

I am using the following type of camera: MMAL Camera

My camera model is: Raspberry Pi Camera Module V2

### Network Connection

My motionEyeOS unit is connected to the network via: WiFi

### Peripherals

I am using the following peripherals that I consider relevant to this issue: None

### Details

Hello! I used a Network Share to test my first recordings successfully and have just added my Google Drive as an additional file storage option. The Share was only to test them because I am fully aware that once my hosting computer is turned off, no recordings can be sent, but I was still hoping to use the Network Share as additional contingency.

My first test upload worked perfectly for both the Share and Drive when my computer was on. When I shut my computer off, however, my Google Drive did not receive any uploads.

I had assumed that my Google Drive would receive recordings even when my computer was offline. Is there a simple solution to this issue? Thank you for reading!

starbasessd, 2020-10-08:
Please confirm your settings in motionEye WebGUI, they should look something like this:
[image]
(IIRC you have your network share in the Storage Device & Root Directory settings)

zdoucette97, 2020-10-08:
> 
> 
> Please confirm your settings in motionEye WebGUI, they should look something like this:
> (IIRC you have your network share in the Storage Device & Root Directory settings)

[image]

I see. Yes, it looks like this

starbasessd, 2020-10-08:
It is probably failing because it is unable to create the 'original' file, thus it can't find the 'original' file to upload to GDrive. 
IOW motionEye creates the movie or snapshot file in the File Storage, Storage Device, first. then does the upload / copy to GDrive. Since it cannot find the place in part 1 it fails.
You would need a NAS (Network Attached Storage) or computer that you are uploading to to be on all the time so the motionEye can create the file there.
 If you cannot, then have it create the file locally and do the GDrive storage. 
Then clean up the files as needed on that local drive.

zdoucette97, 2020-10-08:
ok. thank you very much!

zdoucette97, 2020-10-08:
> It is probably failing because it is unable to create the 'original' file, thus it can't find the 'original' file to upload to GDrive.
> IOW motionEye creates the movie or snapshot file in the File [...]

hi! sorry, but what is the name of the default location for the microsd?

starbasessd, 2020-10-08:
In my screenshot

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-10-08:
My bad.
/data/output/Camera1

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-10-08:
BTW, either will work just change Camera1 or FrontDrive to your camera
name....

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

zdoucette97, 2020-10-08:
ok! thanks a bunch for your swift help!

closed 2020-10-08 by zdoucette97