motioneye-project/motioneyeos #2542: Green line's

ramybt, 2020-10-10:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: (insert your version here, e.g. 20200907).

### Board Model

I am using the following board/model: (insert your board model here, e.g. Raspberry PI 3B+).

### Camera

I am using the following type of camera: (choose from V4L2, MMAL, Network Camera, Fast Network Camera and Simple MJPEG Camera).
Ps3 camera, Dell laptop camera module 

My camera model is: (insert your camera model, e.g. Logitech C290).
Ps3 camera, Dell laptop camera module 

### Network Connection
Wifi

My motionEyeOS unit is connected to the network via: (choose your network connection, e.g. WiFi, Ethernet).
Wifi

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * (e.g. USB WiFi adapter)
 * (e.g. External Hard Disk)

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)
 

I switched from 2019 version to 20200907 and now i have green line's showing when i review recorded videos, only when i review/watch
My movie pass is ON but when i turn in off it seem to fix the green lines but than each video that used to take around 260mb now takes 2GB and thats 30 minutes
So this is what i mean by green line's down here
[image]

Thanks you for any help in advanced

starbasessd, 2020-10-10:
Try downgrading back to 20200606 or upgrading to dev20201008 and report on
whether or not either fix it.
There are several other reports of the same issue, and it is being looked
at.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

ccrisan, 2020-10-11:
At the suggestion of @popoviciri, upgrading to ffmpeg 4.3.1 may bring some fixes with regards to this problem. The next nightly build will contain ffmpeg 4.3.1.

JohanD78, 2020-10-25:
Try the latest prerelease (dev20201025) from here: https://github.com/ccrisan/motioneyeos/releases

Fixed the issue for me on both pi3 and pi4 with omx encoding :)

ramybt, 2021-01-14:
New releases has fix the issue!

closed 2021-01-14 by ramybt