motioneye-project/motioneyeos #2544: Motioneye fails to start on Pi Zero W

mmaxbar, 2020-10-11:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 20200606
### Board Model

I am using the following board/model: Pi Zero W

### Camera

I am using the following type of camera:

My camera model is:  Pi Camera 2 NoIr
### Network Connection

My motionEyeOS unit is connected to the network via: Wifi

After writing a fresh image of 20200606 to SD card and putting wpasupplicant.conf into the root, i have problems starting motioneye os. 
I connected HDMI for debugging: motioneye os starts ok until starting motioneye, this fails after some time and the device reboots. After going back to the (fresh) image 20190911 it works flawless.

starbasessd, 2020-10-11:
The file is wpa_supplicant.conf
It needs to follow a format like:
-----
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
country=US

network={
    ssid="SSID"
    psk="passphrase"
key_mgmt=WPA-PSK
}
-----
with 'country=XX' your country
with 'ssid="SSID" ' with quotes and is case sensitive
with 'psk="password" ' with quotes and is case sensitive

BTW, make sure that on the back of the Pi, it says "Raspberry Pi Zero W
V1.1" ( or Zero WH if it has the GPIO header installed )
Several people have reported recently that they've received original Pi
Zeros (without the W or WH) from somewhat dodgy dealers.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

mmaxbar, 2020-10-11:
> The file is wpa_supplicant.conf It needs to follow a format like: ----- ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev update_config=1 country=US network={ ssid="SSID" psk="passphrase" key_mgmt=WPA-PSK } ----- with 'country=XX' your country with 'ssid="SSID" ' with quotes and is case sensitive with 'psk="password" ' with quotes and is case sensitive BTW, make sure that on the back of the Pi, it says "Raspberry Pi Zero W V1.1" ( or Zero WH if it has the GPIO header installed ) Several people have reported recently that they've received original Pi Zeros (without the W or WH) from somewhat dodgy dealers.
> […](#)
> -- Thanks Kevin Shumaker Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel> N38° 19' 56.52" W85° 45' 8.56" Semper Gumby “Don't tell people how to do things. Tell them what to do and let them surprise you with their results.” - G.S. Patton, Gen. USA Ethics are what we do when no one else is looking. Quis custodiet ipsos custodes? “There is no end to the good you can do if you don’t care who gets the credit.” - C Powell You know we're sitting on four million pounds of fuel, one nuclear weapon and a thing that has 270,000 moving parts built by the lowest bidder. Makes you feel good, doesn't it?

The file is named correctly, wpa_supplicant.conf, it was just a typo in my original post describing the issue, network is up. I can ping the device while it is trying to start motioneye which fails after some time, so I am quite sure that networking is not part of the issue.

starbasessd, 2020-10-11:
> motioneyeos/blob/master/.github/CONTRIBUTING.md guide before opening this
> motioneyeos/wiki/FAQ before opening this issue. motionEyeOS Version I am
> AEZTUHKHKWGWEMZBPCS3ETDSKFVGZANCNFSM4SLT7QOA .

What's the last things it says on the monitor before it reboots?
Which image did you use? Full filename?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

mmaxbar, 2020-10-11:
I used the one from the wiki. 
Now the strange thing is that I cannot reproduce the error right now. 
I am literally doing exactly the same thing I did yesterday, apart from using a different SD card and downloaded the image again. I'm closing it for now, will reopen if I'm able to reproduce it...

closed 2020-10-11 by mmaxbar