motioneye-project/motioneyeos #2545: Unable to SSH into new install - Password incorrect and/or Pi user not found

alsamentertainment, 2020-10-13:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 20200606.

### Board Model

I am using the following board/model: Raspberry Pi Zero W
### Camera

I am using the following type of camera: (choose from V4L2, MMAL, Network Camera, Fast Network Camera and Simple MJPEG Camera).

My camera model is: Raspberry Pi Camera Board v2

### Network Connection

My motionEyeOS unit is connected to the network via: Wifi

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

None

### Log Files

I consider the following log files relevant to this issue:

Unable to
[image]

 I am a new user to Pi, so not great at explaining my issue, but...here goes.
It's a new install, just flashed.  When I try to SSH or PuTTY it says the password is invalid for the pi user.  I am definitely typing raspberry.  My CMD ssh line is: ssh pi@192.168.1.40 

When I log in as root, and I try to access the password change it says the user pi does not exist.

I DO have a Pi4 NAS connected and have the keys set up to run passwordless, but I am not sure that's causing any conflict.  It does make me question the * Generating ssh host keys: done, and I am not sure if I need to have the corresponding key on my computer to log in.

Even using the IP address in a browser is prompting a user name and password, which is failing.

starbasessd, 2020-10-13:
Username is 'admin' (without the quotes, and all lower case)
Password is blank  (no password) until it's changed.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

alsamentertainment, 2020-10-13:
Thankyou so much.  Worked through the Web interface. Will try SSH after work. Cheers.

starbasessd, 2020-10-23:
Status?

closed 2020-10-25 by starbasessd