motioneye-project/motioneyeos #2548: How to update config file using command line

ZKAW, 2020-10-20:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 20200606

### Board Model

I am using the following board/model: Raspberry pi 3a+

### Camera

I am using the following type of camera: (choose from V4L2, MMAL, Network Camera, Fast Network Camera and Simple MJPEG Camera).

My camera model is: NoBrand 5M pixels nightvision

### Network Connection

My motionEyeOS unit is connected to the network via: Wifi

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * (e.g. USB WiFi adapter)
 * (e.g. External Hard Disk)

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)
 
Hello everyone.

I'm running motioneyeOS and I need to know how to apply the changes made to te .conf file using command line instead of the web interface.
I'm looking to do a script to adjust the brightness for day/night cycle.

Thank you, and if you need any more information tell me.

starbasessd, 2020-10-20:
I would use 2 copies, day.conf, and night.conf, and use a crontab to swap
the two files and force a reboot
day.conf
night.conf
crontab
* 6 * * * cp /data/etc/camera-1_day.conf /data/etc/camera-1.conf && reboot
* 18 * * * cp /data/etc/camera-1_night.conf /data/etc/camera-1.conf &&
reboot
This would swap at 6am and 6pm every day.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-10-20:
Oops...
crontab entries should begin with 0 6 * * * & 0 18 * * * not with the *...
My bad...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

ZKAW, 2020-10-20:
> I would use 2 copies, day.conf, and night.conf, and use a crontab to swap
> the two files and force a reboot
> day.conf
> night.conf
> crontab
> * 6 * * * cp /data/etc/camera-1_day.conf /data/etc/camera-1.conf && reboot
> * 18 * * * cp /data/etc/camera-1_night.conf /data/etc/camera-1.conf &&
> reboot
> This would swap at 6am and 6pm every day.
> 
> 
> > Preliminary Docs
> >
> > I confirm that I have read the CONTRIBUTING
> > guide before opening this issue.
> >
> > I confirm that I have read the FAQ
> > <https://github.com/ccrisan/motioneyeos/wiki/FAQ> before opening this
> > issue.
> > motionEyeOS Version
> >
> > I am running motionEyeOS version: 20200606
> > Board Model
> >
> > I am using the following board/model: Raspberry pi 3a+
> > Camera
> >
> > I am using the following type of camera: (choose from V4L2, MMAL, Network
> > Camera, Fast Network Camera and Simple MJPEG Camera).
> >
> > My camera model is: NoBrand 5M pixels nightvision
> > Network Connection
> >
> > My motionEyeOS unit is connected to the network via: Wifi
> > Peripherals
> >
> > I am using the following peripherals that I consider relevant to this
> > issue:
> >
> >    - (e.g. USB WiFi adapter)
> >    - (e.g. External Hard Disk)
> >
> > Log Files
> >
> > I consider the following log files relevant to this issue:
> >
> >    - (attach e.g. motioneye.log)
> >    - (attach e.g. boot.log)
> >
> > Hello everyone.
> >
> > I'm running motioneyeOS and I need to know how to apply the changes made
> > to te .conf file using command line instead of the web interface.
> > I'm looking to do a script to adjust the brightness for day/night cycle.
> >
> > Thank you, and if you need any more information tell me.
> >
> >
> 
> 
> Thanks
> 
> Kevin Shumaker
> 
> 
> Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>
> 
> N38° 19' 56.52"
> W85° 45' 8.56"
> 
> Semper Gumby
> “Don't tell people how to do things. Tell them what to do and let them
> surprise you with their results.” - G.S. Patton, Gen. USA
> Ethics are what we do when no one else is looking.
> Quis custodiet ipsos custodes?
> “There is no end to the good you can do if you don’t care who
> gets the credit.” - C Powell
> You know we're sitting on four million pounds of fuel, one nuclear weapon
> and a thing that has 270,000 moving parts built by the lowest bidder. Makes
> you feel good, doesn't it?

Indeed, but is there a way to do it without rebooting the raspberry PI? Like the same way that when you click apply in the web interface?

starbasessd, 2020-10-20:
You can try:
0 6 * * * cp /data/etc/camera-1_day.conf /data/etc/camera-1.conf &&
/sbin/service motioneye restart
0 18 * * * cp /data/etc/camera-1_night.conf /data/etc/camera-1.conf &&
/sbin/service motioneye restart

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

ZKAW, 2020-10-20:
> You can try:
> 0 6 * * * cp /data/etc/camera-1_day.conf /data/etc/camera-1.conf &&
> /sbin/service motioneye restart
> 0 18 * * * cp /data/etc/camera-1_night.conf /data/etc/camera-1.conf &&
> /sbin/service motioneye restart
> 
> 
> > I would use 2 copies, day.conf, and night.conf, and use a crontab to swap
> > the two files and force a reboot
> > day.conf
> > night.conf
> > crontab
> >
> >    - 6 * * * cp /data/etc/camera-1_day.conf /data/etc/camera-1.conf &&
> >    reboot
> >    - 18 * * * cp /data/etc/camera-1_night.conf /data/etc/camera-1.conf &&
> >    reboot
> >    This would swap at 6am and 6pm every day.
> >
> >
> > Preliminary Docs
> >
> > I confirm that I have read the CONTRIBUTING
> > https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md
> > guide before opening this issue.
> >
> > I confirm that I have read the FAQ
> > https://github.com/ccrisan/motioneyeos/wiki/FAQ before opening this
> > issue.
> > motionEyeOS Version
> >
> > I am running motionEyeOS version: 20200606
> > Board Model
> >
> > I am using the following board/model: Raspberry pi 3a+
> > Camera
> >
> > I am using the following type of camera: (choose from V4L2, MMAL, Network
> > Camera, Fast Network Camera and Simple MJPEG Camera).
> >
> > My camera model is: NoBrand 5M pixels nightvision
> > Network Connection
> >
> > My motionEyeOS unit is connected to the network via: Wifi
> > Peripherals
> >
> > I am using the following peripherals that I consider relevant to this
> > issue:
> >
> >    - (e.g. USB WiFi adapter)
> >    - (e.g. External Hard Disk)
> >
> > Log Files
> >
> > I consider the following log files relevant to this issue:
> >
> >    - (attach e.g. motioneye.log)
> >    - (attach e.g. boot.log)
> >
> > Hello everyone.
> >
> > I'm running motioneyeOS and I need to know how to apply the changes made
> > to te .conf file using command line instead of the web interface.
> > I'm looking to do a script to adjust the brightness for day/night cycle.
> >
> > Thank you, and if you need any more information tell me.
> >
> >
> >
> > Thanks
> >
> > Kevin Shumaker
> >
> > Personal Tech Support https://kevinshumaker.wixsite.com/thethirdlevel
> >
> > N38° 19' 56.52"
> > W85° 45' 8.56"
> >
> > Semper Gumby
> > “Don't tell people how to do things. Tell them what to do and let them
> > surprise you with their results.” - G.S. Patton, Gen. USA
> > Ethics are what we do when no one else is looking.
> > Quis custodiet ipsos custodes?
> > “There is no end to the good you can do if you don’t care who
> > gets the credit.” - C Powell
> > You know we're sitting on four million pounds of fuel, one nuclear weapon
> > and a thing that has 270,000 moving parts built by the lowest bidder. Makes
> > you feel good, doesn't it?
> >
> > Indeed, but is there a way to do it without rebooting the raspberry PI?
> > Like the same way that when you click apply in the web interface?
> >
> >
> 
> 
> Thanks
> 
> Kevin Shumaker
> 
> 
> Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>
> 
> N38° 19' 56.52"
> W85° 45' 8.56"
> 
> Semper Gumby
> “Don't tell people how to do things. Tell them what to do and let them
> surprise you with their results.” - G.S. Patton, Gen. USA
> Ethics are what we do when no one else is looking.
> Quis custodiet ipsos custodes?
> “There is no end to the good you can do if you don’t care who
> gets the credit.” - C Powell
> You know we're sitting on four million pounds of fuel, one nuclear weapon
> and a thing that has 270,000 moving parts built by the lowest bidder. Makes
> you feel good, doesn't it?

Okay I will try this later, thanks. I'll keep you in touch.

starbasessd, 2020-10-23:
Status?

ZKAW, 2020-10-23:
> Status?

Yes, sorry I absolutely forget to reply. It worked perfectly ! Thank you very much ;).

starbasessd, 2020-10-23:
Resolved

closed 2020-10-23 by starbasessd