motioneye-project/motioneyeos #2550: Mass storage not found

SlymHD, 2020-10-22:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: dev20201012

### Board Model

I am using the following board/model: Raspberry PI 4B

### Camera

I am using the following type of camera: MMAL

My camera model is: RPI 4 camera

### Network Connection

My motionEyeOS unit is connected to the network via: WiFi

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

External Hard Disk (formatted ext4)

[motioneye.log](https://github.com/ccrisan/motioneyeos/files/5422341/motioneye.log)
[motion.log](https://github.com/ccrisan/motioneyeos/files/5422342/motion.log)
[dmesg.log](https://github.com/ccrisan/motioneyeos/files/5422343/dmesg.log)
[boot.log](https://github.com/ccrisan/motioneyeos/files/5422344/boot.log)

starbasessd, 2020-10-22:
It discovers SDA1 & SDA2 (and supports UAS)
1.895133] scsi host0: uas
[    1.896690] scsi 0:0:0:0: Direct-Access     WDC WD10 JPVX-75JC3T0
0301 PQ: 0 ANSI: 6
[    1.898733] sd 0:0:0:0: [sda] 1953525168 512-byte logical blocks: (1.00
TB/932 GiB)
[    1.898752] sd 0:0:0:0: [sda] 4096-byte physical blocks
[    1.898971] sd 0:0:0:0: [sda] Write Protect is off
[    1.898989] sd 0:0:0:0: [sda] Mode Sense: 53 00 00 08
[    1.899410] sd 0:0:0:0: [sda] Disabling FUA
[    1.899433] sd 0:0:0:0: [sda] Write cache: enabled, read cache: enabled,
doesn't support DPO or FUA
[    1.900172] sd 0:0:0:0: [sda] Optimal transfer size 33553920 bytes not a
multiple of physical block size (4096 bytes)
[    2.078624]  sda: sda1 sda2
[    2.081888] sd 0:0:0:0: [sda] Attached SCSI disk
Does it not show up under /data?
If you run in console or ssh session:
df -h <enter>
or
mount <enter>
what do they show?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

SlymHD, 2020-10-22:
[image]
[image]

starbasessd, 2020-10-22:
In console or ssh type:
mkdir /data/storage <enter>
mkdir /data/storagee/sda1 <enter>
mkdir /data/storage/sda2 <enter>
mount /dev/sda1 /data/storage/sda1 <enter>
mount /dev/sda2 /data/storage/sda2 <enter>
df -h <enter>
if this works, let me know, I'll give instructions on how to auto-mount

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

SlymHD, 2020-10-22:
Nope, still doesnt work and when I'm trying to mopunt it says that there arent such directories
[image]

starbasessd, 2020-10-22:
Checking

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

cap9qd, 2020-10-22:
Looks like you missed a / in your mount commands.

SlymHD, 2020-10-22:
Where?

cap9qd, 2020-10-22:
On the mounting location. It's not data/... it is /data/... If you compare
your screen shot with what starbassed sent you you will see it.

starbasessd, 2020-10-22:
mount /dev/sda1 /data/storage/sda1 <enter>
mount /dev/sda2 /data/storage/sda2 <enter>
before the data

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-10-22:
Thanks @cap9qd

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

SlymHD, 2020-10-22:
[image]

cap9qd, 2020-10-22:
No worries. I'm sorry I spelled your handle wrong...

starbasessd, 2020-10-22:
Couple of questions:
what kind of external drive?
Are you plugging in before power up or after?
I just tested with a USB Thumb Drive:
[image]
Worked fine.

starbasessd, 2020-10-22:
@cap9qd if that's my worst issue today I'm doing good...

SlymHD, 2020-10-22:
It's a WD Blue HDD 2,5", it's always plugged in, nothing works

SlymHD, 2020-10-22:
And it's internal

starbasessd, 2020-10-22:
Just checked an external usb3 1 TB physical drive
[image]
working auto-mount
[image]
Does your drive work in RaspberryPiOS on the 4? My 4B 4GB and 4B 8GB both work normally.

starbasessd, 2020-10-22:
Internal? as in a HAT? THAT could be why it's not working...

SlymHD, 2020-10-22:
HAT?

starbasessd, 2020-10-22:
A card connected to the GPIO pins to provide SATA connections. Not
connected through USB2 or USB3...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

SlymHD, 2020-10-22:
https://randomnerdtutorials.com/cctv-raspberry-pi-based-system-storage-motioneyeos/ bcs I was following this guide and they also used an internal 2,5 drive

starbasessd, 2020-10-22:
Describe your system completely, please.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-10-22:
Describe YOUR system completely, please.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

SlymHD, 2020-10-22:
[image]
It's a Raspberry pi 4b connect via adapter to a HDD (I hope that you can see everything on the photo)

starbasessd, 2020-10-22:
In my Pi4B 4GB
When I go to the WebGUI, Settings, File Storage, I get the following options:
[image]

"Mass Storage" is no longer an option.
My toshiba 1TB USB3 connected shows (currentl, for testing) as SDA1 and SDA2.
What is listed in yours (not including Custom Path, Network, or the /dev/mmc...)

SlymHD, 2020-10-22:
That's the problem there is nothing
[image]

SlymHD, 2020-10-22:
[image]
when I type in fdisk -l it says that the drive type is microsoft storage spaces even though I formatted it in ext4, is that probably the problem?

starbasessd, 2020-10-22:
where/how did you partition and format the drive?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

SlymHD, 2020-10-22:
I formatted it with mini partittion wizard

starbasessd, 2020-10-22:
In a linux box, or Win10 box?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

SlymHD, 2020-10-22:
Well, I clicked on disk management, then right click on the disk, format and choosed ext4

SlymHD, 2020-10-22:
on win10

starbasessd, 2020-10-22:
you can re-do it in motionEyeOS by doing the following in the command line
(console or ssh)
fdisk /dev/sda <enter>
d <enter>
1 <enter>
d <enter>
n <enter>
p <enter>
1 <enter>
<enter>
<enter>
ignore errors and reboot
back into command liine
mkfs.ext4 /dev/sda1
reboot
should now show in WebGUI, Settings, File Storage

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-10-22:
Win10 does not support ext4. Follow the instructions in the last message.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

SlymHD, 2020-10-22:
omg

SlymHD, 2020-10-22:
finally thank you

closed 2020-10-22 by SlymHD

starbasessd, 2020-10-22:
Glad to help.