motioneye-project/motioneyeos #2551: Pi Camera V2 camera mounted on my Pi Zero W simply will not work

bmeinhard57, 2020-10-23:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: (0.42.1).

### Board Model

I am using the following board/model: (Raspberry Pi Zero W).

### Camera

I am using the following type of camera: (V4L2).

My camera model is: (Raspberry Pi Camera Module v2).

### Network Connection

My motionEyeOS unit is connected to the network via: (WiFi).

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

None

### Log Files

I consider the following log files relevant to this issue:

 * (motioneye.log)
 * (boot.log)
 * (motion.log)

Hello - I have reviewed the guide and wiki before opening this issue.  I apologize for being the "noob", but my Pi Camera V2 camera mounted on my Pi Zero W simply will not work (files attached).  Is my camera not compatible?  Did I miss something in the setup?  Any and all help is most appreciated.  Bob

[boot.log](https://github.com/ccrisan/motioneyeos/files/5427766/boot.log)
[motion (1).log](https://github.com/ccrisan/motioneyeos/files/5427767/motion.1.log)
[motioneye (1).log](https://github.com/ccrisan/motioneyeos/files/5427768/motioneye.1.log)

starbasessd, 2020-10-23:
If it didn't auto-discover the camera, how did you add it? It should be
"Local MMAL Camera" for type.
You can confirm that the camera is working and set up correctly by
installing RaspberryPiOS from here:
https://downloads.raspberrypi.org/raspios_full_armhf_latest
(this is only to confirm the camera is working correctly)
If it works in an application like Cheese, then you can re-install
motionEyeOS-20200606 for the PiZero
and we can continue testing from there. We need to confirm the camera is
working 'normally' first.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

bmeinhard57, 2020-10-23:
Thank you very much.  I will do that and report back.

bmeinhard57, 2020-10-23:
The camera does NOT work with Cheese (it knows a camera is there, but no video) - I am contacting the supplier for a replacement.  Thanks.

starbasessd, 2020-10-23:
Good deal (finding out it's probably a camera issue...) and you are on the
way to resolving it.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

bmeinhard57, 2020-10-24:
The camera is on it's way back to supplier - thanks for the help.  We'll see how the replacement works...

starbasessd, 2020-11-05:
Status? Did you get a replacement camera?

bmeinhard57, 2020-11-05:
Hi - sorry for not posting sooner.  Thanks for the advice.  All is good now - camera is working great!  Bob

closed 2020-11-05 by starbasessd