motioneye-project/motioneyeos #2553: Ethernet connection problem

Dusko28, 2020-10-24:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 20200606

version Board Model

I am using the following board/model: RPI 4 - 4GB model

### Network Connection

My motionEyeOS unit is connected to the network via: Ethernet on raspberry

**Hello guys,
i try multitimes running motionEyeOs via ethernet cable conection. After first boot fresh installed system, everything work fine. DHCP give me IP adress. But after reboot, system cycle in bootloop caused by network. Configuring wired network: no link

Of course, i can set wifi conection on first boot. Thats work, but i wish use ethernet.

I try check connection status via _ip addr_ in terminal
I try manualy set IP addres for eth0 adapter via _ip addr add 192.168.1.22/24 dev eth0_ command

I hope for some advice, and thank you for making motioneyeos  **

[image]
[image]

starbasessd, 2020-10-24:
If you want a static IP address, use the static_ip.conf file when you
create the fresh image, and put it in the /boot partition
https://github.com/ccrisan/thingos/wiki/static_ip.conf
sample text [modify to your network settings]
=====
STATIC_IP="192.168.42.1/24"
STATIC_GW="192.168.42.1"
STATIC_DNS="192.168.42.1"
=====

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-10-24:
An alternative is to go to Settings, Network, and change
Ip Configuration from Automatic (DHCP) to
Manual (Static IP)
and add your settings there.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Dusko28, 2020-10-24:
Thanks for advice, but my problem is not set static IP adress. My problem is ethernet conection after first reboot.
1st time, system boot corectly and obtaine IP from DHCP server. I noting change, just log and give reboot command
second boot faild, ethernet connection dont work.
i must motioneyeos run only via wifi connection :(

starbasessd, 2020-10-24:
Yes, it is. You cannot set a static IP via the command line you posted.
In order to set a static IP address, you have to use one of the 2 options I
posted.
If you run the command you posted, it will break ethernet networking.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Dusko28, 2020-10-25:
ok, pls forget what i say about change IP adress. Thats just my hopefull trying fix problem :)

again:
i have fresh installation
1st boot with ethernet cable OK 
i do nothing, just reboot system
second boot dont work - configuring wired network: no link

I expect motionEyeOS with ethernet cable work anytime

starbasessd, 2020-10-25:
My setup:
RPi4-4GB, Gigabit Ethernet, 4GB microSDCard, image
https://github.com/ccrisan/motioneyeos/releases/download/20200606/motioneyeos-raspberrypi4-20200606.img.xz
No other changes.
First Boot:
login as: admin
Welcome to meye-2ba2d181!
[root@meye-2ba2d181 ~]# cat /var/log/boot.log
---- booting motionEyeOS 20200606 ----
 * Detecting disk device: /dev/mmcblk0
 * Creating data partition: done
 * Formatting data partition: done
 * Mounting data partition: done
 * Copying data skeleton: done
 * Mounting filesystems: done
 * Mounting overlay filesystems: done
 * Setting empty root password: done
 * Copying version file: done
 * Setting UTC timezone: done
 * Loading kernel modules: done
 * Setting hostname: done
 * Configuring CPU frequency: done
 * Starting syslogd: done
 * Loading device-tree overlays: done
 * Starting throttle watcher: done
 * Starting eudev: done
 * Starting watchdog: done
 * Starting rngd: done
 * Preparing AP interfaces: ap0
 * Configuring wired network: dhcp
 * Setting current date using SNTP: Sun Oct 25 11:39:20 UTC 2020
 * Starting ntpd: done
 * Starting crond: done
 * Generating ssh host keys: done
 * Starting sshd: done
 * Starting proftpd: done
 * Starting smbd: done
 * Starting nmbd: done
 * Starting motioneye: done
 # Interface eth0 has IP address 192.168.2.211/24
 # Default gateway is 192.168.2.1
 # DNS server address is 192.168.2.1
[root@meye-2ba2d181 ~]#
no changes, command given: reboot
logged back in as admin
[root@meye-2ba2d181 ~]# cat /var/log/boot.log
---- booting motionEyeOS 20200606 ----
 * Detecting disk device: /dev/mmcblk0
 * Creating data partition: done
 * Formatting data partition: done
 * Mounting data partition: done
 * Copying data skeleton: done
 * Mounting filesystems: done
 * Mounting overlay filesystems: done
 * Setting empty root password: done
 * Copying version file: done
 * Setting UTC timezone: done
 * Loading kernel modules: done
 * Setting hostname: done
 * Configuring CPU frequency: done
 * Starting syslogd: done
 * Loading device-tree overlays: done
 * Starting throttle watcher: done
 * Starting eudev: done
 * Starting watchdog: done
 * Starting rngd: done
 * Preparing AP interfaces: ap0
 * Configuring wired network: dhcp
 * Setting current date using SNTP: Sun Oct 25 11:39:20 UTC 2020
 * Starting ntpd: done
 * Starting crond: done
 * Generating ssh host keys: done
 * Starting sshd: done
 * Starting proftpd: done
 * Starting smbd: done
 * Starting nmbd: done
 * Starting motioneye: done
 # Interface eth0 has IP address 192.168.2.211/24
 # Default gateway is 192.168.2.1
 # DNS server address is 192.1---- shutting down motionEyeOS 20200606 ----
 * Stopping motioneye: done
 * Stopping smbd: done
 * Stopping nmbd: done
 * Stopping proftpd: done
 * Stopping sshd: done
 * Stopping crond: done
 * Stopping ntpd: done
 * Stopping network: done
 * Stopping rngd: done
 * Stopping eudev: done
 * Stopping throttle watcher: done
 * Stopping syslogd: done
---- booting motionEyeOS 20200606 ----
 * Detecting disk device: /dev/mmcblk0
 * Mounting filesystems: done
 * Mounting overlay filesystems: done
 * Loading kernel modules: done
 * Setting hostname: done
 * Configuring CPU frequency: done
 * Starting syslogd: done
 * Loading device-tree overlays: done
 * Starting throttle watcher: done
 * Starting eudev: done
 * Starting watchdog: done
 * Starting rngd: done
 * Preparing AP interfaces: ap0
 * Configuring wired network: dhcp
 * Setting current date using SNTP: Sun Oct 25 11:43:00 UTC 2020
 * Starting ntpd: done
 * Starting crond: done
 * Starting sshd: done
 * Starting proftpd: done
 * Starting smbd: done
 * Starting nmbd: done
 * Starting motioneye: done
 # Interface eth0 has IP address 192.168.2.211/24
 # Default gateway is 192.168.2.1
 # DNS server address is 192.168.2.168.2.1
================================
I am able to access the WebGUI in both sessions (first boot and 2nd boot)
at the same IP address.
No issues.
What are you doing differently?
Can you install RaspberryPiOS and reboot and get ethernet connection each
time?

On Sun, Oct 25, 2020 at 4:31 AM Dusko28 <notifications@github.com> wrote:

> ok, pls forget what i say about change IP adress. Thats just my hopefull
> trying fix problem :)
>
> again:
> i have fresh installation
> 1st boot with ethernet cable OK
> i do nothing, just reboot system [...]

Dusko28, 2020-10-25:
fresh installation first boot log: 
====================================
---- booting motionEyeOS 20200606 ----
 * Detecting disk device: /dev/mmcblk0
 * Creating data partition: done
 * Formatting data partition: done
 * Mounting data partition: done
 * Copying data skeleton: done
 * Mounting filesystems: done
 * Mounting overlay filesystems: done
 * Setting empty root password: done
 * Copying version file: done
 * Setting UTC timezone: done
 * Loading kernel modules: done
 * Setting hostname: done
 * Configuring CPU frequency: done
 * Starting syslogd: done
 * Loading device-tree overlays: done
 * Starting throttle watcher: done
 * Starting eudev: done
 * Starting watchdog: done
 * Starting rngd: done
 * Preparing AP interfaces: ap0
 * Configuring wired network: dhcp
 * Setting current date using SNTP: Sun Oct 25 12:47:57 UTC 2020
 * Starting ntpd: done
 * Starting crond: done
 * Generating ssh host keys: done
 * Starting sshd: done
 * Starting proftpd: done
 * Starting smbd: done
 * Starting nmbd: done
 * Starting motioneye: done
 # Interface eth0 has IP address 192.168.1.7/24
 # Default gateway is 192.168.1.1
 # DNS server address is 213.46.172.38
---- shutting down motionEyeOS 20200606 ----
 * Stopping motioneye: done
 * Stopping smbd: done
 * Stopping nmbd: done
 * Stopping proftpd: done
 * Stopping sshd: done
 * Stopping crond: done
 * Stopping ntpd: done
 * Stopping network: done
 * Stopping rngd: done
 * Stopping eudev: done
 * Stopping throttle watcher: done
 * Stopping syslogd: done
================================
no change, just reboot
second boot:

---- booting motionEyeOS 20200606 ----
 * Detecting disk device: /dev/mmcblk0
 * Mounting filesystems: done
 * Mounting overlay filesystems: done
 * Loading kernel modules: done
 * Setting hostname: done
 * Configuring CPU frequency: done
 * Starting syslogd: done
 * Loading device-tree overlays: done
 * Starting throttle watcher: done
 * Starting eudev: done
 * Starting watchdog: done
 * Starting rngd: done
 * Preparing AP interfaces: ap0
 * Configuring wired network: no link
---- shutting down motionEyeOS 20200606 ----
 * Stopping motioneye: failed
 * Stopping smbd: failed
 * Stopping nmbd: failed
 * Stopping proftpd: failed
 * Stopping sshd: failed
 * Stopping crond: failed
 * Stopping ntpd: failed
 * Stopping network: done
 * Stopping rngd: done
 * Stopping eudev: done
 * Stopping throttle watcher: done
 * Stopping syslogd: done

starbasessd, 2020-10-25:
Something's broken in your setup.How are you rebooting?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Dusko28, 2020-10-25:
just log as admin and put reboot command

starbasessd, 2020-10-25:
Can you try a different ethernet port on your switch?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Dusko28, 2020-10-25:
I try - no effect

on my RPI4 i normaly use other distribution like raspbian or libreelec. Network conecction via ethernet work great. I have more SD card with diferent systems. Im sad :(

Dusko28, 2020-10-25:
I try older version motionEyeOS for RPI4   

version 20200203 and 20190911
same issue

Dusko28, 2020-10-27:
I find problem and fix!

I read initial script in /etc/init.d/S40network

part on line 139 checking if ethernet working - time limit is maybe too short 

i edited this code for my use and put longer time limit - work good

i give original and edited script ( if someone use - remember rename to original name "S40network" )

i write pull request on github page
[S40network.zip](https://github.com/ccrisan/motioneyeos/files/5446941/S40network.zip)