motioneye-project/motioneyeos #2554: Ongoing Motion Eye issue

STUKguy, 2020-10-24:
motionEye Version | 0.41
Motion Version | 4.2.2
OS Version | motionEyeOS 20190911
Device: Raspberry PI4 2GB
Previous issue: #2520 

Hi, 
So for some reason my PI will lose its connection from 1 of my 2 USB cameras and then I have to reboot the whole pi to get it to work again. This will happen randomly. Here are the logs
[messages.log.txt](https://github.com/ccrisan/motioneyeos/files/5433913/messages.log.txt)
[motion.log.txt](https://github.com/ccrisan/motioneyeos/files/5433914/motion.log.txt)
[motioneye.log.txt](https://github.com/ccrisan/motioneyeos/files/5433915/motioneye.log.txt)

There is no power issues it gets a steady 5V3A
[image]

I've tried different USB ports unsure what else to try.

starbasessd, 2020-10-24:
Can you check the cameras by connecting them to a PC or MAC and see if they
crash there?
Are they identical make/model?
Is it always the same Camera or the same port?
Probably asked before and answered, but starting my process over...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

STUKguy, 2020-10-24:
@starbasessd they are both custom built cameras which worked fine for a year before I moved over to a PI4. What I could do is rebuild a laptop and plug the cameras into this but how would I go about this? Install Ubuntu then install motioneyeos?

starbasessd, 2020-10-24:
Install Ubuntu, then install cheese, or VLC and just watch the video.
 If the video never crashes, we've narrowed it down to something dealing
with the Pi4.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-10-24:
USB2 or USB3?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

STUKguy, 2020-10-25:
@starbasessd in that case I could just use windows and install VLC on that. They are both USB2 cameras

starbasessd, 2020-10-25:
Yep.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

STUKguy, 2020-10-25:
@starbasessd ok so i had to do a Ubuntu install managed to get one of the cameras to work in VLC but the one that was causing the issues yesterday is just not work????
[image]
unsure how to check VLC logs

starbasessd, 2020-10-25:
Logging isn't on by default...
https://askubuntu.com/questions/910037/where-is-the-vlc-logfile-for-errors

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

STUKguy, 2020-10-25:
@starbasessd just found the issue if i close VLC and then open it again and choose the other feed it works i just cant have both of them on at the same time????

starbasessd, 2020-10-25:
If they are the same make and model, and USB connected, very few programs understand there are 2 there. Network cams don't have the same issue. I only have 1 of each model of USB on my Pi or hub for the same issue, which is why I asked several times if they were the same make/model.

STUKguy, 2020-10-25:
@starbasessd they are not the same make/model they are custom bult from salvaged laptop webcams

starbasessd, 2020-10-25:
You could even connect them to 2 PiZeros, set to Fast Network Cam and use
them that way.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-10-25:
in the Ubuntu, do a lsusb -t with both connected and report the results

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

STUKguy, 2020-10-25:
@starbasessd 
[image]

starbasessd, 2020-10-25:
Mine:
kevin@camera-server:~$ lsusb -t
/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 5000M
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/10p, 480M
    |__ Port 9: Dev 2, If 0, Class=Video, Driver=uvcvideo, 480M
    |__ Port 9: Dev 2, If 1, Class=Video, Driver=uvcvideo, 480M
    |__ Port 9: Dev 2, If 2, Class=Audio, Driver=snd-usb-audio, 480M
    |__ Port 9: Dev 2, If 3, Class=Audio, Driver=snd-usb-audio, 480M
    |__ Port 9: Dev 2, If 4, Class=Mass Storage, Driver=usb-storage, 480M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=ehci-pci/2p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/6p, 480M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=ehci-pci/2p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
Showing a Generic TIGA Cam made by Pilot Dashcam.
another one shows the same, even though made by Pilot was a HD Dashcam
(differenct model)

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

STUKguy, 2020-10-25:
@starbasessd if im reading this right both are coincidently the same make...thats great

starbasessd, 2020-10-25:
Thats only showing 1 camera...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-10-25:
Port1, Dev2 with 2 interfaces indicates the same camera (one camera)
Video and Audio or 2 video
There would be another Port & Dev for the 2nd if discovered.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

STUKguy, 2020-10-25:
@starbasessd just plugged one camera into one side of the laptop and another into the other side and got this
[image]

starbasessd, 2020-10-25:
That is 2 cameras.
Now do:
lsusb [enter]

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-10-25:
mine:

kevin@camera-server:~$ lsusb
Bus 002 Device 002: ID 8087:8000 Intel Corp.
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 8087:8008 Intel Corp.
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 002: ID 1908:3256 GEMBIRD TIGA   Device
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-10-25:
The ID is for manufacturer and model. That is what most apps see and then
think they are the same device.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

STUKguy, 2020-10-25:
@starbasessd Thats great, i will look into selling th PI4 and getting 2 PI Zero-W with USB to micro USB adapters

starbasessd, 2020-10-25:
Interesting 1st part is the same, but the 2nd is different.
Should (repeat: should) work.
How were they used before you switched to motionEyeOS and Pi?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

STUKguy, 2020-10-25:
@starbasessd Nothing differant had them both connected to a PI2 board with motion eye and had no issues, all these issues started happening when i switched to a PI4

starbasessd, 2020-10-25:
Out of curiosity, can you set the Pi2 back up, for testing, with the
current version of motionEye 20190911, or try both the Pi2 and Pi4 with
20200606, or dev20201025?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

STUKguy, 2020-10-25:
@starbasessd i sold the PI2 to get the PI4

starbasessd, 2020-10-25:
:(
OK, can you try the 2 newer versions on the Pi4? One is a new build last
night.
I am wondering what changed in mEOS on the USB side. I know the USB
hardware did some major changes in the chipset used.
(added USB3, etc), but the USB2 ports should still react the same.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

STUKguy, 2020-10-25:
@starbasessd just flashed dev20201025 and got this PI4 with both cameras working the only setting i have touched in camera rotation and frame rate set to 25FPS. i havent set up Video storage, Timezone, admin password or anything else

starbasessd, 2020-10-25:
Go ahead and make those changes, then watch a while...
If you can have the Ubuntu machine watch ports 8081 & 8082 on the Pi4
(streaming output) in 2 tabs in Firefox too, that'd be great.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

STUKguy, 2020-10-25:
@starbasessd ok.......just set the admin password, set the timezone, and changed the host name. and now the PI is not booting???? going to flash and retry

STUKguy, 2020-10-25:
@starbasessd now im even more confussed so i redid dev20201025 put the SD card back in to the PI but everytime i change a setting that requires a reboot the PI just hangs for ages until i physically go and unplug the PI and plug it back in.

starbasessd, 2020-10-25:
Bad SDCard?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

STUKguy, 2020-10-25:
@starbasessd i would think that but this happened on 2 different SD cards

starbasessd, 2020-10-25:
I'm currently at a loss. Thinking...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

STUKguy, 2020-10-25:
@starbasessd so i found another older SD card and flashed 20200606 on it and pluged this into the PI it boots and reboots bt for some reason the options are greyed out
[image]

starbasessd, 2020-10-25:
Not logged in as admin?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

STUKguy, 2020-10-25:
@starbasessd thats the thing i am, as soon as i remove the cam the menu works but when i add the cameras back the menu greys out

starbasessd, 2020-10-25:
I am at a loss. No clue. I am open to suggestions...
You've tried everything.
I cannot reproduce it at all.
One more thing, are you on the latest firmware and flash on the Pi4?
Install RaspberrypiOS Desktop  (20200820
https://downloads.raspberrypi.org/raspios_armhf_latest) on an SDCard
Boot, then in Terminal:
sudo apt update && sudo apt full-upgrade -y [enter]
sudo reboot [enter]
In terminal:
rpi-eeprom-update [enter]
If it shows an update is available,
sudo rpi-eeprom-update -a
sudo reboot

You should then be able to try the other SDCard(s)
Anything after motionEyeOS-20200606 should have the corrected boot loader
fo4 the Pi4

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

STUKguy, 2020-10-25:
@starbasessd i will try this now, the only option i have is either to get 2 x  PIZeros or get a PI2 and see if i have the same issues there but that would be a even longer processess. Im flashing an image of Raspbian OS now see how this goes.

STUKguy, 2020-10-25:
@starbasessd  im assuming this means everything is up to date
[image]

starbasessd, 2020-10-25:
Yes, you are completely up to date.
I have installed 20190911, 20200203, 20200606 and all the dev2020wxyz
versions, and they all work on the Pi4-4GB I have (as well as the Pi4-8GB).
I've used several SDCards, a 4GB, and 3 8 GB.
I get the can't seem to see the 2nd USB Camera when it's a same
manufacturer (sometimes sees 1, sometime sees the other) when both are
plugged into the black USB ports (USB2)
(I get the issue on a PI2, too, using 20200606)
I am at a complete loss.
I you wanted too, you could try installing the RPiOS and installing
motionEye following these instructions:
https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian
or install the Ubuntu 20.10 for RaspberryPi (64 bit desktop) and use my
instructions
*Part 1: (after imaging)*
sudo apt update && sudo apt upgrade -y
sudo apt install ssh curl
sudo reboot

*Part 2:*
connect via ssh
sudo apt update
sudo apt install python2
curl https://bootstrap.pypa.io/get-pip.py --output get-pip.py
sudo python2 get-pip.py

*Part 3: (I'll need to update these instructions, too much of Python2.7 is
not in      the upstream repositories any more, especially for the
ARM-Ubuntu...)*

sudo su
apt install -y libffi-dev libzbar-dev libzbar0
apt-get install motion ffmpeg v4l-utils
apt-get install python2-dev libssl-dev libcurl4-openssl-dev libjpeg-dev
apt-get install python-pil
pip2 install motioneye
mkdir -p /etc/motioneye
cp /usr/local/share/motioneye/extra/motioneye.conf.sample
/etc/motioneye/motioneye.conf
mkdir -p /var/lib/motioneye
cp /usr/local/share/motioneye/extra/motioneye.systemd-unit-local
/etc/systemd/system/motioneye.service
systemctl daemon-reload
systemctl enable motioneye
systemctl start motioneye

Now you should be able to connect to http://<ip_address>:8765 from another
computer on the same network, or if using Desktop Ubuntu in FireFox
http://localhost:8765

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

STUKguy, 2020-10-25:
@starbasessd ive just reflashed 20200606 and booted this im going to try plugging in 1 camera at a time and reboot after each one but if this fails again i will try both the suggestions. Really unsure why this is happening.

STUKguy, 2020-10-25:
@starbasessd i only have a PI4 2GB will either of these options work on it?

starbasessd, 2020-10-25:
Yes, they work on the Pi4 1GB, too, as well as the 4GB & 8GB versions. The
RaspbianOS version works on the 3B and 3B+, and later today, or tomorrow,
will try the Ubuntu setup on the 3B & 3B+.
I like to have options, and I have a number of the earlier PIs and a few
Zeros, ZeroWs, ZeroWHs & 3B & 3B+'s, and the pair of 4Bs...
That's how I can re-create a lot of user's enviroments

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

STUKguy, 2020-10-25:
@starbasessd ok so moved on to the Guthub instructions ive dones everything it said step-by-step but after the last step it doesnt say where to go to veiw motion eye??? you dont happen to know the port number i need to type do you?

starbasessd, 2020-10-25:
8765 is the admin/webgui port.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

STUKguy, 2020-10-25:
@starbasessd thanks, logged in now
[image]
this could take awhile trying to figure out which one is the feed and i only have one of the camera plugged in

starbasessd, 2020-10-25:
It's going to be one of the bottom 4...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

STUKguy, 2020-10-25:
@starbasessd ok so i have managed to get the Github instructions working with both cameras
[image]
see how this goes, i think it was roughly 1 and half weeks before it crashed so i will report back in 2 weeks or sooner if it crashes again.

starbasessd, 2020-10-25:
OK, no problem.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

STUKguy, 2020-10-25:
@starbasessd just found out that this install is not storing video files, there are no logs so I can't see why it's not storing video files

starbasessd, 2020-10-25:
On RaspberryPiOS or Ubuntu 20.10? What is the path for File Storage?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

STUKguy, 2020-10-25:
@starbasessd Raspbian OS and the path is set to custom /cctv/landing

starbasessd, 2020-10-25:
Does that path exist?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

STUKguy, 2020-10-25:
@starbasessd yep just checked it is there

starbasessd, 2020-10-26:
Going to build an image later today, had to free up the Pi4 4GB.
Will advise.
May just be a permissions issue to that folder tree.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-10-26:
what is under the path:
/var/lib/motioneye/
???

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

STUKguy, 2020-10-26:
@starbasessd 
pi@raspberrypi:/var/lib/motioneye $ ls
Camera1  Camera2  Camera3  Camera4  Camera5  Camera6  Camera7  Front  Landing

starbasessd, 2020-10-26:
I create the /cctv/landing path on my
Pi4 5GB, 128GB, with RaspberryPiOS 20200820, full update & upgrade, and the
commands to install motion/motioneye from here:
https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian
Added 2 cameras
Turned on motion detection on both
recorded correctly to /var/lib/motioneye/Camera1 and Camera2
changed File Storage, Custom Path to /cctv/landing in both
recorded correctly there, too.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

[59 more replies]

closed 2020-11-09 by STUKguy