motioneye-project/motioneyeos #2555: brcmfmac problems on install raspberry zero W

erichuib, 2020-10-26:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: latest version: 20200606.

### Board Model

I am using the following board/model: Raspberry Zero w
### Camera

I am using the following type of camera: (choose from V4L2, MMAL, Network Camera, Fast Network Camera and Simple MJPEG Camera).

My camera model is: rasp V2.

### Network Connection

My motionEyeOS unit is connected to the network via: WiFi
### Peripherals

I am using the following peripherals that I consider relevant to this issue:

intern wifi 

### Log Files

I consider the following log files relevant to this issue:

brcmfmac foul!!

starbasessd, 2020-10-26:
Just to be sure, you downloaded and unzipped:
https://github.com/ccrisan/motioneyeos/releases/download/20200606/motioneyeos-raspberrypi-20200606.img.xz
and burned the image to a microSDCard of 4GB or larger?
Where do you see the error? During boot up? After bootup?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-10-26:
BTW, you need to add a wpa_supplicant.conf file to the /boot folder/drive.
The wpa_supplicant.conf file format is at:
https://github.com/ccrisan/thingos/wiki/wpa_supplicant.conf
or an os.conf file set to OS_NETWORKLESS=false
to temporarily disable the network search.

erichuib, 2020-10-26:
Yes downloaded the latest file from: https://github.com/ccrisan/motioneyeos/wiki/Supported-Devices
doing all the same as usual when installing motion eye.

I have copied a working wpa_supplicant.conf from my files ( use this one on al my raspberry's

when I hook up my screen I see that the install goes normal until the network search. it crashes and I get : brcmfmac fail.

and a panic reboot begins.

over and over again.

starbasessd, 2020-10-26:
> motioneyeos/wiki/Supported-Devices

Try with 20200203. 20200606 works on all my PiZeros except the non-wireless
ones (no W in the name on the back)

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-10-26:
If you can get it to boot (with RaspberryPiOS or similar, you can use the
command
sudo cat /sys/firmware/devicetree/base/model;echo [enter]
one of my PiZeroWs shows:
[root@meye-c0e188a9 ~]# cat /sys/firmware/devicetree/base/model;echo
Raspberry Pi Zero W Rev 1.1

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

erhughesjr, 2020-10-27:
Computer:  Pi Zero W (brand new)
SD: Samsung Pro Endurance
Motion Eye OS Img:  20200606

I've been trying to get this to run on my new Pi Zero W for a day now -- no luck.   I have tried using another raspberry pi 4 to create the SD using Etcher, and with my Windows 10 computer.    My card is a 32 GB Samsung Pro Endurance.   I get 3 volumes -- 31 MB volume (which has the bootcode.bin, config.txt, start.elf,...), a 31 GB volume, and a 231 MB volume (with bin, boot, data, etc directories).  I am using version 20200606.   

I am trying to connect to a 2.4 GB wifi router, which is the same one I'm using on my Raspberry Pi 4 and on my WIndows 10 computer.    Note that I have no trouble with using this same Pi Zero W with the latest img of Raspberry OS connecting to the same router.   I just can't get it to work with the motioneyeos.

I have placed the wpa_supplicant.conf file (as shown in wifi preconfiguration with the correct SSID and password) in the 31 MB volume, which appears to be the boot volume/partition.   Raspberry OS will not allow me to put it anywhere else -- it gives me permission errors.    On my Windows 10 computer, I can only see the 31 MB volume, and had placed the file in there.    

The Error:
When it comes to bootup in my Zero W, the message "Starting wpa_supplicant:" comes up.    It then states "no device".    Next it tries to configure the wireless network and says no device.   Then the device begins the perpetual panic reboot process.    

The SSID and password are correct -- i'm using the same ones to connect to the internet and post this message on my other Raspberry Pi.  "scan_SSID, ssid, and psk" all start in column 5 of the file and the values are surrounded by quotation marks.  When I re-insert the SD back in my RPI 4, the wpa_supplicant.conf is gone.   

I've reformatted the SD numerous times and then re-imaged the SD with motioneyeos.    I've tried creating the file with and without the ctrl_interface line.   I've tried copying the wpa_supplicant.conf file back into the 31 MB volume after it was removed the first time.  I've tried following steps of a number of YouTube channels reviewing the product and their steps don't work.   I've tried to overwrite the /data/etc/wpa_supplicant.conf file, but the /data file is on the 231 MB volume and has no /etc subdirectory.  I found a /etc directory, but I can't copy anything to it.    I have a /boot directory in the 231 MB volume, but Raspberry OS will not permit me to copy or create the file in that directory.   I found my wpa_supplicant.conf file does appear in the 31 GB volume under the /etc directory.   So the wpa file is being seen in the 31 MB volue and appears to be moved or copied to that 31 GB/etc directory before being deleted from the 31 MB volume.  

It should not be this hard, but I've run out of things to try.   Any suggestions?

starbasessd, 2020-10-27:
Try with imaging the SDCard with RaspberryPiOS, and put the
wpa_supplicant.conf file in that first (only) partition that can be seen in
Windows 10.
That will allow you to boot up, and if the wpa_supplicant.conf file fails,
it'll still boot up all the way.
Can you take a pic of the back of the PiZero, the side without the chips,
and post here?
It is also possible that the PiZero is defective.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

erhughesjr, 2020-10-27:
Excellent advice.    I tried what you suggested.   I installed Raspberry Pi OS and inserted the wpa_supplicant.config file into the first partition with my Windows 10 machine.   I inserted it into the Pi Zero and it came right up.   However, I noticed there was no wifi.    So I compared the back of two Pi Zero's I had and they were different.    I've attached these.   It appears the one I was trying to use first was not a -W.    So I reburned the SD with motioneyeos and added the wpa_supplicant.config to the first partition.    I put this in the 2nd Pi Zero -- the one with a W in the name on the back.    It booted up and got a wifi address right away.   I attached the picture when I networked in from my Windows 10 computer.    I'm a little embarrassed -- I ordered 2 Pi Zero Ws from two different vendors.   One vendor sent me the wrong device and I didn't the board itself.    So thanks for that suggestion of looking at the back of the board. 

Thanks very much for your help!
[image]
[image]
[image]

starbasessd, 2020-10-27:
Yes, it happens. Not all PiZero's are created the same. Some dealers are not quite as honest, or don't know what they are selling...
Glad you got it straightened out.

closed 2020-10-27 by starbasessd