motioneye-project/motioneyeos #2557: Holds no media

mroberts77, 2020-10-27:
Camera been running for a couple of weeks, and apparently say no media files on it? Tried to look on the disk on computer, couldn't find any folders to get the images
Indicates it's 50% full, so something must be there? How to get the media ? before I scrub and start again?

All I could find in dmsg log is 👍 [  393.748866] EXT4-fs error (device mmcblk0p3): dx_probe:759: inode #655391: comm python: Directory index failed checksum

[image]

starbasessd, 2020-10-27:
Would help to have answers to the standard questions:
Make/Model of Board
Version of motionEyeOS (20190911, 20200203, 20200606, a dev version?)
For your specific issue, the File Storage path
Logs: motion, motioneye, boot

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

mroberts77, 2020-10-27:
gotcha!

motionEyeOS 20200606

[motioneye.log.txt](https://github.com/ccrisan/motioneyeos/files/5446489/motioneye.log.txt)
[motion.log.txt](https://github.com/ccrisan/motioneyeos/files/5446490/motion.log.txt)
[messages.log.txt](https://github.com/ccrisan/motioneyeos/files/5446496/messages.log.txt)
[boot.log.txt](https://github.com/ccrisan/motioneyeos/files/5446498/boot.log.txt)
[dmesg.log.txt](https://github.com/ccrisan/motioneyeos/files/5446500/dmesg.log.txt)

starbasessd, 2020-10-27:
what board?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-10-27:
If a Raspberry Pi, can you ssh to it, or hook up a monitor and keyboard?
If so, we can confirm issues with the /data partition (the 3rd one) where
the videos are stored by default.
If not, the fastest solution is to re-image, as it appears from the logs
that the /data partition is corrupt.
If you have a Linux machine (laptop or PC) that you can insert the SDCard,
we might be able to check it there, and / or possibly recover data.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

mroberts77, 2020-10-27:
From Adafruit: [https://www.adafruit.com/product/3579](url)
Pi Zero W.
can ssh into it with putty but not having much luck finding the files

mroberts77, 2020-10-27:
Are the media files salvagable? or gone forever? Have reimaged 2x: the first was very spotty. THis go works sometimes, camera not on, and was successfully doing the recording work and saving files. didn't think the files were in jeapordy (lesson!).

starbasessd, 2020-10-27:
ssh in
type the following commands:
mount -o remount rw /boot [enter]
nano /boot/cmdline.txt [enter]
at the end of the line, type in
fsck.repair=yes
[ctrl]x y [enter]
reboot [enter]
then watch the boot up, or check /var/log/boot.log
let me know.
(as to whether they're salvageable depends on how corrupt the SDCard
mmcblk0p3 partition is damaged.)

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

mroberts77, 2020-10-27:
Still doesn't seem to have any media files :( 
boot.log: 
---- shutting down motionEyeOS 20200606 ----
 * Stopping motioneye:                                                                                                                                                                                                                                                                                                                           ---- booting motionEyeOS 20200606 ----
 * Detecting disk device: /dev/mmcblk0
 * Mounting filesystems: done
 * Mounting overlay filesystems: done
 * Loading kernel modules: done
 * Setting hostname: done
 * Configuring CPU frequency: done
 * Starting syslogd: done
 * Loading device-tree overlays: done
 * Starting throttle watcher: done
 * Starting eudev: done
 * Starting watchdog: done
 * Starting rngd: done
 * Preparing AP interfaces: ap0
 * Starting wpa_supplicant: done (brcmfmac)
 * Configuring wireless network: dhcp
 * Setting current date using SNTP: Tue Oct 27 11:58:03 PDT 2020
 * Starting ntpd: done
 * Starting crond: done
 * Starting sshd: done
 * Starting proftpd: done
 * Starting smbd: done
 * Starting nmbd: done
 * Starting motioneye: done
 # Interface wlan0 has IP address 192.168.0.27/24
 # Default gateway is 192.168.0.1
 # DNS server address is 192.168.0.1

mroberts77, 2020-10-27:
Doesn't seem to take any images, so must be that the disk won't allow any more images? 
would be good to get what's on there, since it still appears to be 50% full?

starbasessd, 2020-10-27:
Lets try:
ssh in
type the following commands:
mount -o remount rw / [enter]
touch /forcefsck; reboot [enter]

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-10-27:
ssh in
run command:
df -h [enter]
list results here.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

mroberts77, 2020-10-27:
[root@meye-955e5835 ~]# mount -o remount rw/
mount: can't find rw/ in /proc/mounts
[root@meye-955e5835 ~]# mount -o remount rw
mount: can't find rw in /proc/mounts

then:
[root@meye-955e5835 ~]# df -h
Filesystem                Size      Used Available Use% Mounted on
/dev/root               213.0M    180.4M     17.6M  91% /
devtmpfs                180.3M         0    180.3M   0% /dev
tmpfs                   184.5M     16.0K    184.5M   0% /dev/shm
tmpfs                   184.5M      2.3M    182.2M   1% /tmp
/dev/mmcblk0p1           29.9M     13.3M     16.6M  44% /boot
/dev/mmcblk0p3           13.2G      6.1G      6.4G  49% /data
/dev/mmcblk0p3           13.2G      6.1G      6.4G  49% /home/ftp/sdcard
/dev/mmcblk0p3           13.2G      6.1G      6.4G  49% /home/ftp/storage
overlay                  13.2G      6.1G      6.4G  49% /usr
overlay                  13.2G      6.1G      6.4G  49% /var/log
overlay                  13.2G      6.1G      6.4G  49% /var/lib

starbasessd, 2020-10-27:
There is a space between rw and the / character.
type the following commands:
mount -o remount rw / [enter]
touch /forcefsck; reboot [enter]

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

mroberts77, 2020-10-27:
[root@meye-955e5835 ~]# df -h
Filesystem                Size      Used Available Use% Mounted on
/dev/root               213.0M    180.4M     17.6M  91% /
devtmpfs                180.3M         0    180.3M   0% /dev
tmpfs                   184.5M     16.0K    184.5M   0% /dev/shm
tmpfs                   184.5M      2.3M    182.2M   1% /tmp
/dev/mmcblk0p1           29.9M     13.3M     16.6M  44% /boot
/dev/mmcblk0p3           13.2G      6.1G      6.4G  49% /data
/dev/mmcblk0p3           13.2G      6.1G      6.4G  49% /home/ftp/sdcard
/dev/mmcblk0p3           13.2G      6.1G      6.4G  49% /home/ftp/storage
overlay                  13.2G      6.1G      6.4G  49% /usr
overlay                  13.2G      6.1G      6.4G  49% /var/log
overlay                  13.2G      6.1G      6.4G  49% /var/lib

starbasessd, 2020-10-27:
still need the boot.log

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

mroberts77, 2020-10-27:
:)
---- booting motionEyeOS 20200606 ----
 * Detecting disk device: /dev/mmcblk0
 * Mounting filesystems: done
 * Mounting overlay filesystems: done
 * Loading kernel modules: done
 * Setting hostname: done
 * Configuring CPU frequency: done
 * Starting syslogd: done
 * Loading device-tree overlays: done
 * Starting throttle watcher: done
 * Starting eudev: done
 * Starting watchdog: done
 * Starting rngd: done
 * Preparing AP interfaces: ap0
 * Starting wpa_supplicant: done (brcmfmac)
 * Configuring wireless network: dhcp
 * Setting current date using SNTP: Tue Oct 27 12:19:49 PDT 2020
 * Starting ntpd: done
 * Starting crond: done
 * Starting sshd: done
 * Starting proftpd: done
 * Starting smbd: done
 * Starting nmbd: done
 * Starting motioneye: done
 # Interface wlan0 has IP address 192.168.0.27/24
 # Default gateway is 192.168.0.1
 # DNS server address is 192.168.0.1

starbasessd, 2020-10-27:
If this is the latest, it's not doing the file check. Unless you have a
Linux machine, you won't be able to recover anything, and you need to
re-image.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

mroberts77, 2020-10-27:
how do I access that folder named /home/ftp/sdcard? seems there's file size in there?

Filesystem                Size      Used Available Use% Mounted on
/dev/root               213.0M    180.4M     17.6M  91% /

---- booting motionEyeOS 20200606 ----
 * Detecting disk device: /dev/mmcblk0
 * Mounting filesystems: done
 * Mounting overlay filesystems: done
 * Loading kernel modules: done
 * Setting hostname: done
 * Configuring CPU frequency: done
 * Starting syslogd: done
 * Loading device-tree overlays: done
 * Starting throttle watcher: done
 * Starting eudev: done
 * Starting watchdog: done
 * Starting rngd: done
 * Preparing AP interfaces: ap0
 * Starting wpa_supplicant: done (brcmfmac)
 * Configuring wireless network: dhcp
 * Setting current date using SNTP: Tue Oct 27 12:37:59 PDT 2020
 * Starting ntpd: done
 * Starting crond: done
 * Starting sshd: done
 * Starting proftpd: done
 * Starting smbd: done
 * Starting nmbd: done
 * Starting motioneye: done
 # Interface wlan0 has IP address 192.168.0.27/24
 # Default gateway is 192.168.0.1
 # DNS server address is 192.168.0.1
devtmpfs                180.3M         0    180.3M   0% /dev
tmpfs                   184.5M     16.0K    184.5M   0% /dev/shm
tmpfs                   184.5M      2.3M    182.2M   1% /tmp
/dev/mmcblk0p1           29.9M     13.3M     16.6M  44% /boot
/dev/mmcblk0p3           13.2G      6.1G      6.4G  49% /data
/dev/mmcblk0p3           13.2G      6.1G      6.4G  49% /home/ftp/sdcard
/dev/mmcblk0p3           13.2G      6.1G      6.4G  49% /home/ftp/storage
overlay                  13.2G      6.1G      6.4G  49% /usr
overlay                  13.2G      6.1G      6.4G  49% /var/log
overlay                  13.2G      6.1G      6.4G  49% /var/lib

starbasessd, 2020-10-27:
It's also /data and /home/ftp/storage, they are all 3 connections to the
same space.
you would enter:
cd /home/ftp/sdcard [enter]
or
cd /home/ftp/storage [enter]
then
ls -la [enter]

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

mroberts77, 2020-10-27:
not sure if i can do much: does it seem like they are gonza?

[root@meye-955e5835 sdcard]# ls -la

total 4625
drwxr-xr-x    3 root     root          4096 Oct 17 09:13 ./
drwxr-xr-x    4 ftp      ftp           1024 Jun  6 02:54 ../
-rw-rw-rw-    1 root     root       4724346 Sep 13 20:03 .1600052354.avi
-rw-r--r--    1 root     root             0 Jun  6 02:54 .keep
drwxrwxrwx   10 root     root          4096 Oct  5 14:37 Camera1/

starbasessd, 2020-10-27:
How about
ls -la Camera1 [enter]

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

mroberts77, 2020-10-27:
[root@meye-955e5835 sdcard]# ls -la Camera1
total 48
drwxrwxrwx   10 root     root          4096 Oct  5 14:37 ./
drwxr-xr-x    3 root     root          4096 Oct 17 09:13 ../
-rw-rw-rw-    1 root     root             0 Aug  1 22:19 .keep
drwxr-xr-x    2 root     root          4096 Jul 30 23:54 2020-07-30/
drwxr-xr-x    2 root     root         12288 Jul 31 23:54 2020-07-31/
drwxr-xr-x    2 root     root          4096 Aug  1 22:18 2020-08-01/
drwxrwxrwx   16 root     root          4096 Oct 27 14:00 FrontTL/
drwxrwxrwx    4 root     root          4096 Aug  5 20:48 aug2020/
drwxrwxrwx    3 root     root          4096 Sep 11 22:02 backyard/
drwxrwxrwx    2 root     root          4096 Oct  5 14:37 flower/
lrwxrwxrwx    1 root     root            23 Aug  1 22:18 lastsnap.jpg -> 2020-08-01/22-18-00.jpg
drwxrwxrwx    4 root     root          4096 Aug  2 18:09 moon/

starbasessd, 2020-10-27:
ls -laR Camera1 [enter]

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

mroberts77, 2020-10-28:
lots of files:

-rw-rw-rw-    1 root     root        109898 Aug  2 08:12 08-12-00.jpg
-rw-rw-rw-    1 root     root        110202 Aug  2 08:18 08-18-00.jpg
-rw-rw-rw-    1 root     root        118362 Aug  2 08:24 08-24-00.jpg
-rw-rw-rw-    1 root     root        123105 Aug  2 08:30 08-30-00.jpg
-rw-rw-rw-    1 root     root        126117 Aug  2 08:36 08-36-00.jpg
-rw-rw-rw-    1 root     root        134208 Aug  2 08:42 08-42-00.jpg
-rw-rw-rw-    1 root     root        135696 Aug  2 08:48 08-48-00.jpg
-rw-rw-rw-    1 root     root        139173 Aug  2 08:54 08-54-00.jpg
-rw-rw-rw-    1 root     root        143310 Aug  2 09:00 09-00-00.jpg
-rw-rw-rw-    1 root     root        141068 Aug  2 09:06 09-06-00.jpg
-rw-rw-rw-    1 root     root        139942 Aug  2 09:12 09-12-00.jpg
-rw-rw-rw-    1 root     root        141880 Aug  2 09:18 09-18-00.jpg
-rw-rw-rw-    1 root     root        141601 Aug  2 09:24 09-24-00.jpg
-rw-rw-rw-    1 root     root        145098 Aug  2 09:30 09-30-00.jpg
-rw-rw-rw-    1 root     root        134286 Aug  2 09:36 09-36-00.jpg
-rw-rw-rw-    1 root     root        141002 Aug  2 09:42 09-42-00.jpg
-rw-rw-rw-    1 root     root        136747 Aug  2 09:48 09-48-00.jpg

mroberts77, 2020-10-28:
folder of interest:

Camera1/FrontTL/2020-10-26:
total 38328
drwxr-xr-x    2 root     root          4096 Oct 26 18:47 ./
drwxrwxrwx   16 root     root          4096 Oct 27 18:00 ../
-rw-rw-rw-    1 root     root        922073 Oct 26 00:00 00-00-00.jpg
-rw-rw-rw-    1 root     root        886420 Oct 26 01:00 01-00-00.jpg
-rw-rw-rw-    1 root     root        872344 Oct 26 02:00 02-00-00.jpg
-rw-rw-rw-    1 root     root        857489 Oct 26 03:00 03-00-00.jpg

mroberts77, 2020-10-28:
does this mean that the images are still there, salvageable?

starbasessd, 2020-10-28:
Maybe..
Are you using Windows 10?
If yes, in your search box, bottom left, type in "bash" (without the
quotes) & [enter]
type
cd /mnt/c/User/(your Windows user name)/Downloads [enter]
scp -r admin@(ip address of  pi):/home/ftp/storage ./ [enter]
Answer yes  to Are you sure you want to continue connecting
You should see it all download.
Let me know.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-10-28:
Oops, typo... Users, not User
type
cd /mnt/c/Users/(your Windows user name)/Downloads [enter]
scp -r admin@(ip address of  pi):/home/ftp/storage ./ [enter]
Answer yes  to Are you sure you want to continue connecting
You should see it all download.
Let me know.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

mroberts77, 2020-10-28:
Yes: Windows 10. search for bash only indicates a bash.sh file. SHould the MotionOS be plugged in for Bash to work? Turned it off after it wasn't useful. Looking for a dos prompt or specific one?
 
Have the anaconda dos prompt open, will that work in lieu of Bash?

not balena etcher?

starbasessd, 2020-10-28:
My Windows 10 has the Windows Subsystem for Linux installed, I was hoping
yours did, too.
Where/how do you have PuTTY installed?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

mroberts77, 2020-10-28:
on lap top, start - putty - IP of pi

starbasessd, 2020-10-28:
Do you have pscp as an option under putty apps?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

mroberts77, 2020-10-28:
[image]

starbasessd, 2020-10-28:
Would you go to http://www.putty.org, and click Download, and grab the
pscp.exe program, and place it in C:\Program Files\PuTTY\, please?
Let me know when done or if you don't feel comfortable about it.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-10-28:
It may already be in c:\Program Files\Putty or C:\Program Files (x86)
Putty, by the way...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

mroberts77, 2020-10-29:
Looks like the folder already has an exe file: double click didn't do anything
[image]

starbasessd, 2020-10-29:
OK, have the Pi on and know the ip_address
open a command line on the Windows box (search 'cmd' and click on it)
in the box that opens:
cd c:\Users\your user name\ [enter]
c:\Progra~1\putty\pscp -r -P 22 admin@ip_address_of Pi:/home/ftp/storage/*
Downloads\ [enter]
This should copy anything available to your Windows Download Folder

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

mroberts77, 2020-10-29:
[image]

mroberts77, 2020-10-29:
did I type it wrong (Likely): Feel like a real amateur here.

starbasessd, 2020-10-29:
Yep.
there's a space between storage/* and Downloads\ you missed.
The rest looks good.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

mroberts77, 2020-10-29:
C:\Users\micha>c:\Progra~1\putty\pscp -r -P 22 admin@192.168.0.27:/home/ftp/storage/* Downloads\
admin@192.168.0.27's password:
pscp: wildcard '*' matched no files

starbasessd, 2020-10-29:
Try
C:\Users\micha>c:\Progra~1\putty\pscp -r -P 22 admin@192.168.0.27:/home/ftp/*
Downloads\
remove storage/ from the line.
You missed the last \ after Downloads, that time...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

mroberts77, 2020-10-29:
that command is getting some juice flowing... need to check if it gets the recent stuff.... downloading.... ;0

starbasessd, 2020-10-29:
Fingers crossed...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-10-29:
Only had to jump through a few hoops.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

mroberts77, 2020-10-29:
score man! looks like everything is there! Seems like voodu, but that last command worked! I am going to remove the sd card, put a new one in and start again fresh. ONly thing I'd say is that there are times when I have to physically restart it and times I can log into the web version, but no camera. Maybe this 3rd go will be better. But thank you for hanging in there and supporting me thru this, and being responsive. :)

closed 2020-10-29 by mroberts77