motioneye-project/motioneyeos #2558: cannot access admin page on wifi, how can i set a static ip?

NielsKoomans97, 2020-10-28:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 20200606

### Board Model

I am using the following board/model: Raspberry PI 3B+

### Camera

I am using the following type of camera: V4L2

My camera model is: Logitech C270

### Network Connection

My motionEyeOS unit is connected to the network via: Wifi

### Peripherals

I am using the following peripherals that I consider relevant to this issue: none

 * (e.g. USB WiFi adapter)
 * (e.g. External Hard Disk)

### Log Files

I consider the following log files relevant to this issue: can't generate since i can't access the os via SSH or video output

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)
 
```
C:\Users\niels>ping 192.168.2.4

Pinging 192.168.2.4 with 32 bytes of data:
Reply from 192.168.2.48: Destination host unreachable.
Reply from 192.168.2.48: Destination host unreachable.
Reply from 192.168.2.48: Destination host unreachable.
Reply from 192.168.2.48: Destination host unreachable.

Ping statistics for 192.168.2.4:
    Packets: Sent = 4, Received = 4, Lost = 0 (0% loss),
```

this is the output from command prompt when I try to ping the motioneye os server. 2.48 is my mini desktop pc, and 2.4 is my raspberry pi. Somehow the requests get routed hrough my desktop pc, and I honestly don't know how to fix this since i can't access motion through ssh or the normal admin page. i flushed the dns cache on my desktop, which took the request timed out error away, but gave me the destination host unreachable error in return. 

that's how i came to the idea of giving my pi a static ip on it's wifi, but doing that is a bit hard considering i can't access the terminal, at all.

is there a (hacky) solution to this?

starbasessd, 2020-10-28:
Yes, several.
To set a static ip, you can include the following files, static_ip.conf, &
wpa_supplicant.conf, after you image the SDCard, into the boot folder.
Content of static_ip.conf:
=====
STATIC_IP=192.168.2.4
STATIC_GW=192.168.2.1
STATIC_DNS=192.168.2.1
=====
Content of wpa_supplicant.conf
=====
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
country=US

network={
    ssid="your ssid"
    psk="your wifi password"
key_mgmt=WPA-PSK
}
=====
Change country to your country if not US.

Some instructions on how to include the files into the /boot if needed:
https://github.com/ccrisan/motioneyeos/wiki/Wifi-Preconfiguration

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

NielsKoomans97, 2020-10-29:
i finally got it working. thanks a lot for the help =)

closed 2020-10-29 by NielsKoomans97