motioneye-project/motioneyeos #2560: MotionEyeOS multiple video* on RPi ZW and no video in MEyeOS latest Oct 2020

mrmichaelhallett, 2020-10-28:
Using username "admin".
Pre-authentication banner message from server:
| Welcome to RaspiCam!
End of banner message from server
admin@192.168.50.113's password:
[root@RaspiCam ~]# ls /dev/video*
/dev/video10  /dev/video11  /dev/video12

No display of camera video in console
Logs in fine and I see Logi Camera though no display because I unplugged it for troubleshooting
Raspi camera connected directly to the Raspberry Pi Zero W camera connector

Wireless connection: WPA_Supplicant.conf worked fine for initial wireless connection

About to re image SD card and start again.
OTG plus Logi QCam Orbit worked fine.  Unplugged

Michael

Using username "admin".
Pre-authentication banner message from server:
| Welcome to RaspiCam!
End of banner message from server
admin@192.168.50.113's password:
[root@RaspiCam ~]# dmesg
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.19.83 (root@f8defca1b6bb) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611)) #1 Sat Jun 6 14:23:22 UTC 2020
[    0.000000] CPU: ARMv6-compatible processor [410fb767] revision 7 (ARMv7), cr=00c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT nonaliasing instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi Zero W Rev 1.1
[    0.000000] Memory policy: Data cache writeback
[    0.000000] cma: Reserved 8 MiB at 0x17400000
[    0.000000] On node 0 totalpages: 98304
[    0.000000]   Normal zone: 864 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 98304 pages, LIFO batch:31
[    0.000000] random: get_random_bytes called from start_kernel+0x8c/0x49c with crng_init=0
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 97440
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708_fb.fbswap=1 smsc95xx.macaddr=B8:27:EB:5B:D9:2C vc_mem.mem_base=0x1ec00000 vc_mem.mem_size=0x20000000  dwc_otg.fiq_fix_enable=1 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 rootwait ro rootflags=noload panic=10 quiet loglevel=1
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Memory: 369180K/393216K available (7157K kernel code, 657K rwdata, 2136K rodata, 468K init, 797K bss, 15844K reserved, 8192K cma-reserved)
[    0.000000] Virtual kernel memory layout:
                   vector  : 0xffff0000 - 0xffff1000   (   4 kB)
                   fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
                   vmalloc : 0xd8800000 - 0xff800000   ( 624 MB)
                   lowmem  : 0xc0000000 - 0xd8000000   ( 384 MB)
                   modules : 0xbf000000 - 0xc0000000   (  16 MB)
                     .text : 0x(ptrval) - 0x(ptrval)   (7158 kB)
                     .init : 0x(ptrval) - 0x(ptrval)   ( 468 kB)
                     .data : 0x(ptrval) - 0x(ptrval)   ( 658 kB)
                      .bss : 0x(ptrval) - 0x(ptrval)   ( 798 kB)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] ftrace: allocating 25729 entries in 76 pages
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000030] sched_clock: 32 bits at 1000kHz, resolution 1000ns, wraps every 2147483647500ns
[    0.000078] clocksource: timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 1911260446275 ns
[    0.000200] bcm2835: system timer (irq = 27)
[    0.000605] Console: colour dummy device 80x30
[    0.000634] console [tty1] enabled
[    0.000694] Calibrating delay loop... 697.95 BogoMIPS (lpj=3489792)
[    0.060339] pid_max: default: 32768 minimum: 301
[    0.060889] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.060912] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.062334] CPU: Testing write buffer coherency: ok
[    0.063653] Setting up static identity map for 0x8200 - 0x8238
[    0.065060] devtmpfs: initialized
[    0.076375] VFP support v0.3: implementor 41 architecture 1 part 20 variant b rev 5
[... 196 lines cut ...]
I am using the following board/model: (insert your board model here, e.g. Raspberry PI 3B+).

### Camera

I am using the following type of camera: (choose from V4L2, MMAL, Network Camera, Fast Network Camera and Simple MJPEG Camera).

My camera model is: (insert your camera model, e.g. Logitech C290).

### Network Connection

My motionEyeOS unit is connected to the network via: (choose your network connection, e.g. WiFi, Ethernet).

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * (e.g. USB WiFi adapter)
 * (e.g. External Hard Disk)

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)

starbasessd, 2020-10-28:
MotionEyeOS has no camera outputs or video display in Console or SSH
session. (one exception, will describe later)
When you get it installed, you then open a browser like Edge, Chrome,
Firefox, Opera, etc on another computer, and connect to http://ipaddrress
That is where you administer ,ost of the settings, and view cameras set up.
The one exception is when you set "Fast Network Camera = ON", you can also
turn on a setting for HDMI Output, but it is only for a CSI Pi Cam.
You lose all console access, but SSH and the WebGUI access still works.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

mrmichaelhallett, 2020-10-29:
Not sure how to close this out.
Got the RaspiCam and logitech USB cam working.
Had to do a manual modprobe and (have the cables attached to the RPi zero W camera connector and camera board (duh!)
Then modprobe, then found Video 0 eventually after a couple of reboots 

For the Logitech QuickCam Orbit I found 3 video devices automagically created Video 10,11,12 none worked
Kept installing cameras with the defaults  then the last one chose Video ) and worked (halleleuya)

Then I read about how to mount as rw and edit the config (didnt do it)
but saw that some had edited one of the camera conf (ig) files and changed the device.
Rebooted after replugging the USB and found device Video 1 had been created
oh a whim tried changing the Camera1.conf video device to Video1 instead of Video 10 (edited with nano and saved)
reboot and both are showing

I will probably reflash and set them up again as an exercise and to get them as Camera 1 and 2

In case it helps anyone. 
Michael

closed 2020-10-29 by mrmichaelhallett