motioneye-project/motioneyeos #2561: Inconsistant camera service, drops, interface still there, no camera

mroberts77, 2020-10-29:
Started fresh image.
Started up, applied settings.
Took one image then camera wasn't available next day, appears last image taken was at midnight.
Using a different sd card, had to reboot today to get camera back.
Same issue was having before: inconsistant reliability and thought it was a power issue (Tried solar, NG), so plugged in now. Is the board getting too hot, is the card problematic? What could be the issue that is  the camera unreliable? ANy indication from the logs?
If undeterminable, will give up and accept inperfection, but if it's something I can solve to avoid with a setting or such, would be good to know to improve reliability.... :)

[motion.log-1.txt](https://github.com/ccrisan/motioneyeos/files/5460094/motion.log-1.txt)
[messages.log-1.txt](https://github.com/ccrisan/motioneyeos/files/5460096/messages.log-1.txt)
[boot.log-4.txt](https://github.com/ccrisan/motioneyeos/files/5460097/boot.log-4.txt)
[dmesg.log-1.txt](https://github.com/ccrisan/motioneyeos/files/5460099/dmesg.log-1.txt)

### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version 

motionEye Version | 0.42.1
-- | --
Motion Version | 4.2.2+gitUNKNOWN

I am running motionEyeOS version: (insert your version here, e.g. 20180314).
OS Version | motionEyeOS 20200606
### Board Model

I am using the following board/model: (insert your board model here, e.g. Raspberry PI 3B+).

Pi Zero W from Pimori

### Camera

I am using the following type of camera: (choose from V4L2, MMAL, Network Camera, Fast Network Camera and Simple MJPEG Camera).
MMAL Camera that came with it from Pimori [https://www.adafruit.com/product/3579]() 

My camera model is: (insert your camera model, e.g. Logitech C290).

### Network Connection

My motionEyeOS unit is connected to the network via: (choose your network connection, e.g. WiFi, Ethernet).

Home WIfi

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * (e.g. USB WiFi adapter)
 * (e.g. External Hard Disk)

NONE (SD card)

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)

starbasessd, 2020-10-29:
It's not heat. Temps avg 40's ( peaks were around low 50s) Pretty good.
When you say 'unreliable', what do you mean? Not detecting movement?
That's a setting to play with under "Motion Detection"
Be sure to turn on Show Motion Changes while setting up to see where the
software thinks there's movement.
I have to set mine to .1% for Frame Change Threshhold for one of my cameras.
I also have Motion Gap set to 2
Captured before 10
Captured After 10
and Minimum Motion Frames 2
I turn on Depeckle Filter to help with camera noise.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

mroberts77, 2020-10-29:
Keeps loosing camera functionality.

starbasessd, 2020-10-29:
Loose or bad cable?
If you can ssh in, try the command:
vcgencmd get_throttled [enter]
right after you lose the camera functionality.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

mroberts77, 2020-10-29:
just logged in again and camera not showing. have to constantly reboot. and timelapse not occurring as a result of camera not staying active.

starbasessd, 2020-10-29:
I really hate to suggest it, but you may want to contact
https://shop.pimoroni.com/pages/contact-us and see if they have any ideas.
It may be a faulty camera or PI (I lean more towards the camera) and they
may have better suggestions on diagnosing the connectivity to the camera...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

mroberts77, 2020-10-30:
yeah, could be. Will advise [adafruit- middlefolk] we tried.

mroberts77, 2020-10-30:
ps- throttled=0x0

mroberts77, 2020-10-31:
Replaced and installed a new 32gB micro SD card and new image (per above info). Camera seems to stay on when browser is on and active to the pi. However, when I close the laptop or sleep mode, come back to no camera on, re boot needed. COuld there be something to do with connectivity of an active browser?

starbasessd, 2020-10-31:
No. Could be your browser has lost the connection.
Which browser are you using?
Try closing and re-opening the browser, not just the tab, the whole thing.
If that doesn't work, try rebooting the laptop, before rebooting the Pi
Also, try opening the web page:  http://[ip_address_of_Pi:8081
See if it is still streaming there.
Did you contact Pimoroni or Adafruit support?

mroberts77, 2020-10-31:
Firefox. Sent a note to Adafruit who said go to Pimoroni, posted on the discord, awaiting input (3 day).
Will try a shut down of my computer.
Camera was out of action last 2 days, and wierdly, we had someone spray white paint at our home and deck from the angle the camera would have been seeing. Is there any way someone else could have used the log data above for IP and location to take an opportunity to vandalize while camera was down?

starbasessd, 2020-10-31:
Doubtful. Unless you have a password on your wireless like 'password' or a
commonly hacked router for your internet.
I doubt anyone local to you is watching your conversations here to know
that the camera was down.
Please try restarting the browser first, then rebooting the laptop, if/when
you lose the camera.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

mroberts77, 2020-10-31:
ok. P/w is not easy. Been going good all day so far. have rebooted and restarted browser, tried Edge. Will see. ;0

mroberts77, 2020-10-31:
Have the iphone charger 5V 1A on an extension chord, and a lowere quality micro USB cable, just noticed that the wall plug for pi zero W is 5V 2.5A: is it possible it's a power supply issue?

starbasessd, 2020-10-31:
Most definitely.
If you can ssh in, or use keyboard & monitor, run the command
vcgencmd get_throttled [enter]
If it says anything besides '0x0', you have issues.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

mroberts77, 2020-10-31:
[root@meye-955e5835 ~]# vcgencmd get_throttled
throttled=0x0

starbasessd, 2020-10-31:
try it when you lose the camera.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

mroberts77, 2020-10-31:
got it! if camera lost after checking after an hour or so, will it still report that error, or will that error fix after the actual issue has occurred?

starbasessd, 2020-10-31:
It shows until the next reboot.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

mroberts77, 2020-11-01:
ok, tried 5V 2.5A power, and same issue: No camera this AM, and none after a reboot, 0x0 on tests ok. Will attach the logs files in case you can identify an issue, and will test with a better camera. Is it a matter of attaching the camera and the sd to a different board or will it have a different address and IP?

Do think it must be a hardware connection issue, but why it works when supervised and not when computer is off at night is bizarre.

[motioneye.log.txt](https://github.com/ccrisan/motioneyeos/files/5471511/motioneye.log.txt)
[motion.log.txt](https://github.com/ccrisan/motioneyeos/files/5471512/motion.log.txt)
[dmesg.log.txt](https://github.com/ccrisan/motioneyeos/files/5471513/dmesg.log.txt)
[boot.log.txt](https://github.com/ccrisan/motioneyeos/files/5471514/boot.log.txt)
[messages.log.txt](https://github.com/ccrisan/motioneyeos/files/5471515/messages.log.txt)

starbasessd, 2020-11-01:
If the same model, depends on how your DHCP works, probably not the same IP
address. If you had set a static, it'll be the same.
Checking logs

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

mroberts77, 2020-11-01:
so would have to use monitor and key board approach to get the new IP, or somehow need to log onto the router to check for new one?

starbasessd, 2020-11-01:
Please try dev20201026 from
Releases · ccrisan/motioneyeos
<https://github.com/ccrisan/motioneyeos/releases>
for your board.
Is your PIZero plus motionEyeOS auto discovering your camera or are you
needing to add it?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-11-01:
Correct.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

mroberts77, 2020-11-01:
think I tried this on Friday:  motioneyeos-raspberrypi4-20200606.img.xz and the pi didn't even start.
should it be the motioneyeos-raspberrypi-20200606.img.xz file for pi zero W?

I haven't had to select a camera, just shows up in the view. JUst rebooted again, and it's back.  But will try a different board and cameras including the NoIR.

starbasessd, 2020-11-01:
Yes, it should it be the motioneyeos-raspberrypi-20200606.img.xz file for
pi zero W.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

mroberts77, 2020-11-02:
so far, not so bad. Have it installed in 2 pi's and the drop rate on the intial pi zero seems to be substantially less with the DEV OS. Ran it overnight and appears to have taken images every hour and is capturing motion without a reboot today. Only caveat was this am, it wasn't operating, then needed some settings changed to refresh it, but didn't require a reboot. All seems positive so far [holding breath] ;0

starbasessd, 2020-11-02:
Watching.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

mroberts77, 2020-11-04:
all good so far, couple of days not dropped! ;0 would say that dev was good. "Watching" is .....?

starbasessd, 2020-11-04:
Watching for issues.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

closed 2020-11-06 by mroberts77