motioneye-project/motioneyeos #2562: samba share

jinyistudio, 2020-10-30:
Can i connect motioneyeos(pi2) with samba from other PI ?

starbasessd, 2020-10-30:
Yes

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-10-30:
If you would like a fuller answer, it would be helpful to have answered the
basic questions of your setup,
at a minimum, what is the other PI?
What is it running (which OS and version)?
What SAMBA setup?
Is the other PI on a static IP address?
Are you trying to save everything there, or only as a backup to the local
storage?
Do you have SSH or console access to both PIs? (as well as the WebGUI on
the motionEyeOS PiZero)
There will probably be more questions.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

jinyistudio, 2020-10-31:
I have two sbc(pi2,pi3). pi2 installed motioneyeos 20200606. pi3 has running node-red as gateway. 
my idea is
pi2 setting samba share for still image. i have enable its services/enable samba server. 
pi3 receive motion detection notification/GET from pi2 and then read still image via samba and then send line notify(https://notify-bot.line.me/doc/en/) to user.
**My question is how to create a samba share in pi2 ?**

jinyistudio, 2020-10-31:
pi2 debian, 192.168.1.182, With WebGUI to operating system.
pi3 debian, 192.168.1.181
I have SSH and console access to both PIs.

jinyistudio, 2020-10-31:
Hi, 
I find this https://raspberrypi.stackexchange.com/questions/40974/access-network-samba-share-from-pi-client
I tried many times and finally let me try it out, thank you and Happy sunday

closed 2020-10-31 by starbasessd