motioneye-project/motioneyeos #2563: Jerky video with Zero W...

Bromley47, 2020-10-31:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

motionEyeOS Version 0.42.1

I am running motionEyeOS version: 20200606.

Board Model Raspberry Pi Zero W

I am using the following board/model: Raspberry Pi Zero W

Raspberry Pi Camera: V2

I am using the following type of camera: MMAL Camera

My camera model is: Raspberry Pi V2

Network Connection; wifi

My motionEyeOS unit is connected to the network via: wifi

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * (e.g. USB WiFi adapter)
 * (e.g. External Hard Disk)

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)
 
Have searched forums and seen other peoples questions relating to jerky video on the above arrangement, but cannot find a definitive answer.

In standard mode I am getting 2.30/2.80 fps and video time read out seems to jump  4 or 5 secs at a time and the video movement coincides with this, which makes for choppy video and also on the recording.

I have tried fast network camera and a great improvement but not sure how to add a second camera or if the files are still stored on the SD card and will get transferred as I am doing now. Otherwise happy to use FNC.

Settings are:
Movie quality: 100%
Movie format: H.264 (.mp4)
All services: On
Still images: Off
Streaming: Off.
Motion detection: On

I use sftp to transfer images to a Windows 10 PC.

I will attach logs if considered helpful.

Before motioneyeos I use to use RPi Cam Control v.6.6 and video was excellent but user interface not so, and only get basic controls if second camera added. Much prefer motioneyeos if I can get it to run reasonably.

starbasessd, 2020-10-31:
Fast Network Camera precludes a 2nd camera on the PiZero.
Due to it being a single core processor (effectively an original Pi 1) and
other limitations, you are going to have reduced capabilities.
As a Network Camera (Fast Network Camera & PiCam) it works well, but you
need a Pi3 or Pi4 set up as a camera hub to do management and recording.
If you use SCP to do the file transfers, you can turn off all services, and
improve things a bit more.
SCP is available in PuTTY as another program, and works well.
You can also use a SAMBA (SMB) share to be you primary storage location
directly, and still turn off SMB and FTP services.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Tameiki, 2020-11-17:
Hello Bromley47.

I was also looking for those performance with Pi Zero, and I found this topic. https://github.com/ccrisan/motioneyeos/issues/2325
If you didn't already fount it, you'll read in it some tips to have a descent IpCam with Pi Zero W.

In a easy-readable list:
- 720P@5FPS recording
- (or ?) 30FPS videostreaming without motion detection or recording to a more powerful hub which will motion detection / record.
- Don't look at stream in browser while testing.

For me it works well. (I only tested for now with 800x600 at 30fps stream to a motioneye docker on a more powerful machine)

I'll come later with other test results.