motioneye-project/motioneyeos #2564: Multiple PiZero W network question

questrov, 2020-11-03:
I have (lost count) probably a dozen Pi of different flavors running both networked and Wi-Fi spread out over several acres on a farm.  Several are PiZero W running motioneyeos in Fast Network Camera ported to [https://blueirissoftware.com/](url) on a dedicated Windows PC.  

Wired network to several outbuildings each with enterprise AP's running [https://www.peplink.com/products/wi-fi/](url) mostly AC Mini's and Rugged with a https://www.peplink.com/products/balance-20/ as the main router.  

There is also [https://www.tp-link.com/us/business-networking/outdoor-radio/cpe210/](url) mounted high on a building for several solar powered PiZero W located in pastures.  Some of the solar Pi's have external wi-fi antenna and some don't, depending on distance.  

For months the solar unit Pi (logs attached) ran perfect with no reboots other than occasionally dead batteries.  Now when I try to add a PiZero W to one of the Peplink AP's it causes the solar Pi's on the TP CPE210 to crash.  It's almost like there is some netwrok traffic or issue that motioneyeos does not like and it seems to be generated from the AC Mini when a PiZero is associated.

Wi-Fi channels are spread out and dedicated, for example AC mini's on ch1, TP Link on ch10 and misc on ch6.  Pi's all have fairly current OS and static IP's.  I have noticed the same issue (more or less) with other AP's but can't pin point what the exact issue is.

Any ideas what/how to troubleshoot?  The only issue I can possibly see from the logs is clock/time errors but not sure how to interpret the details.  Seems like a network issue that somehow affects motioneyeOS?

### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 20200606

### Board Model

I am using the following board/model: PiZero W

### Camera

I am using the following type of camera: (choose from V4L2, MMAL, Network Camera, Fast Network Camera and Simple MJPEG Camera).

My camera model is: Fast Network Camera

### Network Connection

My motionEyeOS unit is connected to the network via: Wi-Fi
[messages.log](https://github.com/ccrisan/motioneyeos/files/5483240/messages.log)
[boot.log](https://github.com/ccrisan/motioneyeos/files/5483242/boot.log)

### Peripherals

I am using the following peripherals that I consider relevant to this issue: None

 * (e.g. USB WiFi adapter)
 * (e.g. External Hard Disk)

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)

starbasessd, 2020-11-03:
How are you giving static IPs? Primary router, each individual AP,. Static
on each Pi, something else?
 I use my router to link static IPs to the mac address of each Pi on my
network. Handy for all my testing, too. Works for both wifi and ethernet,
with transparent APs.
If given out by individual APs, are they all on different subnets?
Can you run a command to clear logs, please, then watch 24 hours, and post
the new logs?
command is
logrotate -f /etc/logrotate.conf
Thanks

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

questrov, 2020-11-03:
Thanks starbasessd great tip, clearing the logs will be much help to maybe make sense of the things, mostly cryptic to me! 

I'm giving static IP for motioneyeos (PiZero W) through the meye webpage interface under 'Network'.
Where motioneye (Pi 3) is running on Raspbian static IP is thru DHCP reservation on the Router.
Everything is on the same subnet.

I did notice (and remove) where we had both above for at least one of the PiZero W, not sure why.  Think I tried to reserve IP via Router in the beginning but with Pi's constantly moving/changing (they get used for IOT projects too) it was easier to use the meye interface.

starbasessd, 2020-11-03:
Include all 5 logs, if you can, please.
Is the one that has issues sitting in a 'equal distance' location (equal
signal strength between 2 or more APs with the same AP name?
Or maybe in a low/dead space?
'Standard' channels are 1, 6, & 11 for 2.4 wifi.
I'll wait in the logs...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

closed 2022-01-19 by starbasessd