motioneye-project/motioneyeos #2565: No email attachment after changing Image File Name

Gblenn, 2020-11-03:
I am trying to change the file name of the pictures saved so that each camera ends up in it's own folder but all in the same top folder (by date).

Menu item is Still Images
Image File Name : Year%-Month%-Day%**/CameraN**/Hour%-Minute%-Second%

Pictures are stored on a Network Share (NAS) and this works fine regardless of naming structure. However, after inserting the "/CameraN/" to create an additional folder level, emails no longer have a picture attached... 
Reverting to the original file structure brings back the attachments in emails again. 

Other name changes, that do not involve adding a folder level, like adding "CameraN to the file name, or changing the first level folder name, works fine and pictures are then attached to email.
E.g. Year%-Month%-Day%/CameraN-Hour%-Minute%-Second% sorks fine and generates 2020-11-03/Camera1-15-25-44.jgp as well as attaching the picture to email.

### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: motionEyeOS dev20201026.

### Board Model

I am using the following board/model: PI 3B+.

### Camera

I am using the following type of camera: Simple MJPEG Camera).

My camera model is: (Dlink DCS932L).

### Network Connection

My motionEyeOS unit is connected to the network via: (Ethernet).

### Peripherals

I am using the following peripherals that I consider relevant to this issue:
### Log Files

I consider the following log files relevant to this issue:

starbasessd, 2020-11-03:
If you can, please post motion and motioneye logs.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Gblenn, 2020-11-04:
Ok not sure the files are showing anything helpful at all. But I rebooted and triggered separate 3 events. The first two with standard naming structure and the last one with /Cameraname/ added to the path. 
The first two resulted in emails with pictures attached, whilst the last event resulted in an empty email.

With the third event, the file location gets a new folder /Cameraname/H-M-S and there were 8 pictures in that folder.

motioneye.log looks like this after the Reboot:
2020-11-04 11:38:38: [motioneye]     INFO: rebooting
2020-11-04 11:38:38: [motioneye]     INFO: executing "/sbin/reboot"
2020-11-04 11:38:39: [motioneye]     INFO: interrupt signal received, shutting down...
2020-11-04 11:38:39: [motioneye]     INFO: server stopped
2020-11-04 11:38:39: [motioneye]     INFO: tasks stopped
2020-11-04 11:39:05: [motioneye]     INFO: hello! this is motionEye server 0.42.1
2020-11-04 11:39:10: [motioneye]     INFO: cleanup started
2020-11-04 11:39:10: [motioneye]     INFO: wsswitch started
2020-11-04 11:39:10: [motioneye]     INFO: tasks started
2020-11-04 11:39:10: [motioneye]     INFO: mjpg client garbage collector started
2020-11-04 11:39:10: [motioneye]     INFO: smb mounts started
2020-11-04 11:39:10: [motioneye]     INFO: server started
2020-11-04 11:42:20: [motioneye]  WARNING: 403 GET /login/?_=1604486540756&_username=_&_signature=removed 2.88ms
2020-11-04 11:45:04: [sendmail]     INFO: sending email
2020-11-04 11:45:06: [sendmail]     INFO: email sent
2020-11-04 11:46:27: [sendmail]     INFO: sending email
2020-11-04 11:46:29: [sendmail]     INFO: email sent
2020-11-04 11:48:06: [sendmail]     INFO: sending email
2020-11-04 11:48:07: [sendmail]     INFO: email sent

motion.log looks like this:
[0:motion] [NTC] [ALL] conf_load: Processing thread 0 - config file /data/etc/motion.conf
[0:motion] [NTC] [ALL] config_camera: Processing camera config file camera-1.conf
[0:motion] [NTC] [ALL] motion_startup: Logging to syslog
[0:motion] [NTC] [ALL] motion_startup: Motion 4.3.1 Started
[0:motion] [NTC] [ALL] motion_startup: Using default log type (ALL)
[0:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
[1:ml1:Vardagsrum] [WRN] [NET] netcam_next: called with no data in buffer

starbasessd, 2020-11-04:
Can you post the complete log files, please? (attachments are fine)

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Gblenn, 2020-11-04:
Do I need to do something in particular to get more information into the files. The motion.log is the complete files, and motioneye.log looks pretty much the same throughout...

starbasessd, 2020-11-04:
If you are attaching to the email, github strips before forwarding. You'll
need to go to the issue in github and attach the files there.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Gblenn, 2020-11-04:
Sorry I'm new to this so my apologies if I'm going about this the wrong right way. Are you saying you can not see the text from the two log files that I pasted into the comment (2h ago)? 
The entire file "motion.log" was only 7 rows long, I pasted the whole thing... 

Perhaps I'm not looking at the right files? /data/log/motion.log and motioneye.log ??
I log on to the Pi3 and pull them using filezilla.

starbasessd, 2020-11-04:
If you go to the WebGUI, you can grab 5 files
Log Files
motion.log

motioneye.log

messages.log

boot.log

dmesg.log

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Gblenn, 2020-11-04:
Ok, same files then... 
Only now motion.log has some error messages in it because I switched off the camera...
[motion.log](https://github.com/ccrisan/motioneyeos/files/5487883/motion.log)
[motioneye.log](https://github.com/ccrisan/motioneyeos/files/5487884/motioneye.log)

starbasessd, 2020-11-04:
Are the 3 other files empty?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Gblenn, 2020-11-04:
Sorry, not at all, I thought those were the one's you were looking for. But here they are...
[messages.log](https://github.com/ccrisan/motioneyeos/files/5487913/messages.log)
[boot.log](https://github.com/ccrisan/motioneyeos/files/5487914/boot.log)
[dmesg.log](https://github.com/ccrisan/motioneyeos/files/5487917/dmesg.log)