motioneye-project/motioneyeos #2567: I can't mount usb stick to /data/usbstick

jinyistudio, 2020-11-04:
Hi
I plug usb stick(8gb) to pi2. pi2 know stick in the /dev/sda1. But I can't mount it to /data/usbstick. Why ? Can I plug an external USB used to store still image under motioneyeos ? If i can How to do it ?

starbasessd, 2020-11-04:
How are you trying to mount it? What command(s)?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

jinyistudio, 2020-11-04:
1. i use usb stick to save still image(motion detection)
2. mount /dev/sda1 /data/usbstick

starbasessd, 2020-11-04:
If plugged in before power on, it shows as /data/media/sda1
If you want to mount at /data/usbstick, you'd have to manually add the
folder:
mkdir /data/usbstick [enter]
then manually mount sda1 to the folder
mount /dev/sda1 /data/usbstick [enter]

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

jinyistudio, 2020-11-04:
1. Yes, I have see it in /data/media/sda1 when i pluged it before poweron. 
2. Do i need to umount /data/media/sda1 if i want mount to /data/usbstick ?

starbasessd, 2020-11-04:
1) Good deal
2) No. If you do a df -h, you'll see /dev/mmcblk0p3 is mounted in 3
places...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

jinyistudio, 2020-11-04:
thank's

closed 2020-11-04 by jinyistudio