motioneye-project/motioneyeos #2568: Rasperry pi zero + motioneyeos doesn't detect my PS3 Camera

ade-x1, 2020-11-06:
Hello team!
I am running motionEyeOS version: 20200606 on a raspberry pi zero, this is the first time I am trying to setup motioneyeos with this camera, however it always worked fine with motion. During this first setup, once I select Local v4l2 camera and I click okay the preview doesn't show any video. Is there a specific setup or codec that needs to be installed?

Many thanks!

[image]
[image]

starbasessd, 2020-11-06:
Need some more info:
Can you post your logs? Boot, messages, motion, motioneye and dmesg, please?
Have you tried to test with RaspberryPiOS and raspivid?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

ade-x1, 2020-11-06:
Hi @starbasessd , I have updated my original post with the logs. No I haven't as I had a working system with motion and I thought that would have been compatible right away

starbasessd, 2020-11-06:
can talk to the camera on some systems (like Debian) natively.
What was motion running on before?
Please try running a current version of RaspberryPi OS (like 2020-08-20) on the Pi and see if it sees it.
I also see an Apple keyboard/mouse? Are you running a USB hub? Is it self powered?
If not, can you try connecting the camera directly to the PI, and just use SSH and/or the WebGUI?

ade-x1, 2020-11-06:
The USB keyboard was there before connecting the camera (then I rebooted the pi). Motion was running on Raspian. I'll try to install raspberry pi os now

ade-x1, 2020-11-06:
Is there any other test that I can do? It is extremely slow on the pi zero is still opening chromium since minutes now

starbasessd, 2020-11-06:
I was hoping that it came up without doing anything special.
If the camera comes up, you can look to see if there is a folder under
/lib/firmware for the camera.
If it doesn't come up, it explains why it doesn't work in motionEyeOS.
We can then go about trying to figure out what the issue is.
Can you run a command line on console or ssh?
If so, run:
lsusb -t [enter]
dmesg | grep usb [enter]
We are trying to discover how it's discovered in RaspberryPiOS.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

ade-x1, 2020-11-06:
Sorry if it took a while but it wasn't recognising my mouse and I had to setup the wireless keyboard. I am not really sure how to test the webcam with the desktop OS, but here there is a screenshot of dmesg output. Thank you for your help with this

[image]

starbasessd, 2020-11-06:
OK, it sees an OmniVision camera. Is Cheese installed?
Or, can you run

sudo apt-get install fswebcam [enter]
or
sudo apt install cheese [enter]
then runfswebcam -r 1280x960 --no-banner ./camtest.jpg [enter]
or
cheese [enter]

You should be able to use the file manager to open the file.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

ade-x1, 2020-11-07:
Good morning! With cheese it failed but it worked just fine with fswebcam

ade-x1, 2020-11-07:
I tried to reinstall motioneyeos and now is working fine ¯\_(ツ)_/¯. It's a bit slow to show the movements but I guess that's only a matter of settings

closed 2020-11-09 by starbasessd