motioneye-project/motioneyeos #2569: Adding the Raspberry Pi NoIR Camera Module

mroberts77, 2020-11-08:
Have the latest version of Motion IO running on Zero W, trying to add the NoIR camera, and have seen but can't seem to track down:
1) extra motion commands to convert to black and white (when IR lights used, still pinkish)
2) adjustment for blurryness?

[image]

### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: (insert your version here, e.g. 20180314).

### Board Model

I am using the following board/model: (insert your board model here, e.g. Raspberry PI 3B+).

### Camera

I am using the following type of camera: (choose from V4L2, MMAL, Network Camera, Fast Network Camera and Simple MJPEG Camera).

My camera model is: (insert your camera model, e.g. Logitech C290).

### Network Connection

My motionEyeOS unit is connected to the network via: (choose your network connection, e.g. WiFi, Ethernet).

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * (e.g. USB WiFi adapter)
 * (e.g. External Hard Disk)

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)

mroberts77, 2020-11-08:
Ran using Raspbian only, and seems to be clearer than for OS. Also, can convert to Black and white with this code 
`camera.color_effects = (128,128) # turn camera to black and white`
>  can that be done in Motioneye OS in extra Motion Options?

[image]
[image]

mroberts77, 2020-11-09:
Blurriness seems to have been adjusted with manual focus and restart.
How to convert to black and white only?
[image]

cap9qd, 2020-11-09:
You can set the AWB mode to greyworld by selecting the option in the video
section if using fast network camera or by using the extra motion options
listed in the post below if using the full installation.

https://github.com/ccrisan/motioneye/issues/1370#issuecomment-527471418

mroberts77, 2020-11-10:
tried typing:
mmalcam_control_params -awb greyworld

didn't seem to make a difference, also tried 'tungsten' instead, no apparent effect: probably missed the way to add the info?

Pi-DD, 2020-11-10:
Did you restart Motioneye? Check what level of Motion is installed - needs to be 4.3 at least for “greyworld”. I’m running 4.3.1

mroberts77, 2020-11-10:
Pasted in rebooted, and even "apply" still color, got the version you have: what am I doing wrong?

[image]

[image]

cap9qd, 2020-11-11:
Greyworld isn't black and white. It is just properly white balanced now but
still color.

closed 2021-01-29 by mroberts77