motioneye-project/motioneyeos #2570: Raspberry Pi 4 not detecting PINOIR camera

Drakenns3, 2020-11-09:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I have the latest version of MotionEyeOS installed, 10-20?? maybe. Everything seems to work. Can connect via putty or chrome, but the camera will not display an image. I have the PINOIR camera installed. 

I am running motionEyeOS version: (insert your version here, e.g. 20180314).

### Board Model

I am using the following board/model: (insert your board model here, e.g. Raspberry PI 3B+).

### Camera

I am using the following type of camera: (choose from V4L2, MMAL, Network Camera, Fast Network Camera and Simple MJPEG Camera).

My camera model is: (insert your camera model, e.g. Logitech C290).

### Network Connection

My motionEyeOS unit is connected to the network via: (choose your network connection, e.g. WiFi, Ethernet).

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * (e.g. USB WiFi adapter)
 * (e.g. External Hard Disk)

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)

starbasessd, 2020-11-09:
Please answer as many questions as you can.
Which version of motionEyeOS at a minimum, and incluse the boot, dmesg and
messages log files.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Drakenns3, 2020-11-09:
I apologize. I wrote this at work because I couldn't find an answer after searching all last night. I'll update with logs and all that once I get home.

Raspberry Pi 4
Pi Camera v2 NoIR
Motion v raspberrypi4-dev20201026
Wifi Connection (works and tested)
No other Peripherals than the camera

How do i get the logs off? It's a headless install? Can I FTP to the Pi and download them?

starbasessd, 2020-11-09:
In the WebGUI there are links under "Expert Settings" to the main log files.

Drakenns3, 2020-11-09:
[motioneye.log](https://github.com/ccrisan/motioneyeos/files/5513597/motioneye.log)
[motion.log](https://github.com/ccrisan/motioneyeos/files/5513598/motion.log)
[dmesg.log](https://github.com/ccrisan/motioneyeos/files/5513599/dmesg.log)
[boot.log](https://github.com/ccrisan/motioneyeos/files/5513600/boot.log)
[messages.log](https://github.com/ccrisan/motioneyeos/files/5513601/messages.log)

starbasessd, 2020-11-09:
If you try to manually add an mmal camera, does it see it?
Are you sure the camera is installed correctly? The bare part of the ribbon
cable faces the circuit board.
If both are yes, the try installing RaspberryPiOS (2020-08-20) and go
through the testing here and report:
https://www.raspberrypi.org/documentation/raspbian/applications/camera.md

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Drakenns3, 2020-11-10:
Installed 2020-08-20 and was able to see an image from the camera no problem. Reloaded MotionOS and nothing. The camera is not detected when trying to install a mmal. Is there a way to install it from the command line?

[messages (2).log](https://github.com/ccrisan/motioneyeos/files/5520624/messages.2.log)
[motioneye (2).log](https://github.com/ccrisan/motioneyeos/files/5520625/motioneye.2.log)
[motion (2).log](https://github.com/ccrisan/motioneyeos/files/5520626/motion.2.log)
[boot (2).log](https://github.com/ccrisan/motioneyeos/files/5520627/boot.2.log)
[dmesg (2).log](https://github.com/ccrisan/motioneyeos/files/5520628/dmesg.2.log)

starbasessd, 2020-11-11:
Need to get some info from the RaspberryPiOS install to try and discover
why it's working there and not in motionEyeOS. It usually involves a
missing module or firmware.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Drakenns3, 2020-11-11:
OK. Let me know next steps.
I am having to add the start and fix files from the RASPI image to get motionOS to load.

starbasessd, 2020-11-11:
If the camera is working in RaspberryPiOS, Then look at or post your dmesg
log here.
If you want to try something else, try installing motionEyeOS dev20201026

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Drakenns3, 2020-11-11:
DMESG from RaspberryPiOS? I already posted the one from MotionEyeOS. I'm also already using motionEYEOS dev20201026.

starbasessd, 2020-11-11:
Yes, dmesg from RaspberryPiOS.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Drakenns3, 2020-11-11:
This is the sys log which contains DMESG data, from a raspberry lite install that works with the camera
[syslog.log](https://github.com/ccrisan/motioneyeos/files/5526891/syslog.log)

starbasessd, 2020-11-11:
What app are you using in RPiOS-Lite to view live video from the camera?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Drakenns3, 2020-11-11:
Raspistill

Drakenns3, 2020-11-12:
With a monitor and keyboard attached I get a 5 second preview before it takes the image. I was using "raspistill -o image.jpg"

Drakenns3, 2020-11-13:
Any luck or anything I can do to help?

starbasessd, 2020-11-13:
Waiting to see if anyone else has any ideas. I only have 'official'
Raspberry Pi Cameras for the CSI interface, and they all work. I've seen
other cameras reported here, but they show up in Raspbian & dmesg, and I
can figure out if they use a different driver. your reported dmesg info
doesn't indicate which driver or firmware that I can tell.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Drakenns3, 2020-11-13:
This is an official camera purchased straight from the raspberry pi website.  It's v2.1 NoIR.

Drakenns3, 2020-11-13:
Also, is there a setting I can change somewhere to up the logging by dmesg?

Drakenns3, 2020-11-13:
[syslog.log](https://github.com/ccrisan/motioneyeos/files/5538398/syslog.log)
Attached is the most recent log I pulled. I increased logging to level 6 (I think), by editing the /etc/sysctl.conf and uncommenting and changing the kernel.printk line to read:
kernel.printk = 6       4       1       6

then confirming after reboot:
cat /proc/sys/kernel/printk
"6       4       1       6"