motioneye-project/motioneyeos #2571: Network Share Connection Stops Working

erhughesjr, 2020-11-10:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 20200606 on two cameras.

### Board Model

I am using the following board/model: Camera 1 is on a Raspberry Pi 3B+, and is acting as a hub for Camera 2.  Camera2 is on a Raspberry Pi Zero W.    Both devices have a 32 GB SD card.    Both have substantial power supplies (3A on RPi3B+ and 4 Amp on the PiZero).

### Camera

My camera model is: I am using the following type of camera: Camera 1 is a LoveRPI NOIR camera.  (https://www.amazon.com/gp/product/B07KF7GWJL).  I've also tried the PiCam V2 as well.   This is entered as a Camera Device vc.ril.camera, and Camera Type is MMAL Camera.    Camera2 is a PiCam night vision (see link: https://www.amazon.com/gp/product/B07VSPSNL8).  It is listed as a Camera Device: http://192.168.0.15/config/1 and Camera Type Remote motionEye Camera. 

### Network Connection

My motionEyeOS unit is connected to the network via: a Wi-Fi network with static IP addresses

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

I have setup a network sharing device with a Raspberry Pi 4B with 8 GB of RAM and a 64 GB fast SD card (writes 85 MB/s and reads 130 MB/s).    I set this up with Samba using this link as my guide:  https://magpi.raspberrypi.org/articles/samba-file-server.   It does seem to be working.

I have a second Raspberry Pi 4B 8GB RAM and 64 GB SD card that I use to log into the cameras from its web browser.   I have confirmed this RPI4 can see the network sharing RPI4.     

### Log Files

I have not attached the log files.   I can't read the motioneye.log file because I don't know what format it is written in.   It will not open with the standard UTF-8 setting.  
 
### Issue:   There are no instructions on setting up a network drive connection and it took more than a day to get this to work.   I had to use the IP address for the Server name, and I had to add a sub-directory to the share name.    I left root with /".   So it is currently setup as:

Storage Device: Network Share
Network Server: 192.168.0.159  --- this device is a static IP on my local area network
SMB Protocol Version:   I have confirmed all settings work.
Share Name: share/cam1/
Share Username:   the name setup using smbpasswd on the network sharing RPi4
Share Password:  the password setup using smbpasswd on the network sharing RPi4.
Root Directory: /

The problem I am having is that the storage only works for a short period of time.    When I first enter the above information and click Test Share, it will show success!  Then I press Apply and I can correctly trigger motion video.   But then after may 5 minutes or so I lose video and when I check the network share folder I can see there are no more  new files being stored.  I can usually get about 2-3 events of storage before this stoppage occurs.     If I go back in as admin and hit Test Share again, I get the Accessing network sharing failed message.   I  cannot make this reconnect EXCEPT when I change the SMB Protocol.   If I go from say 1.0 to 2.0 and then hit Test Share, it will show success again.   I can then click Apply and it will work again for 5-10 minutes before loosing video.  So I can repeat this process again and again and get the video back.        >>>>>>>> What can I do to stop the video feed dropout (on the 2nd RPi4) and saving stoppage on the network share folder?   <<<<<<<<<<<        Note that I have run days without stoppage when saving to the local SD card on both cameras.   Also note that I have currently on Camera 1 on network share.   Camera 2 is on disk storage (Custom Path).

starbasessd, 2020-11-10:
If you use the WebGUI, you can access the logs under Settings, Expert
Settings, and when you click on each of the 5 logs, they are all .log
files, in a text format.
If you are able to ssh, you can scp the files (located at /var/log/)
If you are using current SMB (Samba) on the host (file server) you should
be able to select V3.
Do you have a Windows PC to test the shares with?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

erhughesjr, 2020-11-10:
> If you use the WebGUI, you can access the logs under Settings, Expert Settings, and when you click on each of the 5 logs, they are all .log files, in a text format. If you are able to ssh, you can [...]

I tried all versions 1.0, 2.0, 2.1, 3.0, 3.1.1.   They all work briefly before disconnecting.    Using the WebGUI on the RPI4 (the 2nd one -- not the server), I can read the motion.log, boot.log, dmesg.log files as text files.   If I try to open the motioney.log or messages.log, I get an error saying "The document was not UTF-8 valid and suggests I select an encoding from a list of 20 encoder choices.    However, I copied these files and put them on a memory stick and I was able to open them on a Windows Computer.     It appears I am getting a lot of errors  such as these: **2020-11-10 14:25:06: [motioneye]  WARNING: Connect error on fd 15: ECONNREFUSED**
[boot.log](https://github.com/ccrisan/motioneyeos/files/5519837/boot.log)
[dmesg.log](https://github.com/ccrisan/motioneyeos/files/5519838/dmesg.log)
[messages.log](https://github.com/ccrisan/motioneyeos/files/5519839/messages.log)
[motion.log](https://github.com/ccrisan/motioneyeos/files/5519840/motion.log)
[motioneye.log](https://github.com/ccrisan/motioneyeos/files/5519841/motioneye.log)

starbasessd, 2020-11-10:
Where did the messages log come from? It's showing heat issues, network
issues, communications issues.
Only the last 2 boot log entriy sections show anything even close to a
normal boot.
The motioneye log file indicates a connection error:
2020-11-10 08:57:57: [motioneye]    ERROR: failed to mount smb share
"//smb://raspberrypi.local/share/scitreks-cam1/" at
"/data/media/motioneye_smb___raspberrypi_local_share_scitreks_cam1__pi"
is invalid.
If you are trying to set up the connection TO the share, you cannot use the
instructions at RaspberryPi.org.
You need to set SAMBA share up under Settings, File Storage, Storage Device
= Network Share
then enter Server IP address for Name
path, username & password as appropriate, etc.
Then Click Apply, then click Test.
If you are using motionEyeOS to be a Samba File Server, you cannot use
those instructions, either.
You turn on write under Services, and use the admin username & password.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

erhughesjr, 2020-11-10:
All those logs are from Camera 1.   Both cameras have been working for days, and it appears the logs are appending.   It took me a while to get Cam1 working because its was a RPI3B+, and I found motioneyeos would not boot the first time without an RJ/45 cable plugged in.   It ignored the wpa_supplicant.conf file until I could set up the wifi setting completed from the webgui.  I also changed from dhcp assigned IP addresses to static.    So some of that is in the logs. The camera and the RSP3B+ are in an enclosure, which is fanless.  So I am sure it was running hot.   I've got fans arriving tomorrow for this issue, but it is clear there was CPU throttling as a result.

I've just today got share working.   There is very little information available on how to do this, so it took me a while to brute force try all options for settings.    Remember that I have no issues with saving to SD, and camera#2 has no issues saving to the share.   It's been working without error for a number of hours now.    The settings are identical between the two cameras, with the exception of the share folders are share/cam1 and share/cam2 and their names (scitreks-cam1 and scitreks-cam2).    And of course, Camera1 is the hub camera.

I set up both cameras just as you stated via the motioneyeos webgui and logging in as admin.  I go to Settings>File Storage> and then this:

Storage Device: Network Share
Network Server: 192.168.0.159 --- this device is a static IP on my local area network
SMB Protocol Version: I have confirmed all settings work.
Share Name: share/cam1/
Share Username: the name setup using smbpasswd on the network sharing RPi4
Share Password: the password setup using smbpasswd on the network sharing RPi4.
Root Directory: /

The difference in what you said is that I hit Test and then Apply.    I hit Test and if it was successful then hit Apply.    I confirmed the order of Apply and Test make no difference to my problem.     

I have done a few more tests.   I've rebooted everything.   The video files are saving for both cameras when motion triggers it.    Camera 2 (on the PiZW) is working flawlessly with no disconnects.   Camera 1 is saving when there is motion, but the stream does stop and saving stops after a few minutes.    I even tried to create a new folder for Camera 1, but problem persists.   I think I may try to remove Camera 2 to see if Camera 1 being a camera hub is the issue.   Anything else you can think of to try?

starbasessd, 2020-11-11:
The share name is usually one folder, not a folder path.
The folder path is the 'root directory'.
So share name would be share
root folder would be cam1

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

erhughesjr, 2020-11-11:
> The share name is usually one folder, not a folder path. The folder path is the 'root directory'. So share name would be share root folder would be cam1

It will not work if I put anything other than "/" for the root directory.   As long as I put "share/cam1" for the Shared Directory, it will work for a while for cam1.    I found out from a reddit post from a person having a similar problem.  They said their system would run correctly with the name of the folder in the Shared Directory, and they suggested leaving "/" for Root Directory.   They posted it to let others know.   Note that this person was running 2 cameras as well.   Here's the post: https://www.reddit.com/r/raspberry_pi/comments/e7871p/motion_eye_on_raspberry_pi_3b_network_share/ 

By the way, Cam2 has continued to work as "share/cam2 and "/" without any interruption for 24 hours.    Only Cam1 seems to be the problem, and this is the one that was acting as a hub.   So I removed cam2 away and left cam1 as a standalone (no hub).    Since then, Cam2 has been working flawlessly as before, but Cam1 is still having the disconnects.   The only way it reconnects is by changing the smb version number and then hit Apply, or rebooting.     I confirmed all settings between the two cameras are identical with the exception of names and folder names.    Cam2 is dhcp addressed, and Cam1 is static ip addressed.   But I tried setting Cam1 to dhcp addressed and it made no difference.    The hardware is different of course -- Cam1 is on a RPi3B+ and Cam2 is on a RPiZW.

Is there someway there could be a corruption of motioneyeos on that cam1 computer (the RPi3B+)?    Should I reburn the SD card with the OS?

erhughesjr, 2020-11-11:
I have found out what the issue was.    I had used the same network drive samba username and password for both cameras -- cam1 and cam2.   I decided to enable Guest on samba on the Network Share device, and logged in using Guest on my motioneye cam1.   Cam2 was logged in using the username (pi) and password I had used before.    Since doing this, I have not had a disconnect in over 2 hours.    I also added cam2 to cam1 (it becomes a hub again), and that also works.    I've tried this on Windows 10 Pro, and a Raspberry Pi 4B with the same results -- it works!   So the important points I discovered on this whole process:
(A)  For setting up a Network Share to save movie data, use these steps
    (1) Storage Device:  Network Share --- Its best to use an IP address for the Network Share -- not the device name.   There may be instances the name may work, but the IP address works for both Windows and Raspberry Pi OS.
    (2) Share Name must be <shared folder name>/<folder name for specific camera>.  In my example this was share/cam1/
    (3) Root Directory should be left as "/".    If I used Share Name share and Root Directory /cam1/, it would not work.   
    (4) As of 11/11/2020, all protocol versions of smb (1, 2, 2.1, 3, 3.1.1) will work fine for all motioneyeos cameras when the network drive is uses samba.
    (5) Username and Password:   If you are using Samba on your Network Drive, then you should use either "Guest" for all 
cameras or use a different username and password for each camera.  
(B)  Use Remote motionEye Camera when adding a camera, and enter http:/<ip address>to see other motioneyeos cameras on the network. 
(C) When setting up Samba on your Network Share device, follow the steps listed here:  https://magpi.raspberrypi.org/articles/samba-file-server.   The only change to this is the last line -- you should enter  "sudo /etc/init.d/smbd restart" instead of "sudo /etc/init.d/samba restart"

Thanks for your help starbasessd.   Your comments gave me some ideas on things to check.

erhughesjr, 2020-11-11:
Given my above tests, I have solved my issue as described above.

closed 2020-11-11 by erhughesjr