motioneye-project/motioneyeos #2573: Mount disks on UUID for consistent mountpoints

fredrik-rambris, 2020-11-12:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: dev20201026

### Board Model

I am using the following board/model: Raspberry Pi 4 Model B Rev 1.1

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

* USB hard disk

I am having issues where the disk gets remounted to a higher and higher device. Probably my disk powersupply is a bit shaky or something else. So after a while the disk is mounted on /media/sdb1, sdc1, sdd1 and so on.

Would it be possible to mount it with another variable than deviceid? Like so in udev rules

`SUBSYSTEMS=="scsi", ENV{ID_FS_TYPE}=="vfat|ext2|ext3|ext4" RUN+="/bin/mkdir -p /data/media/%k", RUN+="/bin/mount /dev/%k /data/media/%E{ID_FS_UUID}"
SUBSYSTEMS=="scsi", ENV{ID_FS_TYPE}=="ntfs" RUN+="/bin/mkdir -p /data/media/%k", RUN+="/usr/bin/ntfs-3g /dev/%k /data/media/%E{ID_FS_UUID}"`

That way if it ever gets a different device name the mount point will stay the same.

starbasessd, 2020-11-12:
When you notice the USB drive mounted on sdb1, does mount command or df -h command show anything still mounted on sda1? 
It may be a band-aid fix, but it does sound like either the USB interface may be running at the ragged edge of available power, or may be failing. 
I would test it by using a self-powered USB hub for the drive. I have several 1TB physical drives I use for various Pi3 & Pi4 projects, and if I notice any random flakeyness that's my first solution. Trying to power them with a PI is asking for trouble. USB Thumb drives don't ever seem to exhibit the issue because they require MUCH less power.  Even USB SSD don't have the issue nearly as often, and the issue usually follows a dying USB to SATA interface which is pulling too much power.

starbasessd, 2020-11-12:
Oh, and you should be able to modify the udev rule to use a fixed point.
It would not be practical for other users who swap thumb drives regularly. 
If you need help temorparily mounting the file system read/write, let me know.

fredrik-rambris, 2020-11-12:
It is a separately powered 3.5" USB harddrive. I will check the mounted
status next time i notice. But the problem is still there if you have
several USB disks. uuid are consistent, device-names are not always.

fredrik-rambris, 2020-11-12:
Maybe a configuration if one wants deviceId named mounts or filesystem
named mountpoints? There are several fields that are available.

How does one change the read-only filesystem permanently (and survive
updates)?

starbasessd, 2020-11-12:
You don't. It's designed to protect the OS.
You can modify the /boot & /root partitions by turning on 'Debug' in Expert
Settings, but you want to turn it back off after making your mods.
You can also use the method of

mount -o remount,rw /

to make mods, but rebooting resets it back to read-only.
As to surviving an Upgrade, since the upgrade over-writes those 2
partitions, it wouldn't survive, either, you'd have to ree-apply your mods.
An alternative would be to 'compile your own' and not use the built in
upgrade path.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

closed 2022-01-19 by starbasessd