motioneye-project/motioneyeos #2574: multiple zero-length snapshots

zxzzxxzx, 2020-11-13:
[messages.txt](https://github.com/ccrisan/motioneyeos/files/5539030/messages.txt)

### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: (insert your version here, e.g. 20180314).

### Board Model

I am using the following board/model: (insert your board model here, e.g. Raspberry PI 3B+).

### Camera

I am using the following type of camera: (choose from V4L2, MMAL, Network Camera, Fast Network Camera and Simple MJPEG Camera).

My camera model is: (insert your camera model, e.g. Logitech C290).

### Network Connection

My motionEyeOS unit is connected to the network via: (choose your network connection, e.g. WiFi, Ethernet).

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * (e.g. USB WiFi adapter)
 * (e.g. External Hard Disk)

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)

starbasessd, 2020-11-13:
Please attach boot.log.
You didn't answer any of the standard questions, such as version of motionEyeOS, What peripherals, what network, etc. 
I see references to gps base. Are you using motionEye on another OS, like Ubuntu?
Are the photos stored locally on the Pi, or on a NAS?

zxzzxxzx, 2020-11-13:
[boot.log](https://github.com/ccrisan/motioneyeos/files/5539145/boot.log)

Thank you for quik reply. I am using motioneyeOS on SDCard on Raspberry 3B+ about 2 month late and everything worked fine, except for today. Raspberry is connected to local wifi router. Photos are stored locally on SDCard, every week they are deleted. Could it be a problem of broken SDCard, or a failed internet connection? If an electricity is down and then up in a while no zero-length photos present, Raspberry restarts and then all photos are stored as usual.

Thank you.

starbasessd, 2020-11-13:
I was thinking the internet connection, but apparently not.
Might be the SDCard /data partition is corrupted. Need to see the other
logs, dmesg, motioneye.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

zxzzxxzx, 2020-11-13:
is it possible that failed internet connetcion causes motioneyeos to fail and restart? actually, there is a record in logs about it 

[image]

[motioneye.log](https://github.com/ccrisan/motioneyeos/files/5539340/motioneye.log)
[motion.log](https://github.com/ccrisan/motioneyeos/files/5539341/motion.log)
[dmesg.log](https://github.com/ccrisan/motioneyeos/files/5539342/dmesg.log)

anyway, shouldnt it write something about the cause of error in such case?

starbasessd, 2020-11-13:
Yep, depending on why it lost connection, that was the symptom I was
looking for
How large of a  power supply are you using?
Distance from wifi router?
What camera?.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

zxzzxxzx, 2020-11-13:
power supply: regular for Raspberry 1A Power adapter 220V
Distance from wifi router: about 30 cm, very close,
what camera: regular piCamera, something like this

[image]

starbasessd, 2020-11-13:
If you are using a 5v 1A power supply, you are way underpowered for a Pi3,
especially with anything (HDMI, USB anything, CSI PiCam, etc)

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?