motioneye-project/motioneyeos #2575: Which camera it is better for MotionEyeOS with night vision? Pi NoIR Camera or Pi IR Camera

ivanjf1978, 2020-11-15:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: (insert your version here, e.g. 20180314).

### Board Model

I am using the following board/model: (insert your board model here, e.g. Raspberry PI 3B+).

### Camera

I am using the following type of camera: (choose from V4L2, MMAL, Network Camera, Fast Network Camera and Simple MJPEG Camera).

My camera model is: (insert your camera model, e.g. Logitech C290).

### Network Connection

My motionEyeOS unit is connected to the network via: (choose your network connection, e.g. WiFi, Ethernet).

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * (e.g. USB WiFi adapter)
 * (e.g. External Hard Disk)

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)
 
Hi everyone,

I am looking for a good camera with night vision and cheap as well, please could you give me options

I am not very sure the differences between Cameras: 

Pi NoIR Camera and Pi IR Camera, I have read a lot of information on the internet, but, not very clear. I am just looking for a good quality Night vision camera suitable for MotionEyeOs in a good price or cheap

I have tried this camera
5MP 1080P Camera Module 160° Fish Eye + IR Night Vision For Raspberry Pi

https://www.ebay.co.uk/itm/New-5MP-1080P-Camera-Module-160-Fish-Eye-IR-Night-Vision-For-Raspberry-Pi-/254707723403?_trksid=p2047675.l2557&ssPageName=STRK%3AMEBIDX%3AIT&nma=true&si=6JJkRLssGN9DwAUTOMuSEIcPBK4%253D&orig_cvip=true&nordt=true&rt=nc

and I am not very happy with the quality

I have read that this camera could be perfect and exactly what I am looking for, but the price it is not too cheap

Pi NoIR Camera V2

https://www.raspberrypi.org/products/pi-noir-camera-v2/?resellerType=home

Thank in advance for your comments

Regards

Ivan

starbasessd, 2020-11-15:
If you can supply your own IR LED light source, and don't care about
daylight, AND want something designed from the get-go for PIs, the PI NoIR
V2 is high resolution and works well.
If you need wide angle, the chinese camera comes with one, but otherwise
has the same limitations.
I personally would prefer the Sony sensor on the real PiCam V2, but that's
me.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

ivanjf1978, 2020-11-16:
Thank you very much @starbasessd

closed 2022-01-19 by starbasessd