motioneye-project/motioneyeos #2576: RPi 4 4G - Does not load Web Gui and camera feed is lost

soflyj, 2020-11-17:
Hi

I am using a Raspberry Pi 4 4GB and using the OEM power supply. My SD Card has no errors.
Running the latest build, dev20201026.
Connect via the ethernet cable and set a Static IP.
I have connect 3 IP Cameras which work perfectly with no interruptions. (I confirmed this using iSpy on windows)

I can the following problems:
Sometimes the connection will fail, no Web Gui and cannot connect via SSH.
Sometimes the camera feeds will just disconnect and reconnect.

I have installed a fresh copy and rebooted, I still get the above errors.

A quick fix is always disconnect and reconnect the power supply. But I have to do this everytime I get these errors.

Please can you assist?
[dmesg.log](https://github.com/ccrisan/motioneyeos/files/5555628/dmesg.log)
[motion.log](https://github.com/ccrisan/motioneyeos/files/5555629/motion.log)
[motioneye.log](https://github.com/ccrisan/motioneyeos/files/5555630/motioneye.log)

[messages.log](https://github.com/ccrisan/motioneyeos/files/5555626/messages.log)
[boot.log](https://github.com/ccrisan/motioneyeos/files/5555627/boot.log)

starbasessd, 2020-11-17:
My first steps would be to change the ethernet cable to another port on the
switch or router, and/or use a new/different ethernet cable.
Your dmesg log shows the ethernet connection 'bouncing' up and down.
CPU temp is high, 60`C  (and a few reports of 61`C) in the area where
throttling occurs, and exceeds in messages.log)
It is possible the chips that control the ethernet connection have gone
flakey due to the heat on the PI

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

soflyj, 2020-11-17:
Thanks @starbasessd, will try changing the port.
Looking at the motion.log, I notice the codecs were throwing errors so I change the movie format from OMX to V4L2M2M.

Maybe I'll switch to the PI's wifi connection?

starbasessd, 2020-11-17:
Wifi is an option, at least for testing.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-11-23:
Status?

closed 2022-01-19 by starbasessd