motioneye-project/motioneyeos #2577: Control when to record or not record

9mido, 2020-11-18:
It would be nice if there was a way to decide when to record or not record. Or a way to temporarily turn off one camera, certain cameras, or all cameras for a certain period of time. Something like a way to schedule certain times of the day when you want to record and when you do not want to record.  

An example would be when I leave my house and I want to see what happens when I leave. When I am home there is no need to record. Being able to do it immediately like a press of a button when I am about to leave the house or on a schedule like when I am going to work for the day or when I am sleeping. 

Another example is when I am going outside to let the dogs out or do yard work, be able to turn off the recording because there will be a lot of movement. 

You should be able to customize each day of the week and whatever times you want for them and have them be recurring. Then also be able to customize or override days/times in the future for something custom. An example would be during a holiday when you have guests over you do not want to record but the holiday may be a weekday and can interfere with the regular weekly schedule. You would want to override that for that specific holiday date.

starbasessd, 2020-11-18:
Work Schedule is an option to do a fixed schedule (like off Mon-Fri, 8a-6p
and on all other times, or the reverse.
You can do 'pauses' on/off toggles, but not auto-toggle back, via scripting.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

9mido, 2020-11-18:
Cool good to know!

Do you or anyone else have any examples of such scripts? What programming language would they need to be written in?

starbasessd, 2020-11-18:
Here's a discussion I had with another user concerning how to use
start/pause/status commands (basically a web page call)
https://github.com/ccrisan/motioneyeos/issues/2311
There are a few others in the issues lists.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Pedder007, 2020-11-19:
Hi again, haven't been here for a couple of months.

I just set up a complete new installation on a Raspi4B as outlined in (https://github.com/ccrisan/motioneye/issues/1747#issuecomment-727176326)
But running into some trouble last night (>3000 photos due to spider-net) I realized, that the http/shell commands seems not working any longer?
E. g. using: http://192.168.xxx.xxx:7999/3/detection/status
responds:
 _Bad Request 
The server did not understand your request._ 

Has there something changed in the syntax meanwhile?
I already checked multiple times the config with regard to adresses, cam No and also the respective line in the motion.conf, but everything seems to be correct ( the commas in the first lines are only here in to prevent a strange format ;-) )
*************************
,# @enabled on
,# @normal_password XXXX
,# @admin_username XXXX
,# @admin_password XXXX
,# @normal_username XXXX

webcontrol_interface 1
webcontrol_port 7999
setup_mode off
webcontrol_parms 2
webcontrol_localhost off

camera camera-3.conf
camera camera-4.conf

netcam_use_tcp off'
***************************
Does someone have an idea what to check else?
Cheers
Pdder

starbasessd, 2020-11-19:
> I just set up a complete new installation on a Raspi4B as outlined in (ccrisan/motioneye#1747

Which version of motionEyeOS are you using?
Did you try 20200203, 20200606, and / or dev20201026?
20200606 has many issues that are broken.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Pedder007, 2020-11-20:
Hi @starbasessd, thanks for your quick reply.
Where can I check this? The frontend shows as following:

motionEye Version | 0.42.1
Motion Version | 4.2.2
OS Version | Raspbian 10

I also quickly scrolled through montion.conf and also motioneye.conf and couldn't find there anything.

What I'm wondering in addtion about is the fact, that I found two directories in /etc
/motion
/motioneye
As the motion.conf in the directory motioneye has a timestamp from yesterday, that seems to be the working-directory. There I also have the motioneye.con and my two current camera-files.
In the directory motion, the timestamps are from August (set up the PI now roughly 1 week ago) and there are 4 cam-xxx.conf files in, wheras I only do have two cams. May I delete this directory?

starbasessd, 2020-11-20:
You're not running motionEyeOS. You are running motion/motionEye on
RaspberryPiOS.
Which set of instructions did you follow to install motion/motionEye?
Are you using HomeAssistant or other home automation system?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Pedder007, 2020-11-20:
I'm running on a NOOBS Raspbian image.
For the installation I followed these instructions: https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian
As HA I'm using Pimatic, running on another PI4B.
Commands from there look like: wget -q "http://192.168.xxx.xxx:7999/4/detection/pause"
But as written, the error message comes back when using the command in a browser (regadless if chrome, edge or firefox).
... and motion detection doesn't stop any longer. Nor by the shell command, nether by the browser command.

starbasessd, 2020-11-20:
Both cause issues.
Noobs does not use the 'standard' partition arrangement.

I didn't see any good info on integrating motionEye with pimatic, and a few
where people gave up and went to Home Assistant (which also has issues)

If you use a straight RaspberryPiOSfrom here:
https://downloads.raspberrypi.org/raspios_armhf/images/raspios_armhf-2020-08-24/2020-08-20-raspios-buster-armhf.zip
(Desktop)
or
https://downloads.raspberrypi.org/raspios_lite_armhf/images/raspios_lite_armhf-2020-08-24/2020-08-20-raspios-buster-armhf-lite.zip
(Lite)
(you don't need Desktop for motionEye unless you want to view cameras
directly on the Pi Console)
and follow the install instructions here:
https://github.com/ccrisan/motioneye/wiki/%28Install-On-Ubuntu-%2820.04-or-Newer%29
(due to some issues with the upstream repositories and Python 2.7 installs)
You should find that motion/motionEye work as expected.

I am getting ready to test/confirm the pause/start functionality of the
webcall on a clean install of RaspberryPiOS in a few minutes, I'll let you
know.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-11-20:
burn image 2020-08-20 RaspberryPiOS to clean 32 GB SDCard using
Win32DiskImager
copy ssh.txt & wpa_supplicant.conf files to /boot partition
Eject SDCard and install in RP4-4GB (1Gbit Ethernet, HDMI monitor connected)
Power on RPi4
Connect via ssh to ethernet connection
Confirm WiFi connected
Run: sudo apt update && sudo apt upgrade -y [enter]
Run: sudo raspi-config [enter] to change password, video resolution (text
on 4k 43" tv monitor is so small as to be unreadable except at 4" from
screen)
Reboot
Follow install instructions from:
https://github.com/ccrisan/motioneye/wiki/%28Install-On-Ubuntu-%2820.04-or-Newer%29
after confirm up and running at http://[ip_address]:8765, added 2 cameras
edit /etc/motioneye/motion.conf
sudo nano /etc/motioneye/motion.conf [enter]
change line webcontrol_localhost on
to
webcontrol_localhost off
sudo systemctl restart motioneye
now, if you go to
http://[ip_address]:7999/[camera#]/detection/status
it should display the status of 'Paused' or 'Active'
to pause motion detection use:
http://[ip_address]:7999/[camera#]/detection/pause
to re-start motion detection use
http://[ip_address]:7999/[camera#]/detection/start
you can also use commands like 'curl' to do the same thing...
2 discussions:
https://github.com/ccrisan/motioneyeos/issues/2311
https://github.com/ccrisan/motioneyeos/issues/1778

let me know if this helps any.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Pedder007, 2020-11-20:
Well ok, wasn‘t aware of the fact that a NOOBS image could cause issues, as I used it also in my previous setup on the 3B. But maybe that have been also the issues in the beginning of the usage of that one, as we discussed it in the respective thread, you mentioned above.
However I will give your instructions a try in the next days - have to find some free hours. Good that is getting cold currently and no one runs around in the garden 😉
Thanks and will come back here soon!

starbasessd, 2020-11-20:
You _may_ be able to modify an existing install with the section following
the motion/motioneye install (no guarantees)

edit /etc/motioneye/motion.conf
sudo nano /etc/motioneye/motion.conf [enter]
change line webcontrol_localhost on
to
webcontrol_localhost off
sudo systemctl restart motioneye
now, if you go to
http://[ip_address]:7999/[camera#]/detection/status
it should display the status of 'Paused' or 'Active'
to pause motion detection use:
http://[ip_address]:7999/[camera#]/detection/pause
to re-start motion detection use
http://[ip_address]:7999/[camera#]/detection/start
you can also use commands like 'curl' to do the same thing...
2 discussions:
https://github.com/ccrisan/motioneyeos/issues/2311
https://github.com/ccrisan/motioneyeos/issues/1778

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Pedder007, 2020-11-21:
Ok, I just took some time now :-)
Changing the parameter webcontrol_localhost off/on (incl restart) didn't help.
And if I'm right, I already tested that yesterday.

But:
I remembered back to that issue #2311 and what at least brought the problem. At that time, it was a cam which was already configured in montioneye, but was not attached, resp. not in use. When I connected the cam, the problems went away - you may remember to that!?
Now I remembered to the fact, that in this new setup, I first configured 2 cams according to the setup I had before (2 JPEG connected cams, via http:// ...)
To get then a better video quality, I configured the same cams in additon, but now connected via rtmp://...
In sum I had then 4 cams in the system.
As the rtpm connections worked stable, I then deleted the first two cams (cam1, cam2) from the setup and the other ones stayed as cam3 and cam4.
So, now this evening I checked about the browser links (http://...../status) on cam1 and cam2, which worked well, allthough these cams I formerly deleted in the setup. It feelt as I looked at ghost-cams.
Seeing this, I renamed manually the cam-file-names, edited the respective entries in the motion.conf, as well as a few lines in the camxxx.conf, and what do you think??? - it works! :-)
Also my shell commands from Pimatic do work again.

Seeing this it seems for me to be, that motioneye has a problem with cams that might not be 100% configured as it has been thought by design (in regard of the usage of the web commands).
1. Example (#2311): A cam which is configured but not switched on.
2. Example (this here): Deleted Cams, or maybe the fact that the remaining cams are No. 3+4 and not No, 1+2
In both examples the issues didn't start directly, but after a view days - if I remember right.

However I will observe this and come back and report if it stays stable now.
Thanks for the moment.

starbasessd, 2020-11-21:
Actually, I think those are motion issues, not motioneye. the toggle for
pause/start is a motion feature.
I don't know, I just support this stuff the best I can...  :)

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Pedder007, 2020-11-21:
Hi again, you might be right. There seems something to be in the Cam ID handling between motion and motioneye. Sorry for not earlier differentiating between them.

And @starbasessd, to say that right clear: I‘m really happy that you reacted that fast here, as it always helps to discuss respective challenges with one other. For this it is regardless if you or others then exactly give the right hint. At least only by jointly thinking and exchanging about something, generates the right ideas .... 😊👍

What do you think, might I creat an official issue about what we saw here?

starbasessd, 2020-11-21:
As I always say, can't hurt, the worst that can happen is they laugh at you
over at motion... :)

starbasessd, 2022-01-19:
https://github.com/ccrisan/motioneyeos/issues/1778

closed 2022-01-19 by starbasessd