motioneye-project/motioneyeos #2578: [Support] RPi Zero W fresh install problems to connect to WiFi

ovz93br43v7, 2020-11-19:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: (insert your version here, e.g. 20180314).
motioneyeos-raspberrypi-20200606.img

### Board Model
I am using the following board/model: Raspberry Pi Zero W v1.1

### Camera
I am using the following type of camera: Raspberry Pi NoIR Camera V2

### Network Connection

My motionEyeOS unit is connected to the network via: WiFi

### Peripherals

I am using the following peripherals that I consider relevant to this issue: nothing

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)
[boot.log](https://github.com/ccrisan/motioneyeos/files/5566288/boot.log)
 

I tried three ways of installation:
- writing image manually to sd card with imagewriter and adding `wpa_supplicant.conf` manually according wiki [https://github.com/ccrisan/thingos/wiki/wpa_supplicant.conf](https://github.com/ccrisan/thingos/wiki/wpa_supplicant.conf)
- writing image manually to sd card with imagewriter and adding `wpa_supplicant.conf` manually according guide in this issue (extended in comparison to wiki with country etc) --> https://github.com/ccrisan/motioneyeos/issues/2544#issuecomment-706698106
- `writeimage.sh` with providing WiFi key and password

But I wasn't able to get a connection. As far as I understand the log file `wpa_supplicant.conf` is correct, WiFi key and password was copy and paste.
My only idea is now that this problem is caused by some special characters in my WiFi password....
Or do you have another idea?

starbasessd, 2020-11-19:
>    in comparison to wiki with country etc) --> #2544 (comment)

2 thoughts immediately come to mind:
1] Install RaspberryPiOS 20200820 with the same process (copy
wpa_supplicant.conf file to /boot partition] and see if that works.
2] use dev20201026 or 20190911 with the same process and see if either of
them work for you.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

ovz93br43v7, 2020-11-19:
Ok I will try that, just one question: 
How is the wpa_supplicant.conf file correct:

update_config=1
ctrl_interface=/var/run/wpa_supplicant

network={
    ssid="mynet"
    psk="mynet1234"
}

or

ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
country=US

network={
    ssid="SSID"
    psk="passphrase"
key_mgmt=WPA-PSK
}

starbasessd, 2020-11-19:
I use this format:
-----
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
country=US

network={
    ssid="ssid"
    psk="password"
key_mgmt=WPA-PSK
}
-----
Latest kernels and wpa requires country code or it will shut down wifi, and
I use the same setup whether for RaspberryPiOS or motionEyeOS.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

ovz93br43v7, 2020-11-20:
I made some trails with the images you mentioned.
When I put the `wpa_supplicant` file manually to `/boot/wpa_supplicant.conf` then I don't find any entry in `boot.log` about any try to access the WiFi.
I tried several combinations:
* your file configured for my case (DE, my SSID, my password)
* tried modified `wpa_supplicant.conf` versions according to several guides or other problem solving requests: [https://wiki.ubuntuusers.de/WLAN/wpa_supplicant/](https://wiki.ubuntuusers.de/WLAN/wpa_supplicant/), [https://www.raspberrypi.org/forums/viewtopic.php?t=204756](https://www.raspberrypi.org/forums/viewtopic.php?t=204756), [https://www.raspberrypi.org/forums/viewtopic.php?t=201850](https://www.raspberrypi.org/forums/viewtopic.php?t=201850)
* because I assume problems with special characters in my password I tried to translate it with wpa-passphrase to a hex coded password --> [https://fecifi.wordpress.com/2018/06/15/wpa-passphrase-with-special-characters/](https://fecifi.wordpress.com/2018/06/15/wpa-passphrase-with-special-characters/) but it did't changed anything
* because I assume problems with special characters in my password I tried to use a mobile hot-spot from my mobile phone with extra simple SSID and extra simple password, laying next to the pi

When I use the `writeimage.sh` from [https://github.com/ccrisan/motioneyeos/wiki/Installation](https://github.com/ccrisan/motioneyeos/wiki/Installation) then I find in the boot.log at least a try to access the WiFi:
```
Starting wpa_supplicant: failed (brcmfmac)
Configuring wireless network: no link
```

When I search for `brcmfmac` then I see that this is the wireless driver but I didn't find out until now what exactly failed. That would be useful :-/

When I use the `writeimage.sh` script I recognized that:
* I don't find at `/boot/wpa_supplicant.conf` before the first boot to possibly detect a difference to my manual created `wpa_supplicant.conf`
* `writeimage.sh` don't request any country code but I can't add because the file `/boot/wpa_supplicant.conf` wasn't created
*  but at `/etc/wpa_supplicant.conf` was created by the `writeimage.sh`script but only with following content:

```
update_config=1
ctrl_interface=/var/run/wpa_supplicant

network={
    scan_ssid=1
}
```

ovz93br43v7, 2020-11-20:
After several trails I made it working.
My solution:
Don't put the `wpa_supplicant.conf` in `/boot/wpa_supplicant.conf`, instead put it directly into `/etc/wpa_supplicant.conf`

My file looks like this:
```
update_config=1
ctrl_interface=/var/run/wpa_supplicant GROUP=netdev
country=XX

network={
    ssid="XXX"
    psk="XXX"
    key_mgmt=WPA-PSK
}
```

closed 2020-11-20 by ovz93br43v7