motioneye-project/motioneyeos #2579: add tuya based camera to motion eye os

jarmodb, 2020-11-19:
is it possible to add an tuya based camera to my excisting motion eye os setup?

### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 20200606

### Board Model

I am using the following board/model: raspberry pi4 

### Camera

I am using the following type of camera: Tuya based ip outdoor camera https://www.action.com/nl-nl/p/dummy-3003011/

### Network Connection

My motionEyeOS unit is connected to the network via: ethernet.

starbasessd, 2020-11-19:
Not enough info on their site to determine.
Did not see the brand on Connecting iSpy to IP Cameras
<https://www.ispyconnect.com/sources.aspx>to know if they have a valid URL
to connect to.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

jarmodb, 2020-11-19:
unfortunately there isn't more info online about these cams... I also can't reach the manufacturer

starbasessd, 2020-11-23:
Feel free to reopen or open a new issue if/when you get more info on the camera.

closed 2020-11-23 by starbasessd