motioneye-project/motioneyeos #2582: motioneyeos frequently unreachable

BETLOG, 2020-11-21:
My motioneyeos camera is often unreachable when i try to load it's interface (or live video) in chrome.

At first I suspected it was rebooting, but now I don't know if thats a good assumption.
So rather than make any assumptions about the cause I'll just describe the setup.

I have tried downgrading the video resolution and framerate, but all trials seem to randomly 'disconnect' equally.

I am making a new issue because the existing issues assume the pi is rebooting, or have other associated issues that make assumptions that I want to avoid making. I also I want to include full logs/descriptions in a way that does not distract from the other issue's points.

The pi is mounted into the front of a small commercial pizero case, the back excluded and i have a heatsink on the GPU, and the whole unit is mounted vertically. So it gets fairly good airflow, and despite high ambient temperatures at this time of year (currently 15-32C-ish) it is not in the sun. 
Power is via a small 500Ma PWM wallsocket converter, along a rather long 5m USB cable. I have previously run larger loads (full size Pis and accessories/tablets,etc) on this cable, so although it might be problematic I don't suspect it.
For the first few days of operation I had a power meter attached inline with the USB at the power source end, and it consistently showed ~220ma of current draw, so the current PSU should be very adequate. The power meter is no longer attached.

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.
I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version
20200606

### Board Model
Raspberry Pi Zero W

### Camera
MMAL - Raspberry Pi Zero Camera

### Network Connection

WiFi
...to OpenWRT router ~6m away
MediaTek MT7621 ver:1 eco:3
OpenWrt 18.06-SNAPSHOT r7835-89808e211c / LuCI openwrt-18.06 branch (git-19.218.48077-4468345)
Kernel Version	4.14.137

### Peripherals
none

### Log Files

I consider the following log files relevant to this issue:

[image]

[image]

This would have been much easier to do if you stated HOW to obtain the logs. ie:
ssh root@pizeroeye1 cat /data/log/motioneye.log /data/log/motion.log

[2020-11-21--19-15-48_motioneye.log](https://github.com/ccrisan/motioneyeos/files/5577311/2020-11-21--19-15-48_motioneye.log)
 
[2020-11-21--19-24-10_operwrt.log](https://github.com/ccrisan/motioneyeos/files/5577321/2020-11-21--19-24-10_operwrt.log)

starbasessd, 2020-11-21:
motionEyeOS 20200606 has many known issues
I would first try 20190911, 20200203, or dev20201026.
To get the logs in the WebGUI, you go to Expert Settings, and the big 5
have links for doewnloading.
The boot.log, messages log would also be helpful.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-11-21:
Since you are able to ssh into the PiZeroW, it would also be helpful to get
the output of the following commands, especially while / after seeing the
issues but before a reboot.:
vcgencmd get_throttled
vcgencmd measure_temp

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

BETLOG, 2020-11-22:
[root@meye-d5ff8645 ~]# vcgencmd get_throttled
throttled=0x0
[root@meye-d5ff8645 ~]# vcgencmd measure_temp
temp=57.3'C
As can be seen on the screenshot of the GUI (above) that is a fairly typical temp. I don't think it is ever overheating.

I just realised I accidentally had it switched off for most of the day, so i doubt the following logs will reveal anything.
...and I plan to try dev20201026 right now, so I'll repost here if that solves the problem.

[boot.log](https://github.com/ccrisan/motioneyeos/files/5579284/boot.log)

[dmesg.log](https://github.com/ccrisan/motioneyeos/files/5579285/dmesg.log)

BETLOG, 2020-11-23:
dev20201026 was a bit of a farce
At first boot the configuration menus at  left were greyed out. Finally I deleted the autodetected camera and added another manually, and was then was able to enter info into the menus, but as soon as i hit apply the menus were greyed out again.
Far too broken at a basic level to persist with.

So now I'm using 20190911, and aside from the mask editor not showing the camera image (it'ts just red squares and black squares) forcing e to guess the mask locations while saving/editing numerous times... it seems ok so far. Will continue to monitor for dropouts.

closed 2021-03-04 by starbasessd