motioneye-project/motioneyeos #2583: writeimage.sh -n should allow for name=MAC (with colons)

BETLOG, 2020-11-21:
suggestion
re: https://github.com/ccrisan/motioneyeos/wiki/Installation
...the instructions for wifi ssid/pass
You can preconfigure a wireless network connection:
 ./writeimage.sh -d /dev/mmcblk0 -i "/path/to/motioneyeos.img" -n 'yournet:yourkey'

Given that MAC addresses contain colons (:), and BSSID's are usually MAC's, and I use the BSSID as my SSID (i name the wifi the same as it's MAC), the separator character in your writeimage.sh argument disallows me from doing this.
I tried numerous variations of single and double quotes, but ultimately it was easier to just edit it manually to repair wpa_supplicant after the script smashed it's name/ssid fields to pieces.

Suggest:
Different separator character (like a space), and specify that ssid and psk both be separately quoted
ie: 
-n 'yournet' 'yourkey'   or    -n "yournet" "yourkey"
eg:
-n '00:11:22:33:44:55:66' 'specific radio passphrase'   or    -n ":11:22:33:44:55:66" "2.5 and 5Ghz radio passphrase"

starbasessd, 2020-11-21:
I found:
https://bttn.freshdesk.com/support/solutions/articles/5000529141-known-issue-in-old-firmware-wi-fi-ssid-and-password-cannot-have-spaces-or-special-characters
and many references to what are acceptable characters and not across many
OSs, and WiFi software. In you case, if it were me, I'd use a different
separator, or no sepaerator.
AA-00-BB-11-CC-22
or
DD33EE44ff55

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

BETLOG, 2020-11-22:
The only thing that has a problem with it is motioneye, but you raise an excellent point. Next time I go through the setup process I'll try a different syntax.
It's just a little tedious that motioneye messes up colons, and yet every wifi router I have ever owned gives MAC addresses with colons in them.

closed 2020-11-22 by BETLOG

BETLOG, 2020-11-22:
I should also mention that it bothers me to be entering passphrases into a command line, so that I then need to remember to precede the command with a space or delete the line from .bash_history later.
It would be better if by invoking the -n parameter with no arguments the script then prompted for both ssid and psk separately. This would also work around the colon issue.

starbasessd, 2020-11-22:
You could also create a wpa_supplicant.conf file with your SSID and a coded
passphrase (using wpa_passphrase https://linux.die.net/man/8/wpa_passphrase)
that goes into the /boot partition
and eliminate the need for command line entries at all.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

BETLOG, 2020-11-23:
> wpa_passphrase

I was actually wondering about that. I assumed that the commands needed to be invoked on the target hardware because crypto, but nothing I'm seeing actually specifies that.
So yeah, thats probably the ideal solution.