motioneye-project/motioneyeos #2586: ShellCode [Expl] threat detected

cesigen, 2020-11-24:
Been using motioneyeos for a while now but today I got the following warning from Avast Antivirus:

"We've safely aborted connection on 192.168.x.x because it was infected with Win32:ShellCode [Expl]"
Threat name: Win32:ShellCode [Expl]
URL: http://192.168.x.x/picture/2/current?_=160xxxxxxx&_username=admin&_signature=6b8471xxxxxxxxxxxxxxxxxxxxxxxxxxxx

This happened when accessing the motioneyeos dashboard (installed on a raspberry pi 3) on my computer. So has my raspberry pi been infected with malware somehow? it only has motioneyeos installed on with secure passwords

[image]

### motionEyeOS Version

I am running motionEyeOS version:

motionEye Version 	0.42.1
Motion Version 	4.2.2+gitUNKNOWN
OS Version 	motionEyeOS 20200606

### Board Model

I am using the following board/model: Raspberry Pi 3

### Camera

Vstarcam c7824wip

### Network Connection

My motionEyeOS unit is connected to the network via: WiFi

starbasessd, 2020-11-24:
If you reboot the Pi and try to connect again, do you get the same error?
Most of motionEyeOS is read only, and should not be 'infectable'.
If you re-flash, do you get the same error?
I'd bet that it's a false positive, due to an Avast update.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

cesigen, 2020-11-24:
I've turned it off because I was worried it would try to spread it to other computers/devices on the network. I did boot it up for a minute to grab the motioneyeos version and no warnings appeared this time. I do hope it is a false positive and will wait to see if anyone else has experienced it the latest avast update

starbasessd, 2020-11-24:
https://www.zdnet.com/article/security-firms-dismiss-english-shellcode-threat/
Most 'attacks' are a half-decade old. Since it would be an 'english' (clear
text) trigger, and a short one, as the researchers say, the odds of it
appearing randomly in unexpected places is higher than it being real.
If your PI is not exposed to the internet, it is set to default read only,
and you don't get more than random reports, I doubt it's real.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

closed 2021-03-04 by starbasessd