motioneye-project/motioneyeos #2587: Run a command not executing arecord

computersandstuff, 2020-11-24:
Hello,
Motioneye works great for me, but I wanted audio so I tried get an audio recording from a USB mic on my raspi.
I am running raspbian lite (no UI) with docker & motioneye. I access the pi from SSH.
I followed some instructions [here](https://medium.com/@elkentaro/nothing-to-see-here-the-not-so-charger-62a51e3aab22)
I created a script to be executed in the directory /home/pi/scripts. It simply recorded an audio file after being triggered from a motion detection from "run a command" under motion detection. After some trouble I did the following:

**(Each and every time I tested the command I put into "run a command" from the command line.)**
**(I was able to get recordings by running the scripts or commands from CLI, but not the "run a command" box)**

- Changed the script location from /home/pi/scripts to /etc/motioneye

- Changed parameters of the arecord

- Changed file permissions of the script, finally changed them to 777

- Tried using the direct path of arecord /usr/bin/arecord

- Simply put the command "arecord" into the "run a command" box instead of running a script. (I also tested this from the CLI and it worked there)

- Changed the permissions of the command arecord

- Changed the sudoers file to allow less restricitons

- Tried a few other commands such as mkdir (which did work, it created the directory  /etc/motioneye/test )

It seems that it does not want to execute arecord for some reason. Weather it is being executed from a script or directly. Like I said before I ran these from the command line and they worked just fine, recorded a file and I could play it back.

I also cannot locate the logs for motioneye as they are not under /var/logs to see what the problem might be (I can put this is another thread if you would like)

Help would be much appreciated!

starbasessd, 2020-11-24:
The docker image has issues passing data / triggers / etc in and out of the
docker container. May I suggest running RaspbianOS Lite or Ubuntu Server
20.04 and install motion/motionEye from here:
https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian
or
https://github.com/ccrisan/motioneye/wiki/%28Install-On-Ubuntu-%2820.04-or-Newer%29
Also, you may want to take this to
https://github.com/ccrisan/motioneye/issues as more people who use
motion/motionEye in various setups (not motionEyeOS) and you might get more
suggestions...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

computersandstuff, 2020-11-24:
So I installed [this](https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian) on raspberry pi as you said and it worked! Now it can execute scripts. I assume the docker version was the issue. Thanks!

starbasessd, 2020-11-24:
Probably.
Glad to help.
Don't forget to close the incident in a few days, or close it and if you
need to re-open it, feel free to do so in the motion/motionEye github...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

closed 2020-11-24 by computersandstuff