motioneye-project/motioneyeos #2588: WiFi camera as Extender for other cameras / devices - Feature request

pineapplee85, 2020-11-24:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: (20200606).

### Board Model

I am using the following board/model: (Rapberry Pi Original (External WiFi, ras pi 2b and 4 could also be used if need be)).

### Camera

I am using the following type of camera: (V4L2).

My camera model is: (Logitech C110).

### Network Connection

My motionEyeOS unit is connected to the network via: (Ethernet).

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

(USB WiFi adapter)

Hi there,

Love the project. Has helped me a lot. 
I currently have one wired connection out to my garage but the wifi signal does not reach far enough to where I want to put another camera.
Could we have an option where a wired Pi can use the internal (if fitted) or external PnP USB WiFi dongle to broadcast an SSID to link another camera to? (Like a wired WiFi access point)
This may be too hardware intense but thought it better to ask.

Thanks for your great work!
Ethan

Ps. If I did something wrong with this feature request I'm sorry, I'm fairly new to github.

starbasessd, 2020-11-24:
A couple of ideas:
RPi project on AdaFruit.com turns RPis into Access Points.

https://learn.adafruit.com/setting-up-a-raspberry-pi-as-a-wifi-access-point
Cheap 5 port 100MBit switch <$20US, seen them even cheaper.

https://www.amazon.com/s?k=5+port+switch&s=price-asc-rank&qid=1606253435&ref=sr_st_price-asc-rank
My D-Link DCS-5020L cameras have the feature built in.

Good request. It might be better on a Pi4, Pi3B+ for me barely does the AP
by itself when used by several devices (phones and tablets...) I doubt any
of the Pi1s, Pi2s, or PiZeros have enough horsepower to do it justice...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

pineapplee85, 2020-11-27:
Makes sense, I thought about an ethernet switch but it won't really fir where I am trying to put the pi.

I wasn't sure if the other pi's would be able to handle it. I was thinking maybe a maximum of 2 or 3 other motion eye cameras only to limit the power required?

starbasessd, 2020-11-27:
How about a cheap b/g/n wifi router (like something like this:
https://www.amazon.com/TP-Link-N300-Wireless-Wi-Fi-Router-TL-WR841N/dp/B001FWYGJS/
Using the Pi as an ap/extender, take into consideration of wifi bandwidth,
and the fact you can't use 2 wifi's as a bridge on the same Pi.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

pineapplee85, 2020-12-14:
Hi,
I am using the pi on ethernet so it could just be a bridge from ethernet to wifi for one other motioneye camera.
I understand there are limits to the power of such devices, it would just be a handy to have.
Thanks!

closed 2020-12-14 by pineapplee85