motioneye-project/motioneyeos #2589: I can not use MotioneyeOS mobile APP to see my network cameras

ivanjf1978, 2020-11-26:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: (insert your version here, e.g. 20180314).

### Board Model

I am using the following board/model: (insert your board model here, e.g. Raspberry PI 3B+).

### Camera

I am using the following type of camera: (choose from V4L2, MMAL, Network Camera, Fast Network Camera and Simple MJPEG Camera).

My camera model is: (insert your camera model, e.g. Logitech C290).

### Network Connection

My motionEyeOS unit is connected to the network via: (choose your network connection, e.g. WiFi, Ethernet).

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * (e.g. USB WiFi adapter)
 * (e.g. External Hard Disk)

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)
 
I tried to use Motioneyeos in my mobile phone to watch my network cameras with MotionEyeOS outside my network (without connected with the same wifi network) and I can not use it. I tried to do port fowarding in my router but not working. 

I am using android 8.0.0

I am using Raspberry Pi 4B 4GB Ram

I am using MotioneyeOS 20200606

Please any advice? What I have to do to watch my motioneyeos outside my network with my mobile phone?

Thanks

Ivan

starbasessd, 2020-11-26:
Try the dev20201026 for the Pi4, there are issues with 20200606. I will try
to reproduce later (Thanksgiving in US...)
Make sure your outside port is something like 8080, pointing to the Pi4 IP
address and port 80 if using motionEyeOS, or port 8765 if using motionEye
on another OS

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

ivanjf1978, 2020-11-27:
@starbasessd Thank you very much for replied to me,

I am not IT expert, please, when you have time could you tell me step by step how can I do that?
Pointing to the Pi4 IP address and port 80 using motionEyeOS

Could you please help me with all items that we have to fill up on the APP? see picture 
[image]

Thank you very much

Best regards

Ivan

ivanjf1978, 2020-11-27:
Fixed my friend....finally

I can see the cameras from my network our side my network

Thank you very much

Best regards

Ivan

starbasessd, 2020-11-27:
I don't support modem / routers that people use for internet access, there
are simply too many of them.
When you get the port forwarding set up, you don't need any 'app' you can
use the android's default web browser. There is no need to give up any
personal data to some programmer / company. The Android app is not part of
motioneye / motioneyeOS.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-11-27:
Close the issue ticket, please.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

ivanjf1978, 2020-11-27:
I tried to connect it from my mobile phone date and take too much time and shows this. But, if I connect my phone to my wifi (local Network) I can see the cameras. But if I use the internet from my mobile phone data I can not see motioneyeOS
[image]

ivanjf1978, 2020-11-27:
> I don't support modem / routers that people use for internet access, there are simply too many of them. When you get the port forwarding set up, you don't need any 'app' you can use the android's default web browser. There is no need to give up any personal data to some programmer / company. The Android app is not part of motioneye / motioneyeOS.
> […](#)
> -- Thanks Kevin Shumaker Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel> N38° 19' 56.52" W85° 45' 8.56" Semper Gumby “Don't tell people how to do things. Tell them what to do and let them surprise you with their results.” - G.S. Patton, Gen. USA Ethics are what we do when no one else is looking. Quis custodiet ipsos custodes? “There is no end to the good you can do if you don’t care who gets the credit.” - C Powell You know we're sitting on four million pounds of fuel, one nuclear weapon and a thing that has 270,000 moving parts built by the lowest bidder. Makes you feel good, doesn't it?

That`s mean the APP is unuseful? why this APP exist?

starbasessd, 2020-11-27:
Someone is trying to get all the other data the app requires access to.

closed 2021-03-04 by starbasessd