motioneye-project/motioneyeos #2591: [Question]does motioneye support auto ISO ?

VoeGalore, 2020-11-28:
Hi,
I tried to change raspimjpeg.conf parameters ('iso 0' or 'iso auto') and didn't work, no image output unless I manually set a value to 100-800 again.
I'm using IR-CUT cameras as outdoor CCTV and lighting condition varies over the day, would be great to make it work so I get less noisy feeds during daytime.
Thanks for any help!

VG

starbasessd, 2020-11-28:
Are you using motion/motioneye or motionEyeOS?
If motion/motionEye which install instruction setup?
If motionEyeOS which version? 20190911, 20200203, 20200606, or dev20201020?
Which Pi?
Are you using Fast Network Camera in motionEyeOS?

VoeGalore, 2020-11-28:
> Are you using motion/motioneye or motionEyeOS?
> If motion/motionEye which install instruction setup?
> If motionEyeOS which version? 20190911, 20200203, 20200606, or dev20201020?
> Which Pi?
> Are you using Fast Network Camera in motionEyeOS?

I'm using motioneyeos.
I used the image from Crisan's github releases, 20190911 at first and it's updated to 20200606.
Device in question is a raspberry pi 3B, but I have a range of other RPi models in the CCTV set up too, from zero W to 3A+/3B+).
I'm using fast network camera mode in all the cameras, including the one in question.

starbasessd, 2020-11-28:
Can you try dev20201026, please?

VoeGalore, 2020-11-28:
> Can you try dev20201026, please?

I just update from the web portal and looks like my control panel is now covered in grey layer, let me do write a new SD card and climb the ladder.

starbasessd, 2020-11-28:
Since 20190911 you can save the settings (backup) and then do restores to newer images...

starbasessd, 2020-11-28:
And I do suggest the dev20201026 image, there a re quite a few issues with motion/ffmpeg and such in 20200606.

VoeGalore, 2020-11-28:
> And I do suggest the dev20201026 image, there a re quite a few issues with motion/ffmpeg and such in 20200606.

I did a fresh installation of dev20201026, looks like as soon as I turn it into Fast Network Camera mode the control interface would grey out and render me a duck. I can still SSH into it but I'm really not good at that.

Shall I take it to a new thread and report this then? I can't collect log in web portal from this state now that it is locked out, which file should I extract for debugging?
[image]

starbasessd, 2020-11-28:
creating a new setup, hold on.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-11-28:
Just did a PiZero with PiCam, installed dev20201026. Turned on Fast Network
Camera and rebooted.
Works fine. You may need to clear your Browser Cache if you had an Admin
Password set.
I will escalate this as a 'feature request / enhancement' as there is no
'auto iso' in the WebGUI.
I took a quick look over at https://github.com/Motion-Project/motion/issues
in both open and closed issues, and I see where they were trying to do
something similar to your request, but unable to find what their ultimate
solution was, if any.
You may want to look at
https://motion-project.github.io/motion_config.html#auto_brightness
There are a lot of motion settings that may be of help.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

VoeGalore, 2020-11-28:
Big thanks man. I look forward to the next release.

I'll revert to 20200606 since it works. auto ISO is a minor thing that might help with noise, I keep it at ISO800 usually and get decent visibility overall.

It's just that the controls in raspimjpeg got me thinking a bit. I ran other RPi cameras too and sometimes configuration allows me to turn on auto iso, specific projects include:

https://elinux.org/RPi-Cam-Web-Interface (gives you a suite of options in web interface, including ISO auto toggle, I use it on an HQ camera attached to a microscope)

http://www.davidhunt.ie/raspberry-pi-zero-with-pi-camera-as-usb-webcam/ ,or
https://github.com/showmewebcam/showmewebcam
(makes RPi zero into a usb webcam, tweaks are done directly in command line via '/usr/bin/v4l2-ctl', there are iso_sensitivity and iso_sensitivity_auto options)

It's quite beyond my own capacity to figure out the details and implement fix or change, I hope the information would be useful to you and others.

And yes I just got in control panel thru a private browser session, so clearing cache should also work.

ETE-Design, 2020-11-30:
@starbasessd Will there be and new Version Released soon? Seems like development stopped in September?

starbasessd, 2020-11-30:
There was a dev20201026 (end of October), and as @ccrisan sees things fixed
upstream with motion (a whole different can-o-worms) (motionEye and
motionEyeOS are a front end to motion) he tries to incorporate them into
newer version of motionEye & motionEyeOS.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

closed 2021-03-04 by starbasessd