motioneye-project/motioneyeos #2592: Red outline mean motion or recording

ITCMD, 2020-11-29:
Hi there, I'm using motioneyeos on a raspberry pi zero with movies set to continuous recording. I have no need for motion detection so I set the threshold all the way at 20% and set max frames to 1. 

However, I am still seeing a red outline over the camera. Does this mean it is recording (as it should be) or that it is detecting motion (which it should not be)?

starbasessd, 2020-11-29:
You can disable motion detection by clicking the on/off button on the left
of the Motion Detection title bar.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

ITCMD, 2020-12-02:
Huh, that's a neat feature. Thanks.

You know it'd be nice if it was put somewhere obvious and visible, like to the left of the title bar.

Im stupid. Thanks again.

starbasessd, 2020-12-02:
We've all been there. Don't forget to close...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

ITCMD, 2020-12-03:
Thanks and will do :)

closed 2020-12-03 by ITCMD