motioneye-project/motioneyeos #2593: How to record hour long movies continuously?

parthpatel05, 2020-11-29:
I am trying to get it to record hour long movies so that I could watch them after the hour is over, but it only records one movie that is an hour long then won't record another hour.

starbasessd, 2020-11-29:
Try dev20201026.
Also please answer the system questions, they help to diagnose if they
are known issues, or new issues, and allow us to try to reproduce.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

parthpatel05, 2020-11-29:
### motionEyeOS Version

I am running motionEyeOS version: motionEyeOS 20200606

### Board Model

I am using the following board/model: Raspberry pi 4 1gb

### Camera

I am using the following type of camera: Fast Network Camera and Simple MJPEG Camera

My camera model is: Pi high quality camera

### Network Connection

My motionEyeOS unit is connected to the network via: wifi for cam, and ethernet for hub

starbasessd, 2020-11-29:
There are definite issues with 20200606 for all model boards. There are boot issues with 20200203 and earlier for Pi4 (you have to have the 'right' boot loader to work
Please try dev20201026 at least for the Pi4, but you can use 20190911 for Pi3B+ and older models. Please be sure to grab the correct version for the board you are using.

closed 2021-03-04 by starbasessd