motioneye-project/motioneyeos #2594: What is the ssh password?

Aspiringpentester, 2020-11-29:
I am using ssh through the windows subsystem for linux, and when I try to log into the pi with ssh it doesnt accept any password I use.  I have tried the admin password, the user password, leaving it blank, and "password." Is there something I'm missing?

starbasessd, 2020-11-29:
If you didn't change it in the WebGUI, the username is 'admin' << all lower
case
and the password is blank. If I use the ssh in Linux subsystem, it doesn't
even prompt me for a password on a clean install with no changes.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-11-29:
I use
ssh admin@[ipaddress] [enter]
If it is already registered, it goes straight in

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Aspiringpentester, 2020-11-30:
I guess I just didn't know how to use ssh lol. I hadn't been setting the username when I ran the command, so no password worked because the username I was using didnt _have_ a password.

starbasessd, 2020-11-30:
No problem. Please close if this resolved your issue.

closed 2020-11-30 by Aspiringpentester