motioneye-project/motioneyeos #2595: Xfinity XFi router issues, not able to view camera outside of home network

Goldglv, 2020-11-30:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: tried DEV20201026, 20200606, 20200203, 2090911.

### Board Model

I am using the following board/model: Pi 3B+.

### Camera

I am using the following type of camera: (choose from V4L2, MMAL, Network Camera, Fast Network Camera and Simple MJPEG Camera).

My camera model is: Raspberry Pi Camera Module V2.

### Network Connection

My motionEyeOS unit is connected to the network via: tried Wifi as well as Ethernet.

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * (e.g. USB WiFi adapter)
 * (e.g. External Hard Disk)

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)
 
Since upgrading to a new Xfinity XFi modem, I'm unable to view MotionEye outside of my home network (through the App or the Web).  Tried above versions of MotionEye and nothing works.  Can confirm Port Forwarding is set up correctly on the router.  Not sure if this is a ipv4/ipv6 issue, heard new Xfinity routers won't allow port forwarding on anything other than ipv4.

Been literally trying to figure this out for months......any help would be appreciated.

starbasessd, 2020-11-30:
Standard port forwarding would be outside port 80 on your Internet IP address to internal IP address on port 80.
Example would be 237.238.239.241 port 8080 external to 
192.168.0.50 port 80 on the Pi
Quick Google found this:
https://www.xfinity.com/support/articles/xfi-port-forwarding
You can contact XFinity support to kill IPV6 if it is possible.
You aren't the only one having issues with port forwarding on those modems. Can you force a static IP address (either through DHCP on the modem/router, or manually on the PI?

Goldglv, 2020-11-30:
Yes, I have already forced a static IP address, currently port forwarding to 8081.  Re-read that link, thank you, I do see the PiCamera in my Xfinity Connected Devices and the Pi is a static IP set up with Port Forwarding.

Like I mentioned, seemed to only happen after I got the new modem from Xfinity, anything else I can try?  I love MotionEye and would hate to get rid of it but I'm at my wits end trying to figure this out....

I'm guessing MotionEye is using IPV6 now?  Is that the issue with XFi router?

starbasessd, 2020-11-30:
Hate to ask this, but do you have streaming turned on for Camera1 on the
Pi? You are not looking at the WebGUI page on 8081...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Goldglv, 2020-11-30:
[image]

starbasessd, 2020-11-30:
I use Spectrum for my ISP.
I have a business class cable modem set to pass-through
I have an Ubiquity Router where I control everything. Internally I have
Gigabit or better everything, with various brands of switches, and fiber
and copper interconnectivity.
I have my motionEyeOS or motionEye hubs set with port forwarding.
I have IPV4 enabled, IPV6 disabled internally.
I have several Wireless APs of different brands, and one Raspberry Pi3B+
set up as an emergency backup if the rest of the network takes a dump.
I do not have issues, other than every few months I have to call Spectrum
and have their techs set the modem back to 'pass through', as, in their
efforts to make my life easier, they make my life more difficult.
If I were in your situation, I would either contact XFinity support, or try
something like my setup. Google points to an issue with the Xfi router.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Goldglv, 2020-11-30:
So you think trying to disable IPV6 is worth a try?

starbasessd, 2020-12-02:
Can't hurt, unless something you are running is using IPV6.

Goldglv, 2020-12-03:
The closer I looked at the settings, I see nothing for IPV6, only that IPV4 is enabled.  So I'm guessing that's not the issue, all port forwarding is set up correctly, Pi has a static IP, anything else I can look at to try and figure out why I can't see MotionEye outside my home network?

starbasessd, 2020-12-03:
https://www.xfinity.com/support/contact-us

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

lanche11, 2020-12-04:
Go to MEOS Expert Settings assign a HTTP: port and Apply
Go to https://internet.xfinity.com/connect/network and select Advance Settings
Select Port Fowarding , select ADD at bottom of page
Select your Device from the list 
On the Pop Up Add your port number
Your router has an IP Addresss XX.XXX.XXX.XXX:your port number

closed 2021-03-04 by starbasessd