motioneye-project/motioneyeos #2596: Saves to wrong google drive folder

Jaizan, 2020-12-01:
motionEye Version | 0.42.1
Motion Version | 4.2.2+gitUNKNOWN
OS Version | motionEyeOS 20200606
Raspberry Pi4B
PS2 camera
Wired LAN connection

I had this all working OK for about a year, then had an error whilst trying to back up the SD card, so I did a fresh install on a new SD.

**With the new install, everything is OK, except when saving to google drive, it insists on creating a new folder, with an obscure and illogical folder naming structure.   The pictures end up on a sub-folder about 7 levels down !**
  
On the last install, all the pictures went to the intended google drive folder and it created a new sub folder for every day.   Very logical.   That's what I want with the new install. 

I've tried several times to set the new installation up, with new folder names and so on, but it insists upon doing it's own thing.
What am I doing wrong ?

starbasessd, 2020-12-02:
Try the dev20201026 release and let us know.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Jaizan, 2020-12-02:
Hi 
Thank you.
I tried the new dev20201026 release.     Whilst it creates a folder for each day, it still does not put the pictures in the folder I selected.
It creates a folder within Drive named https:
Below that is one called:drive.google.com
Below that one called drive
And so on.    Several levels down I finally get to the pictures.
It ignores my folder MotionEye2020

Whatever path I paste in the MotionEye location box is created as a folder structure within the google drive.
So rather than pasting the full location, I then tried pasting the location it generates only   e.g. **5pPXGphfX******************

It then creates a new top level folder with the name 5pPXGphfX****************   and puts the pictures in that.   This is acceptable, but not ideal, as I would like to use a folder "MotionEye2020"
If I put MotionEye2020 as the location in MotionEye, of course it cannot trace the folder.

I get the same issue using a different google account.   Also, re-loading the original software which worked fine previously, same issue.

Also, is dev20201026 stable ?    I'm going on holiday for a few weeks, so hope it works as well as the one I've been using for the last year.

Jeff

starbasessd, 2020-12-02:
To answer your last question, first, I've been running it on a PiZero
since a few days after it came out, and it's been working fine except when
I switch out the card to do other testing.
As to settings, under File Storage, Upload Media files, I have:
Upload Media Files ON
Upload Pictures ON
Upload Movies ON
Upload Service Google Drive
Location /camtest/ <<my test location in GDrive
Include Subfolders ON
Clean Cloud OFF
Where is the https: stuff coming from?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Jaizan, 2020-12-02:
I'm running on a Pi4.    The same Pi4 that's operated MotionEye properly for the last year, until I had to re-install.

Essentially, whatever I now specify as the Location for the upload service, is used literally within google drive.   If I put 5pPXGphfX****************, which is the code google gives me for my intended MotionEye folder, then instead of writing to the MotionEye folder, it creates & writes to 5pPXGphfX****************.     If I put the full URL in, it creates a multi-level folder structure, with the top folder called https: and so on.

I suspect Google have changed something, as even with the same software level I installed last year, it gets the problem.

starbasessd, 2020-12-02:
In File Storage:
Upload Media Files ON
Upload Pictures ON
Upload Movies ON
Upload Service Google Drive
Location /camtest <<my test location in GDrive
Include Subfolders ON
Clean Cloud OFF
In Still Images
filename set to %Y-%m-%d/jpgs/%H-%M-%S
In Movies
filename set to %Y-%m-%d/movies/%H-%M-%S
so on GDrive I get
/camtest/2020-12-02/jpgs/07-04-00.jpg

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Jaizan, 2020-12-02:
Thankyou starbasessd .    Am not quite sure where it went wrong.    When I tried typing ONLY the folder names directly in yesterday, it didn't work.   Now today I try a third folder name and it works fine.

closed 2021-03-04 by starbasessd