motioneye-project/motioneyeos #2597: Reboot issues with Odroid HC2

cap9qd, 2020-12-02:
### Preliminary Docs

Yes. 

### motionEyeOS Version

Dev20201026 (doesn't reboot)
20200606    (doesn't reboot)
20200203    (doesn't reboot)
**_20190911    (DOES! reboot)_**

### Board Model
 
Odroid HC2 Octa-Core (based on Odroid XU4) 

### Camera(s) 

I plan to use the following:
Several RPi Zero W FNC (latest with RTSP streaming) 
ReoLink 5MP RLC-410-5MP
IQEye 720p H264 (not sure the model but stream over rtsp) 

However, I have NOT added any cameras yet due to this reboot issue.

### Network Connection

Ethernet 1000M

### Peripherals

HC2 has an SATA controller but I have yet to connect an HDD. Only things connected to the HC2 is network, 12V 2A power supply and SD card.

### Log Files

Here are the log files from Dev20201026:
[boot.log](https://github.com/ccrisan/motioneyeos/files/5626832/boot.log)
[dmesg.log](https://github.com/ccrisan/motioneyeos/files/5626833/dmesg.log)
[sntp.log](https://github.com/ccrisan/motioneyeos/files/5626839/sntp.log)
[messages.1.gz](https://github.com/ccrisan/motioneyeos/files/5626837/messages.1.gz)
[messages.zip](https://github.com/ccrisan/motioneyeos/files/5626838/messages.zip)

### Issue:
So I got my new Odroid HC2 board hoping to get all my cameras into one hub. I loaded the latest DEV on a brand new new Samsung Pro endurance 32gb (sdhc class 11 and UHS1). It booted well and seems to run; I made changes to disable the WiFi (board doesnt have wifi) and static IP setup that required a reboot. However it hung on reboot with the red light. If I pull power and reapply it boots fine again.

I read about maybe the SD card could be the issue but I tried an old Samsung EVO card I use for all my other motioneye installs and it does the same. I also tried a class 3 SanDisk Extreme card and a handful of others (see pictures).

I downloaded and tried the XU4/HC1/HC2 Ubuntu 20.04LTS (Kernel v5.4) on the SanDisk Extreme card. (https://odroid.in/ubuntu_20.04lts/XU3_XU4_MC1_HC1_HC2/) It was able to reboot via ssh, once I was able to SSH into the installation (fyi default is root/odroid).

I have ordered a serial console cable (uses some screwy-to-me 3.5MM pitch Molex connector I didn't have) so maybe that will shed some more light on things. This HC2 board doesn't have an HDMI or any video output so it is hard to see what is happening on the hung reboot.

I walked back through the releases and got to 20190911 and this one reboots properly.  It did strangely startup with two cameras/video devices (one at /dev/video30 and one at /dev/video31). Not sure what these are...maybe something with the GPU...?  Ill run the info commands via ssh against them shortly...

Any thoughts? Anything I should try to figure out what is going on?

cap9qd, 2020-12-02:
Forgot to add the pictures... 
[image]

starbasessd, 2020-12-02:
After reboot, is it pingable? (ones that seem to not reboot)
Without making ANY changes, does it reboot normally?
Why Disable Wifi? How did you disable WiFi?
With your Ubuntu 20.04 LTS did you try the motion/MotionEye install here:
https://github.com/ccrisan/motioneye/wiki/%28Install-On-Ubuntu-%2820.04-or-Newer%29
(out of curiosity)

cap9qd, 2020-12-02:
It is not ping able when it's in this hung state.

All the attempts after the first install were vanilla with no changes; just
flash image to card, boot HC2 on the card and select reboot once the web
interface comes up.

I just disabled wifi by the network section in the web gui. See the
attached screenshot. The first img I burned from my RPi4 RaspiOS with my
standard script line with the Odroid XU4 image which has the wifi enabled.
The standard image I flashed after the first one doesn't have the wifi
enabled.

I have not tried motion/motioneye in thr latest Ubuntu. Thats the route I'd
go if there's not a solution for motioneyeos but thought I'd try to help if
there truly is a bug in MotionEyeOS.

starbasessd, 2020-12-02:
FYI attachments don't seem to get through when attaching to email replies, but I'll take your word for it.
I don't have any Odroids in my pile here to test. I do not know if it is a ThingOS / buildroot issue or something else.

Somewhat off-topic, have you thought about using it as a NAS, and just point the cameras to it for storage?

cap9qd, 2020-12-02:
Ah I'll attach them on the issue page. Sorry about that!

I have a Drobo 5N NAS I use as the storage (purchased pre Google drive
redible availability). I thought the HC2 might be good to be as a NAS but
will probably run MotionEyeOS/Ubuntu-20.04LTS for just cameras. I was
hoping that I could avoid making a dedicated multi-core server with this
octa-core HC2... But we will see.

cap9qd, 2020-12-02:
Here is the picture of the wifi disable. Like I said I used the writeimage.sh on an RPi4 with my normal arguments for an RPiZeroW (with the right odroid image) so it enabled it by my original image. Subsequent attempts have been with no changes to the image or install after boot. 
[image]

cap9qd, 2020-12-02:
Well not to pile on but it won't do an in place update to dev20201026 either from 20200606 (it did in place update from the 20190911 to 20200606 as you can see from the above screen shot). I have this on a "smart" plug plug so I can bonk-it after a hung reboot without having to be sitting at my workbench.

starbasessd, 2020-12-02:
Under Expert Settings, did you enable prereleases?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

cap9qd, 2020-12-02:
Yes.

starbasessd, 2020-12-02:
I am confused by this:
      Like I said I used the writeimage.sh on an RPi4 with my normal
arguments for an RPiZeroW (with the right odroid image) so it enabled it by
my original image.
I use Win32DiskImager, or RPi Image Writer. I use a wpa_supplicant.conf
file in the /boot partition. I use a ssh.txt (empty) in the /boot partition
to enable ssh at boot

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

cap9qd, 2020-12-02:
Sorry accidently sent the just yes.

I enable pre-releases and it says update avaliable. When I select update is
says updating and I get the spinny thing for a just a minute or so...then
says updated and doesn't reboot or show new version when I manually reboot
(halt, bonk, boot).

starbasessd, 2020-12-02:
Would you please run the following command in an ssh session?

sudo logrotate -f /etc/logrotate.conf

Then run the updates from the WebGUI
Then download and attach the motioneye & messages log files from the WebGUI
Please?

On Wed, Dec 2, 2020 at 10:04 AM cap9qd <notifications@github.com> wrote:

> Sorry accidently sent the just yes.
>
> I enable pre-releases and it says update avaliable. When I select update is
> says updating and I get the spinny thing for a just a minute or so...then
> says [...]

cap9qd, 2020-12-02:
I ran the log rotate and here are the files. Some were empty...and wouldn't upload here. 

[messages.log](https://github.com/ccrisan/motioneyeos/files/5630762/messages.log)
[motioneye.log](https://github.com/ccrisan/motioneyeos/files/5630763/motioneye.log)
[image]

cap9qd, 2020-12-02:
I ran log rotate and then tried the update. Sorry.
[image]
[image]
[image]

cap9qd, 2020-12-02:
Maybe I should have enabled debugging. I will do that and do my reboot trick. Then logrotate and then try the update.

Here are the files with debug enabled;only the motioneye log was > 0Bytes.
[motioneye.log](https://github.com/ccrisan/motioneyeos/files/5630891/motioneye.log)

cap9qd, 2020-12-02:
Looks like a failure to extract the firmware. Humm...

starbasessd, 2020-12-02:
Yep. Interesting. This could indicate other issues (like the reboot not
rebooting because something else is
broken.
I am not a programmer by any stretch of the imagination, but I can break
things good, and act as a translator between end users and programmers. I
wish I had unlimited resources to but all the SBCs CCrisan supports... PI's
I have at least one of each. I can build a VM for almost every OS. I have a
really whacked up network to imitate a lot of issues. But no Odroids or
other clones...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-12-02:
Yep. motioneye.log caught what I was looking to flag to @ccrisan
It's a problem in updating firmware. I'll flag this incident as a 'bug'.
Any chance you can download and flash dev20201026 to an SDCard?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

cap9qd, 2020-12-02:
Yep sure thing I'll get it back to the latest DEV when I get home... I can
VPN and ssh quite a bit on my lunch break from work but not do the physical
stuff.

I'll probably work on the Ubuntu 20.04LTS install to get motioneye/motion
up and running but I'll keep my MotionEyeOS card at the ready for any
testing if CCrisan needs anything from me or test a new release. I want to
help any way I can as I have gotten a lot out of this project and don't
feel I've contributed much!

I'd like to continue my testing to see if I can get all 7 or 8 cameras up
and runnon on this single board so I'll play with Ubuntu for now...

cap9qd, 2020-12-02:
I did some some poking to see if I could download the firmware manually and extract it manually and found ccrisan pos in #1527. I tired this manual firmware steps and got the attached firmware.img not found error. 
[image]

cap9qd, 2020-12-02:
I manually downloaded the firmware to /data/.fwupdate/firmware.img.xz and tried. Still failed on extract. 
[image]

cap9qd, 2020-12-05:
Okay so my serial interface cable came today. I can see the bootloader output but the console seems to die/inactive after the kernel boots. 

However, it appears that on reboot the device actually stops but fails to boot.  I will figure out the console TTYs for this board from my Ubuntu 20.04LTS (the console works the entire time) and see if I can update the boot.ini.

Please see the attached boot log pulled from my putty session connected to the serial console.  Like I said the serial console seems to stop right after the kernel is started.  This may be due to my serial settings; it appears the serial console in Ubuntu and the bootup seems to be auto-detected as I fat-fingered the baudrate to 115200.  Okay I added the bootargs "console=ttySAC2,115200n8" from the Ubuntu boot.ini to the MotionEyeOS boot.ini to get the serial console going after doing the "debug mode" thing.

Boot log from Serial console:
[Odroid_XU4_MotionEyeOS_dev20201026_BootLog.txt](https://github.com/ccrisan/motioneyeos/files/5646173/Odroid_XU4_MotionEyeOS_dev20201026_BootLog.txt)

Reboot Log from serial console:
[Odroid_XU4_MotionEyeOS_dev20201026_RebootLog.txt](https://github.com/ccrisan/motioneyeos/files/5646183/Odroid_XU4_MotionEyeOS_dev20201026_RebootLog.txt)

Interestingly I do not get an interactive console from the serial port in MotionEyeOS like I do in Ubuntu.

Also, I struggled (in the few days I worked on it) to get a stable and running Ubuntu 20.04LTS system with MotionEye and Motion.  I built FFMPEG from source (enabling the patches I found on here), build motion 4.2.2 from source and installed MotionEye.  I got a VERY high CPU usage with just 4 cameras (one 720P H264 with passthrough, one 5MP ReoLink 410 camera and 2 FNC MJPEG cameras of reasonable FPS and resolution); I enabled the GPU driver for XU4 like @jasaw suggested in an issue I found and that helped the MJPEG cameras. Adding the 4th 5MP ReoLink camera would cripple it.  

With MotionEyeOS I have all of this running at 30% to 40% load looking at the output of TOP cmd.  I cant reboot but it runs much much better...  I am setup the BuildRoot for MotionEyeOS if I can figure it out and see what I can come up with.

starbasessd, 2020-12-05:
You are now way beyond my ability to help. Sorry, and good luck, maybe
another user of an ODroid can help...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

cap9qd, 2020-12-05:
> You are now way beyond my ability to help. Sorry, and good luck, maybe another user of an ODroid can help...
> […](#)
> -- Thanks Kevin Shumaker Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel> N38° 19' 56.52" W85° 45' 8.56" Semper Gumby “Don't tell people how to do things. Tell them what to do and let them surprise you with their results.” - G.S. Patton, Gen. USA Ethics are what we do when no one else is looking. Quis custodiet ipsos custodes? “There is no end to the good you can do if you don’t care who gets the credit.” - C Powell You know we're sitting on four million pounds of fuel, one nuclear weapon and a thing that has 270,000 moving parts built by the lowest bidder. Makes you feel good, doesn't it?

No worries! I appriciate the help and have used things you've listed on here in my learning. 

I subscribe to these issues on GitHub, read all the emails (with my dropins on the threads I see I can help) and can see all the help you provide and your demonstration of pure zen patience most of the time. I appriciate it greatly even if you can't help me directly with this issue.

cap9qd, 2020-12-13:
I got a new HC1 (Same board but with a heatsink for a 2.5" drive and 5VDC instead of 12VDC) and have the same issue.  It hangs and does not reboot.

ccrisan, 2020-12-13:
What I know from my Odroid XU4 is that it refuses to reboot. Anytime I need a reboot, I have to manually pull out the plug and plug it back. I have no idea why this happens but I have never seen it as a serious issue.

starbasessd, 2020-12-13:
My google-fu is strong today, so I checked.
Many references to failure to 'sudo reboot', but if it's a 32GB or smaller
SDCard, it seems to reboot more consistently...
Just out of curiosity, what size is your SDCard?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

cap9qd, 2020-12-14:
It's normally not an issue. I have the network reboot enabled so I'll
disable it so it doesn't unexpectedly reboot. There's lots of talk on the
hardkernel about sd card issues but none I tried seems to help.
MotioneEyeOS works so so much better than my attempt at an Ubuntu install
with Motion/MotionEye and I greatly appriciate the excellent project!

I was really hoping the serial console would shed some light but was next
to useless.

I'll close the issue since it's a known issue.

cap9qd, 2020-12-14:
> My google-fu is strong today, so I checked. Many references to failure to 'sudo reboot', but if it's a 32GB or smaller SDCard, it seems to reboot more consistently... Just out of curiosity, what size is your SDCard?

I have lots of 32GB cards I tried, I tried a 16GB and a 64GB. All had the same issue.

Since it's not a huge deal... It was only an "issue" on the initial setup and now it's been fine.

cap9qd, 2021-01-23:
In an attempt to troubleshoot a stability issue with my HC1 hardware (it will just shutdown/reboot with no messages in the serial console) I modified the boot.ini setenv line to the following by removing the `quiet loglevel=1`.

`setenv bootargs "console=tty1 console=ttySAC2,115200n8 root=/dev/mmcblk1p2 rootwait ro no_console_suspend panic=10"`

Now the board reboots.  Not sure why this would have any effect but it appears to point to an issue.  I have rebooted about 10 or 15 times in a row with no issues!

closed 2021-02-07 by cap9qd

benjvoigt, 2021-07-01:
@cap9qd @starbasessd I have the same issue with my Odroid XU4Q, it won't reboot by default. It just shuts down (without any error) when I enter reboot. What works is indeed the modification of the `boot.ini` as described by @cap9qd. Awesome! But it's weird. I tried various combinations for console and log settings but the only line that works is this:
`setenv bootargs "console=ttySAC2,115200n8 root=/dev/mmcblk1p2 rootwait ro no_console_suspend panic=10"`

Once you add either `quiet`, `console=tty1` or `loglevel=1` it won't reboot. I also tried `tty` and `tty0` without any luck.

So can this issue be reopened pls? There are some hints now to investigate it further I guess.

starbasessd, 2021-07-01:
Please open a new issue, answer the questions reference your setup, and refer to this issue ticket.
I do not have any odroids, and I am not sure if Calin does, either, for testing.

cap9qd, 2021-07-01:
I believe that Calin's production machine is an Odroid XU4 (same image as
HC1/2), as he mentions it elsewhere if not in this issue, and has reboot
issues. I don't think he sees it as a priority. With the changes I mention
my odroid works well and has been working well ever since I posted and
closed this issue. I do have issues where it panic reboots occasionally and
so these changes are needed for me to have a stable system.