motioneye-project/motioneyeos #2598: script and Use GPIO

sRspectrect68, 2020-12-03:
i'm trying to create a pyton script to activate a buzzer on the GPIO, but i'm getting import errors on the libraries, any ideas?

`#!/usr/bin/python

import sys
import os
import signal
import time
import cStringIO
import pycurl
import RPi.GPIO as GPIO`

starbasessd, 2020-12-03:
You could try to turn on 'debug' mode (under Expert Settings (enables
system r/w) then when completed, turn it back off.
Any files you need to write to need to live under the /data folder tree.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

sRspectrect68, 2020-12-03:
Hi! . 
I tried to get the same error ...
`#!/usr/bin/python
#Libraries

import sys
import os
import signal
import time
import cStringIO
import pycurl
import RPi.GPIO as GPIO

e warnings (optional)
GPIO.setwarnings(False)
#Select GPIO mode
GPIO.setmode(GPIO.BCM)
#Set buzzer - pin 23 as output
buzzer=23
GPIO.setup(buzzer,GPIO.OUT)
import socket
UDP_IP = "xxx.xxx.xxx.xxx"
UDP_PORT = 13000
sock = socket.socket(socket.AF_INET, # Internet
socket.SOCK_DGRAM) # UDP
sock.bind((UDP_IP, UDP_PORT))
while True:
     from datetime import datetime
     today = datetime.now()
     dt_string = today.strftime("%d/%m/%Y %H:%M:%S")
     data, addr = sock.recvfrom(1024) # buffer size is 1024 bytes
     text = str( data )[2:-1]
     print ("RX message",dt_string," - ", text)
     if ('ALLARM' in text):
            GPIO.output(buzzer,GPIO.HIGH)
            print ("Beep")
            sleep(1) # Delay in seconds
            GPIO.output(buzzer,GPIO.LOW)
            print ("No Beep")
            sleep(1)

`

starbasessd, 2020-12-03:
did you install all the libraries before running your python?
I

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

sRspectrect68, 2020-12-03:
HI. I have installed the latest version of motioneye for Raspy Pi3.  From what I understood the pyton libraries were already installed.  So should I update pyton or just load libraries?

starbasessd, 2020-12-03:
Python would be updated to the day before the date of the version of
motioneEyeOS, but the libraries you need may not be there, or
pre-compiled.(pyc) and not usable in your tools.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

sRspectrect68, 2020-12-04:
Hi! 
Ok, I'll try to update Pyhton and the libreires and then try again...

sRspectrect68, 2020-12-04:
sorry, before getting into trouble, there is a special procedure to update Python, raspberry commands work

sRspectrect68, 2020-12-04:
On motioneye I don't think it is possible to install anything, so I can't update Python. I had tried to install Motioneye on Raspberry (Pi3 + latest rasbian version), but I got many errors and it was not possible.

starbasessd, 2020-12-04:
You can install files on a 'normal' RaspberryPiOS, then copy the files to
the motionEyeOS (if you turn on Debug)
, but you are correct, you can't do 'normal' 'apt updates' or 'apt
install's in motionEyeOS. We do the install/copy for other things, like USB
wifi adapters, USB camera adapters, etc. I've copied python programs and
bash scripts over on a regular basis. Not "happy making" but can be done.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

sRspectrect68, 2020-12-04:
Hi!
Thanks for the replies. I just managed to install Motioeye on a normal raspberryPi3 distribution. Now I should be able to run all Python scripts without any problems.

starbasessd, 2020-12-04:
There is always that <grin>
You can use Raspi-Lite, and it still works. Don't forget access changes to
http://[ip_address]:8765

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

sRspectrect68, 2020-12-04:
😀👍

closed 2021-03-04 by starbasessd