motioneye-project/motioneyeos #2599: RPI 4 - fails to load 20200606, sticks on rainbow screen dev20201026

xxxTark77xxx, 2020-12-06:
Greetings, I've seen this question previously but not solved in my case.  I have an RPI 4 8G

I installed the RPI4 20200606 release and the load failed with the text below.  I tried it a second time with the same result.
Raspberry Pi - 8GB
Bootloader: c305221a Sep 3 2020
Config: 86697050
………
This board requires newer software
Error: 00000044
Insert SD-CARD

I then found the development verion DEV20201026.  The first time through, everything loaded correctly and I was able to set up and use the application.  After making some camera adjustments, the system wanted to reboot and froze in the startup rainbow screen.  Rebooting ended stuck at the rainbow screen.  I reinstalled DEV20201026 and it froze at the rainbow screen right away.

Any ideas?
Thanks, Barry

starbasessd, 2020-12-06:
I am able to reproduce it with my Pi4-8GB.
I will see if I can find a work around other than the install RaspiOS and
motion/motionEye.
I will report shortly.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-12-06:
20200606:
I have tried flashing back to previous eeprom versions, no luck.
dev20201026:
unable to reproduce. 
install on 8GB SdCard
add wpa_supplicant.conf and ssh.txt to /boot partition
install in RPi4-8GB
first boot, normal, add mjpeg network cam, change framerate, change timezone, etc.
reboot
get rainbow screen for a few seconds (<5) and boots normally.
add another camera (rtsp) repeat, reboot
boots normally.

xxxTark77xxx, 2020-12-06:
Thanks @starbasessd.  I've tried a few more combinations based on your post, but no luck.  In summary, with dev20201026, the system will start and function the first time, but after a reboot or shutdown, it starts up to the permanent rainbow screen.

System: RPI4 8G / 32G Samsung EVO

Setup
* Format drive with SD Card Formatter
* Flash with balenaEtcher and RPI installer
* Create wpa_supplicant.conf and ssh.txt with Atom
* Eject drive

Method 1 - Single camera
Rainbow screen 5 secs then entered setup 
I changed - Admin password, Camera settings (framerate and size)
Rebooted
Rainbow screen permanent

Method 2 - Single camera
Rainbow screen 5 secs then entered setup 
Changed nothing
Rebooted
Spinning icon in the program indefinitely
Hard rebooted
Rainbow screen permanent on startup

Single camera
No Rainbow screen?  Went straight to setup this flash 
Changed nothing
Power down, wheel spun then displayed message "Powered Off"
Turned system back on and had rainbow screen permanently

starbasessd, 2020-12-06:
Do you have anything that can read the SDCard with ext4 partition? I'd
really like to get a look at the logs.
BTW, I use Win32DiskImager, I don't pre-format the SDCard, I use generic
MicroCenter SDCards in 8-64GB sizes.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

xxxTark77xxx, 2020-12-06:
I'd be happy to send the logs but not sure how based on what you asked.  

I'll try again with Win32DI

Based on the order of operations you mentioned, I noticed that you installed the cameras after startup while I had them plugged in.  I just tried it with no cameras and the result is below.  I have an RPI4-4G currently in use for something else but I'll try a build with it now and see what happens. 

No Camera
Rainbow screen 5 secs then entered setup 
Changed nothing
Power down, wheel spun then displayed message "Powered Off"
Turned system back on and and it reloaded correctly
Rebooted, Spinning icon in program for a long time and then rainbow screen permanently

xxxTark77xxx, 2020-12-06:
The trial with Win32DI had the same results.  I'll try the RPI4-4G now.

Same process with Win32DiskImager, no pre-format
No Camera
Rainbow screen 5 secs then entered setup 
Changed nothing
Rebooted.  System came back on and reloaded correctly
Rebooted, Spinning icon in program for a long time and then rainbow screen permanently
Manual reset, rainbow screen permanently

starbasessd, 2020-12-06:
Have another SDCard to test?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

xxxTark77xxx, 2020-12-06:
Tried the build on RPI4-4G / 32G Samsung EVO, with similar results.

No camera
Entered setup, no rainbow screen
Changed nothing
Rebooted.  Rainbow screen 5 sec then system came back on and reloaded correctly
Rebooted.  Rainbow screen permanently
Manual reset, loaded correctly.
Rebooted.  Rainbow screen permanently
Manual reset, rainbow screen permanently
Manual reset, rainbow screen permanently

starbasessd, 2020-12-06:
I'd like to see the logs after you get to the rainbow screen. To do so, you
need a Linux box that can read the SDCard's ext4 partitions, or you could
try DiskInternals free tool:
https://www.diskinternals.com/linux-reader/ (freeware version)

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

xxxTark77xxx, 2020-12-06:
I've been looking around and I don't have a spare SD card, my only other one is running my Octoprint server so I can't erase it.  

I don't have a Linux box.  With diskinternals, does that run on a PC and I ssh to get the files from the pi?  Or does it go on the pi and I grab the files.  I'm afraid that once the system reboots and goes into the locked rainbow screen, I won't be able to access the drive anymore.

starbasessd, 2020-12-06:
It installs on the PC that you burned the image on (Win10) and you stick
the SDCard back into the PC.
Most of the time, it will allow you access to the /root and /data
partitions (not always) and you can copy the log files off the SDCard...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

xxxTark77xxx, 2020-12-06:
I really appreciate all of your help!

If I understand correctly.  I'm going to install it on my PC.  Then after the SD card in the Pi fails, I'm putting it back into my PC and looking for the log files?

starbasessd, 2020-12-06:
Correct.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

xxxTark77xxx, 2020-12-07:
I installed DiskInternals and did some checks.  I couldn't find any log files on the SD card before or after running motioneyeos.  I poked around and also searched for "log".  Is there something different that I should be looking for? 

After Image flashed, launched DiskInternals. no log files found
Rainbow screen 5 secs then entered setup 
Changed nothing
Power Shut Down, wheel spun then displayed message "Powered Off"
Put SD card in PC, launched DiskInternals.  Found new folder /data, nothing in it.  No log files found.  
Put card back in RPI4
Rainbow screen permanently
Put SD card in PC, launched DiskInternals.  No log files found.

starbasessd, 2020-12-07:
should be under /var/log or /data/var/log

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

xxxTark77xxx, 2020-12-07:
Within DiskInternals, the /data folder says "There are no recoverable files in this folder".  I searched the drive for *log and nothing either.

starbasessd, 2020-12-07:
It happens to a few of my SDCards, too.
I am at a loss.
If you image the SDCard with RaspberryPiOS 2020-12-02 or 2020-08-20, they
boot & reboot normally?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

xxxTark77xxx, 2020-12-07:
Yes.  I just used Raspberry Pi Imager and flashed Raspberry PI OS (32-BIT) 2020-12-02.  I'm able to view both USB cameras using VLC.  

Any ideas?  Is it possible to load the 20200606 version of motion/motionEye on top of my installed RPiOS?

starbasessd, 2020-12-07:
Not exactly, but you can do motion/motioneye using these instructions:
https://github.com/ccrisan/motioneye/wiki/%28Install-On-Ubuntu-%2820.04-or-Newer%29
Ignore it says Ubuntu. Also the WebGui will be http://[ip_address]:8765

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

xxxTark77xxx, 2020-12-07:
A million thanks for your continued help.  The installation instructions are really clear and the install was successful.  Thanks for highlighting the port is 8765, I would have missed that.  All cameras are up, viewable remotely, and I can reboot the system repeatably.  The only note during install is that Python 2.7 reached the end of its life on 1/1/2020 and updating is recommended.

Thanks for your work on this application, it's great.  I was planning on multiple mjpg streams, this is way better.   I'll definitely donate to the project.  

Thanks, Barry

closed 2020-12-07 by xxxTark77xxx

starbasessd, 2020-12-07:
Yeah it's known that 2.7 is on it's last leg, that's why we needed the
different install instructions.
If you don't need to use the Pi ad a desktop, I recommend using
RaspberryPiOS Lite (command line version,) No real need to have the Desktop
installed and taking up un-needed resources and space.
Glad to help.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

xxxTark77xxx, 2020-12-07:
That's a good point.  Any tricks for copying the instruction text from the browser into the command line?  I think that if I don't have a browser, I'll be hand typing the lines.

starbasessd, 2020-12-07:
Once you get it up on the Pi I ssh into the Pi from another computer and do
the copy/paste thing there.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

xxxTark77xxx, 2020-12-07:
Got it.  Can I ssh into the RPi over wifi using PuTTY from my PC?  If so, do I just use the local IP?

starbasessd, 2020-12-07:
Yep.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

xxxTark77xxx, 2020-12-07:
Sorry for the additional question.  Is there any documentation on the wiki for setting up port forwarding, I couldn't find anything?  I read through issue notes and got the gist.  

The RPi has a static IP.  
On my router, I created a Port Forward service for that IP address
Selected Service Type HTTP.  
SEt External Start and End Port to 8765

Is this correct?  What address do I use externally to connect?

starbasessd, 2020-12-07:
Depending on how often it changes:
https://whatismyipaddress.com/
shows current.
You can use a dynamicDNS service like dyndns.com
I have another Linux box email me the external IP address using a basic
script every 2 hours:
I did:
sudo apt install sendemail [enter]
to give me an easy way to send emails.

in
sudo crontab -e [enter]
I added a line:
10 0,2,4,6,8,10,12,14,16,18,20,22 * * * /boot/extip.sh
so ever even hour, 10 minutes after the hour, It runs a script called
/boot/extip.sh

in that script I have:

#!/bin/bash
myip=`wget -q -O - checkip.dyndns.org|sed -e 's/.*Current IP Address: //'
-e 's/<.*$//'`
sendemail -f my_gmail_email_address -o tls=yes -f  my_gmail_email_address
-t  my_gmail_email_address -s smtp.gmail.com:587 -xu
my_gmail_email_account-xp my_password -u "ExtIP" -m $myip

Works well enough for me.
I check for the last email, get the external IP address, and then do:

http://[extIP_address]:/port  << whichever port is being forwarded to the Pi

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

xxxTark77xxx, 2020-12-07:
Kudos, that is a very clever solution!

xxxTark77xxx, 2020-12-14:
Greetings Kevin, happy Monday!

Thanks for your help getting MotionEye running, 2 cameras are working great.  On a Pi4B-8GB, how many USB cameras would you expect could run simultaneously when directly connect to the RPI's USB ports?  When I connect a 3rd USB camera, I'm not able to stream, even with frame rates and quality at the lowest settings on all 3.  Recall that my setup used the full RPI OS and Motion vs MotionEyeOS because the install kept failing.  I'm not sure how to tell if the RPi has limitations, camera limitations, or the OS is taking too much resource.  If you said it would help, I could RaspberryPiOS Lite vs the desktop.  Any thoughts on what the RPI4 should be able to handle?

Thanks, Barry

starbasessd, 2020-12-14:
On network cameras I can run 5 fairly easily on a Pi4-4GB (32gb USB drive).
RaspberryPiOS Lite with no other HW connected. (no monitor or keyboard,
just ethernet)
If all are connected via USB, you have several issues to be concerned about:
Powered by PI? Not a good idea.
Powered by self-powered USB-3 hub? Bandwidth in use.
You can run top, or neofetch (have to turn on CPU usage).
For bandwidth usage on the USB bus I found a cool tool:
https://ostechnix.com/how-to-find-usb-device-bandwidth-usage-on-linux/
need to force load in modprobe usbmon, instructions here:

To load a module on boot, sudo nano  /etc/modules-load.d/usbmon.conf

this file can have any name, but must end in .conf.

In the file, add a single line with the name of the module you want to load
like so:

usbmon

Reboot and then

sudo usbtop

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-12-14:
I have a TIGA generic dashcam that shows in
sudo usbtop:
 Device ID 2 :                 142.06 kb/s     11368.36 kb/s

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

xxxTark77xxx, 2020-12-14:
Great, I'll do a rebuild with RPiOS.  I couldn't tell, what is the ideal setup for multi cameras if not using the direct USB ports on the RPI for power concerns?  If I get a powered USB hub (through USB 3.) am I not creating a USB bandwidth constraint?

starbasessd, 2020-12-14:
How about: It Depends.
How well does the USB3 self-powered hub translate data speed of USB 2
devices? Or are your cameras USB3?
You have a theoretical max data rate for USB3. Should be a lot faster for
USB3 than USB2.
Since ethernet is no longer on the same bus as the USB, you don't have that
interfering.
Whole 'nuther can-o-worms.
I'd do a minimal Desktop or Raspian Lite (preferred) install to minimize
other software using resources...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

xxxTark77xxx, 2020-12-14:
=)
My take-away is to try 1) rebuild with minimal RPiOS Lite and direct camera plug in.  If not working 2) Get a powered hub and monitor the usb bandwidth and CPU usage.  Correct?

starbasessd, 2020-12-14:
You're probably tired of me saying:
Depends. This time solely on power needed by USB Camera(s)
You can install the cameras, then run the command:
vcgencmd get_throttled to see if you get a low voltage indication. If it's
anything but 0x0, post it here and I'll tell you what failed (temp or
voltage, or you can look here:
https://harlemsquirrel.github.io/shell/2019/01/05/monitoring-raspberry-pi-power-and-thermal-issues.html

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

xxxTark77xxx, 2020-12-14:
No at all!  I greatly appreciate your time.  I'm pretty new to RPIs and learning as I go.  Your insights and sharing are awesome, I can't thank you enough.  Every new command is gold, I'm not sure where else to find out about these.

I'll do a rebuild with RPiOS Lite, check the voltage, and post back.

Cheers! Barry

starbasessd, 2020-12-14:
top shows all processes running and statistics of cpu usage, memory usage,
etc.
usbtop shows bandwidth usage of all the usb connected devices
lsusb -t shows what devices are connected to which usb ports, and details
like vendor and manufacturer ids.
you can usually tpe
man xxxx [enter] and get a good description and usage instructions
or in a browser, most of the time you'll get the same thing...
The only dumb question is the one that doesn't get asked...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

xxxTark77xxx, 2020-12-15:
The monitors were really helpful.  I played with the 3 cameras with both monitors running and here is what I learned.

Setup - I used a powered USB Hub 2.0. No camera combo causes a Temp or Volage failure.
In Motion, I configured 3 cameras the same with Frame Rate 15, Quality 50%, 960x720.  Everything else was disabled.

USBTOP KiB/s (To device  //  From Device)
Device 11  141   //  15000 (Generic 1080p)
Device 12   141  //  3000   (Logitech c920) - not sure why so low relative to the other two, the picture is better.  
Device 13   141  //  15000  (Logitech c310)

Any 2 cameras will run at the same time and draw %CPU - 50% motion, 25% meyectl, rest negligible.
If all 3 cameras are enabled, 1 will go blank or show a still image.  It's not always the same camera.  The bandwidth monitor then alternates from all 3 cameras at 0 KiB/s to normal and back again to 0. 

It seems that 3 cameras running utilizes 33 MB/s bandwidth which is way below USB 2.0 limits, and the CPU % doesn't exceed 75%.  When I was experimenting, I could run 2 cameras fine at 30 FPS and utilize CPU% 150% fine.  Unless I'm not understanding correctly, it doesn't appear the issue is a lack of processing capacity or data bandwidth.  So, not sure why 3 won't work.

Any ideas?

starbasessd, 2020-12-15:
Can you split the 2 Logitechs? One on the hub, one direct to the Pi? Then
rotate between the 3 (which is on the Pi vs which 2 are on the hub)?
See if any combo allows all 3 to show up consistently.
Can you get the vendor:device ID from lsusb for all 3 camera? I'm looking
for any that have the same vendor:device. I need the part after ID

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

xxxTark77xxx, 2020-12-15:
I tried a ton of camera and powered hub combos, there seems to be a camera prioritization that I don't understand.

Device IDs - Headless setup. Powered USB 2.0 Hub (empty), 3 Webcams.  The blank cam is named PHY.
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 006: ID 046d:082d Logitech, Inc. HD Pro Webcam C920
Bus 001 Device 005: ID 046d:0990 Logitech, Inc. QuickCam Pro 9000
Bus 001 Device 004: ID 1224:2a25
Bus 001 Device 003: ID 0424:2507 Standard Microsystems Corp. hub
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

**Observations**

- Any two cameras work together, in any USB port, in the Pi or in the Hub.
- 3 cameras never work together, in any port combination, in the Pi or in the Hub.  When I plug a 3rd camera in, 1 camera always drops, but the dropped camera changes based on the combo and hub use.
- If no hub is used (all 3 into the RPI), the LT9000 always drops.
- If 2 cameras are direct and 1 is through the hub, the dropped camera depends on the combo.  With either the PHY in the hub or LT9000 in the hub and the other 2 direct, the LT9000 drops.  With the LT920 in the Hub and the others direct, the LT920 in the Hub drops.
- With 2 cameras in the hub and 1 camera direct, the dropped camera depends again on the combo in the hub and direct, but also which USB port is used.
- When all 3 cameras are in the hub, the LT9000 is always dropped.

The number of permutations got crazy so I thought I'd check back in to see if you have any ideas how/why the cameras are being prioritized to drop.  And why 1 always drops and it's never the PHY.

Thanks!
Barry

starbasessd, 2020-12-15:
I really am at a loss. Should work, unless there is a limitation of # of
usb cameras allowed in motion/motionEye.
Well I typed that and did a google. Found this:
https://motion-project.github.io/motion_config.html
USB cameras take a lot of bandwidth. A USB camera connected to a USB2 port
or hub consumes virtually all the bandwidth that the port can handle. Even
with a small framesize and low framerate you should not expect to have more
than one camera per USB controller. If you need more than one USB camera it
is possible to add extra USB PCI cards to your computer. It is also
possible that your computer has more than one controller for different USB
slots. For example, if the computer has six USB ports, the three on the
left may be on controller A versus the three on the right may be on
controller B. In this situation, connecting two cameras to ports only on
the left would not work due to excessive bandwidth. However connecting one
camera to port on the right and the other to the port on the left may work
since they are on different controllers.

It's not clear to me if this is the case here, but...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

xxxTark77xxx, 2020-12-15:
I'm stuck too.  The bandwidth monitor shows far less usage than capacity.  I read about other people using more than 2 cameras.  You mentioned that you have ~5 I think.  What is your setup?

starbasessd, 2020-12-15:
Only 1 USB, Generic TIGA Dashcam, 3 D-Link wifi ptz DCS-5020Ls and 1 FosCam
RTSP PTZ for production. 1 USB TIGA Dashcam, 1 PiCam on a PiZeroW, A coupe
other cameras with BT/WiFi.
Production Server is a Lenovo i5 / 16GB / 250 GB on Ethernet with Ubuntu
20.04 TFF PC
Test Pis can be PiZero, PiZeroW, Pi1, Pi2, Pi3B and B+, Pi4 4GB & 8GB, or
any flavor Linux on Ryzen 7, 32GB / 10TB / 4 ethernet ports.
House is wired for 1Gbit ethernet, Running Cisco 2960 managed switch, and
several Belkin 5 port Gbit switches. 11AC and 11B,G,N Wifi Access Points
(Camera systems on their own VLAN)
Ubiquity Router fed pass-through from ISP Cable modem.
This stuff is what's related to my support of motionEye & motionEye OS. Son
and his girlfriend and 2 wannabe grandsons with their own equipment, PCs,
tablets, and only god knows what else and IoT stuff not inventoried...
Watcha think?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

xxxTark77xxx, 2020-12-15:
OMG!! That is a setup for sure.

So, I have a print farm that I need to monitor remotely with ~6 cameras.  It's all in the same location so the cameras can be wired but the "server" won't reach a hub/router with ethernet.  What setup would you recommend?  I've tried MJPG Streamer and Motion, but can't get things working.

xxxTark77xxx, 2020-12-19:
The fun continues!  I thought that maybe the one cheap camera may have been the culprit.  I rebuilt the system with RPiOS lite and ordered two decent webcams (the same model).  Happily, I was able to get the 3rd camera working.  When I plugged in the 4th, I found that the system assigns different device numbers to the 2 cameras, but the device id's are the same.  Is there any way to use both if the ids are the same?

starbasessd, 2020-12-19:
That is another issue we've found. 2 cameras with the same vendor:model USB
ID confuses the system,
After I've had breakfast and caffeine, I'll give you some ideas for that
issue.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-12-19:
take a look at
ls -la /dev/v4l/by-id/
and see which cameras are listed there.

Then take a look at the /etc/motioneye/camera-x.conf files.
You are looking for files that have the same stream_port or videodevice

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

xxxTark77xxx, 2020-12-30:
Thanks for all of your ongoing help.  I have done a ton of troubleshooting and have 3 cameras working reliably now.  I've ended up with 2 RPIs and 5 webcams to figure this out and I learned way more Linux, webcams, and the bandwidth monitoring than I meant to with this project=) The combination of problems is/was:

-  The wifi signal/bandwidth is a limiter.  Given the randomness of how different cameras dropped and connected, I thought it was a USB or RPI processing issue.  Working close to the router through wifi or plugging into the router solved some of the issues.
- My original RPI has a finiky USB port that stops working in certain positions.
- 2 of the cameras have the same device id and can't operate at the same time.
- One camera works only sometimes.   I guess that's what I get for buying the $10 one.

Thanks again for all of your help Kevin!  It's been fun and I hope we connect again in the future.
Barry

starbasessd, 2020-12-30:
You're welcome.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

rrirower, 2021-05-06:
I'm experiencing the same problem on my Pi4 8gb as the original poster @xxxTark77xxx .  I've tried several incantations of formatting and installing.  I've used the Pi imager and balenaEtcher, but, no luck.  The boot stops with the same messages as in the first post of this issue.

Is there a resolution?

matthew1471, 2021-05-20:
Same issue with a new Pi4 8GB for me too flashed : motioneyeos-raspberrypi4-20200606.img.xz