motioneye-project/motioneyeos #2601: motioneyeos smtp settings not being retained

peterd550, 2020-12-07:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

motionEyeOS 20200606

### Board Model

I am using the following board/model: Raspiberry pi zero w.

### Camera

I am using the following type of camera: /dev/v4l/by-id/usb-lihappe8_Corp._USB_2.0_Camera-video-index0

My camera model is: (insert your camera model, e.g. Logitech C290).

### Network Connection

My motionEyeOS unit is connected to the network via: wifi 2.4 ghz

when I set the smtp server it works and sends email

when I turn off send notification my smtp settings are lost

[motion (4).log](https://github.com/ccrisan/motioneyeos/files/5650466/motion.4.log)
[motioneye (3).log](https://github.com/ccrisan/motioneyeos/files/5650467/motioneye.3.log)
[dmesg (1).log](https://github.com/ccrisan/motioneyeos/files/5650468/dmesg.1.log)
[messages (2).log](https://github.com/ccrisan/motioneyeos/files/5650469/messages.2.log)

peterd550, 2020-12-07:
I am also having an issue saving videos to my Nas 
I can connect but when I set it to be a 30 sec continues video I get 1 video

I set it to be 1 sec video and it works get 60 videos in a minute.

starbasessd, 2020-12-07:
The continuous video recording is a known issue. It should work as expected
with 20190911 or dev20201026.
Checking the email issue, but, if I recall correctly, that behavior has
been there since before 20190911, by design.
If you want to not have it send emails for motion detection, you can
disable motion detection by script or toggle.
It assumes you want to set up an email when you enable it, ignoring any
previous setup.
This could be a feature request, to do a Working Schedule for Email, too.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

peterd550, 2020-12-08:
Why is it by design?
When is the next release for this ?

starbasessd, 2020-12-08:
It doesn't assume you want to keep previous settings, rightly or wrongly.
When is up to the owner of the project, @ccrisan

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

peterd550, 2020-12-09:
Can I make a request that it does settings are retained.

peterd550, 2020-12-28:
When is the development version going to be a production version?

peterd550, 2021-11-08:
Any update on this issue?

starbasessd, 2021-11-08:
No further updates at this time. I can write you some scripts to save the settings, if you need, but I doubt they would be 'official'...

peterd550, 2021-11-08:
Yes please that would be great!
What language will they be in?

Do you have any API's for motion eye

starbasessd, 2021-11-08:
APIs, no, the code is open source, though (mostly python scripts).
I've written them before and posted either here or in motionEye, if you look for them
(they are related to SMTP or email alerts) and find them, let me know. 
I am working on updating and confirming motionEye on RaspiOS Bullseye since it was
officially released today (even though it's dated 10/30/2011) and as soon as I get that completed, I will look them up and/or post new ones. They are bash scripts, as all it does is copy the Camera_X.conf scripts to 2 other files, then you can either crontab them (fixed times) or Action Button them (manual times) or both...

starbasessd, 2021-11-08:
Here's a link to a similar situation where they wanted one camera setup for daytime, and a different setup for night. You could just adjust the settings you want for smtp for when you want emails, and one for when you don't want emails (but still want motion detection), and adjust the crontab time settings as needed. If you want Action Buttons, let me know and I can point you to help... If you need more assistance, or have additional questions, I believe my email is in my profile.

starbasessd, 2021-11-08:
I suppose it might have been helpful to include the link, but my brain is mush from doing the Docs for RPiOS Bullseye. Have I ever said how much I hate github MD lanquage? :)
https://github.com/ccrisan/motioneyeos/issues/2779#issuecomment-882773284